hzyitc/openwrt-redmi-ax3000 #1: 开贴庆祝一下🎉

fecent, 2022-12-03:
感谢某F，某P，某M，某H，某Y使这个项目得以实现，庆祝一下🎉

Pinghigh, 2022-12-12:
请问有教程吗，开ssh之类的

hzyitc, 2022-12-13:
README已更新

jessehoo89, 2023-01-29:
大佬加油，cr8809已饥渴难耐了，感谢大佬适配

NBSOD, 2023-02-28:
C8806完美刷机成功，十分感谢大佬的无私分享

mmfmkuang, 2024-03-18:
> 目前手头几台8808情况： 生产2021.11-m81-IPQ5000 生产2022.3-m79-IPQ5000 生产2023.10-m79-IPQ5018（主板印20230713）
> 
> 补充： 最新IPQ5018的8806 8808主板好像是一样的 IPQ5018的机子叫881x，环境变量写的是881x 目前有免拆机开telnet方法，实测880x和881x可行
> 
> 需要先登入后台： 启用telnet： http://<路由器IP地址>/cgi-bin/luci/;stok=<页面地址的STOK>/api/misystem/set_telnet?enable=1 返回：{"code":0}
> 
> 查看开启状态：http://<路由器IP地址>/cgi-bin/luci/;stok=<页面地址的STOK>/api/misystem/get_telnet 返回：{"enable":1,"code":0}
> 
> 查看设备数据：http://<路由器IP地址>/cgi-bin/luci/;stok=<页面地址的STOK>/api/xqsystem/bdata
> 
> telnet启用ttl： nvram set boot_wait=on nvram set uart_en=1 nvram commit
> 
> 判断5000还是5018：ttl在右上角的是5000，在底部的是5018（竖排是m81，横排是m79，板子左上角也有写）在uboot下均显示5018

想问一下5018ttl针脚定义......

mmfmkuang, 2024-03-18:
> cr8816官方历史固件：(IPQ5018) http://cdn.cnbj1.fds.api.mi-img.com/xiaoqiang/rom/cr8816/miwifi_cr8816_firmware_2266a_6.40.44.bin changelog: https://cdn.cnbj1.fds.api.mi-img.com/miwifi/a5215f0e-ab1e-436f-b652-87fbab02a7e7.html hash: 16ed09cc5b1fbf9e4ac207872c32266a
> 
> http://cdn.cnbj1.fds.api.mi-img.com/xiaoqiang/rom/cr8816/miwifi_cr8816_firmware_fe24c_6.40.45.bin hash: 6748a7a9da4866e0f8b4c779eadfe24c

之前解包发现有单独的wolink和cuinform插件，不知道是啥

mmfmkuang, 2024-03-19:
> > 之前解包发现有单独的wolink和cuinform插件，不知道是啥，cuinform看着很眼熟，6.40.45里的api还在，而且也不知道5018实际上能不能刷这个openwrt，实际信号效果好不好......
> 
> 最近大佬活比较忙没空适配5018的包，目前5018的switch需要重新适配，信号排后测试，大概要拖一两星期才能动工了。如果目前要用的话还有个曲线救国的办法，交换机不复位，保留uboot的配置，至少网口能通
> 
> 另外粗略下5018的usb功能，好像是用的uniphy去接的wifi，uniphy是usb和pcie共用的，他还有一组pciex2，usb3引不出，而且飞线难度也大

原来如此，我水平不行还以

> > 想问一下5018ttl针脚定义......
> 
> 理论上两个版本的ttl都是一样的，第三孔应该都是gnd，第一孔是vcc，另外的rx，tx不怕反。第一孔是不是vcc这个等大佬测试了

原来如此，我再试试，之前不懂定义怎么试都看不到跑码，最后应该是不小心接到第一孔vcc把手头其中一台8816烧了……

Ecstasy69, 2024-10-04:
I have a Router CR8808 ver B, running version 7.0.1, I find out if this is the version of 8806, so I can run again version 1.4.01, tks

ken-kuro, 2024-11-28:
> I have a Router CR8808 ver B, running version 7.0.1, I find out if this is the version of 8806, so I can run again version 1.4.01, tks

I got exact same model with 7.0.1 firmware as well, might I asked why you would want to switch back to the original 1.4.01 firmware? Afaik, that firmware may make your router getting remote locked by China Mobile

trongtinh1212, 2025-01-28:
> cr8816官方历史固件：(IPQ5018) http://cdn.cnbj1.fds.api.mi-img.com/xiaoqiang/rom/cr8816/miwifi_cr8816_firmware_2266a_6.40.44.bin changelog: https://cdn.cnbj1.fds.api.mi-img.com/miwifi/a5215f0e-ab1e-436f-b652-87fbab02a7e7.html hash: 16ed09cc5b1fbf9e4ac207872c32266a
> 
> http://cdn.cnbj1.fds.api.mi-img.com/xiaoqiang/rom/cr8816/miwifi_cr8816_firmware_fe24c_6.40.45.bin hash: 6748a7a9da4866e0f8b4c779eadfe24c
> 
> (?)(web 7.0.1)(factory)

Hi, i am using CR8808 China Mobile but actually was CR8816 and default ROM installed is 7.0.1, how i can download newer updates in the future

jeojeo1225, 2026-10-28:
所以？到底什么CPU 什么版本，什么型号，对应什么固件？

jeojeo1225, 2026-10-28:
> 30\. 所以？到底什么CPU 什么版本，什么型号，对应什么固件？
openwrt-ipq50xx-arm-cmcc_rax3000q-squashfs-nand-factory.ubi：这是rax3000q的？
openwrt-ipq50xx-arm-redmi_ax3000-squashfs-nand-factory.ubi：这是ax3000的？ax3000分了联通的CR8806、移动的CR8808、电信的CR8809,M81、M79 A版、M79 B版,其中M79 B版的CPU为IPQ5018，其他两款是IPQ5000
openwrt-ipq50xx-arm-xiaomi_cr881x-squashfs-nand-factory.ubi：这是cr881x的？

fecent, 2026-11-10:
> 所以？到底什么CPU 什么版本，什么型号，对应什么固件？

据我所知，最简单就是看电源接口，a版的貌似都是4mn的dc。b版开始改为5.5的常用dc了，b版刷881x，a版刷ax3000。恢复原厂固件在上面找对应版本刷