hzyitc/openwrt-redmi-ax3000 #2: Compatible with Xiaomi AX3000(ra80)?

Elkowak, 2022-12-06:
Hello, is this openwrt build compatible with Xiaomi ax3000 (ra80)? the hardware is much similar, same soc, ram size, flash size...

Elkowak, 2022-12-08:
i can live without 2.4g, it's already disabled since all my devices there support 5ghz and there is no benefit using 2.4g.

i think i will need to enable ssh, but i didn't find any way to enable.
802.11ax works on this openwrt build? i heard it has problems on mediatek based xiaomi routers running openwrt

hzyitc, 2022-12-13:
I have updated the README. You can try it. This method should work for all xiaomi/redmi router.

Elkowak, 2022-12-31:
Flashed on my Xiaomi ax3000 and booted without issues, wifi 5g works only on 80mhz and ethernet thoughput is very low, i have 1gbps internet and cant get more than 300mbps while using openwrt, probably due to missing offloading options. i reverted to xiaomi firmware and will try again another time

hzyitc, 2022-12-31:
> wifi 5g works only on 80mhz

160mhz should be able to work. You need to set country first. And wait about 3 minutes for radar detection. Then the wifi will appear.

> ethernet thoughput is very low

Yes. CPU ethernet has problem. And i didn't add the nss nat support. As you known this router is more like a AP, so i will make wifi well first and then other problems.

daiaji, 2023-01-03:
现在当AP用基本上没问题？
我现在还在用k2p，令人感慨。

NakanoSanku, 2023-01-09:
> 160mhz should be able to work. You need to set country first. And wait about 3 minutes for radar detection. Then the wifi will appear.
>
>
> Yes. CPU ethernet has problem. And i didn't add the nss nat [...]

I think there's something wrong with your direction,the ax3000 and cr880x hardware are not identical,cr8808 produced in 22 and other cr880x used 2.4g fem of model 7223,ra81 and cr8808 produced in 21 used 2.4g fem from Taiwan.I think this is the core of the problem with 2.4g！

Elkowak, 2023-01-10:
> Flashed on my Xiaomi ax3000 and booted without issues, wifi 5g works only on 80mhz and ethernet thoughput is very low, i have 1gbps internet and cant get more than 300mbps while using openwrt, probably due to missing offloading options. i reverted to xiaomi firmware and will try again another time

i tried built openwrt from qsdk and the results are promissing, network ethernet is good, nss seems to work, no wifi drivers yet,
[image]

i used some of your commits to adapt qsdk buildroot
[openwrt-ipq50xx-ipq50xx_32-redmi_ax3000-squashfs-nand-factory.zip](https://github.com/hzyitc/openwrt-redmi-ax3000/files/10378934/openwrt-ipq50xx-ipq50xx_32-redmi_ax3000-squashfs-nand-factory.zip)

fecent, 2023-01-10:
> 现在当AP用基本上没问题？ 我现在还在用k2p，令人感慨。

其实原厂固件当AP也是没问题的，目前固件开发中不稳定，可以刷着尝鲜

cjz9032, 2023-01-11:
may I know what's the final result of it ? 
Does Xiaomi AX3000 suppoted?

daiaji, 2023-01-11:
AX3000的原厂固件无线连接速率非常低。
只有400-500 MBPS。
这合理吗？

xmaojian, 2023-01-21:
> I'm not sure. You can have a try.
> 
> Probably, you need to change the dts and the BDF file of WiFi.
> 
> BTW, this firmware is still in development. There are lots of problems. For example, due to the small RAM, you probably need to disable 2.4G WiFi to avoid oom.

请问大神，有兴趣适配下中移动RAX3000Q么？硬件配置和红米AX3000一模一样
该型号路由使用目前固件可正常开启SSH，个人是门外汉，只备份了相关分区（一共24个分区，23和24提示系统忙，其余正常备份），如有需要我可提供。

hzyitc, 2023-01-21:
> 有兴趣适配下中移动RAX3000Q么？

请另开一个issue。提供下启动日志和分区表(`/proc/mtd`)。

以及确保**你自己能使用TTL在U-boot下救砖**。

vamanea, 2023-02-18:
For others searching this firmware boots ok on AX3000 RA82 which I think is international version. UART method seems to work, I also used the netmode method(which only works for RA82, from [here](https://forum.openwrt.org/t/adding-openwrt-support-for-xiaomi-redmi-router-ax6s-xiaomi-router-ax3200/111085/936?u=alexq)) with success.

If you wish to add ax3000 RA82 to your supported list, the link to Xiaomi original firmware is this: https://cdn.awsusor1.fds.api.mi-img.com/xiaoqiang/rom/ra82/miwifi_ra82_all_7eb36_1.3.27_INT.bin (taken from [here]( https://forum.openwrt.org/t/add-openwrt-for-xiaomi-ax3000-2021/104751/30) )There is a issue with secure boot in uboot, so I am booting manually at the momement, but all in all it's very promissing.

DoveKi, 2023-03-26:
> i tried built openwrt from qsdk and the results are promissing, network ethernet is good, nss seems to work, no wifi drivers yet, [image]
>
> i used some of your commits to adapt qsdk buildroot [...]

Do you test qsdk open ath11k drivers?
https://insidelinuxdev.net/article/a0dovj.html

dandek01, 2024-05-17:
Hi,
I successfully managed to convert AX3000 RA80 to openwrt using this firmware.
Brief instructions are available on this [link](https://forum.openwrt.org/t/adding-openwrt-support-for-xiaomi-redmi-router-ax6s-xiaomi-router-ax3200/111085/1875?u=dandek01)

eeddww, 2024-07-27:
> Hi, I successfully managed to convert AX3000 RA80 to openwrt using this firmware. Brief instructions are available on this [link](https://forum.openwrt.org/t/adding-openwrt-support-for-xiaomi-redmi-router-ax6s-xiaomi-router-ax3200/111085/1875?u=dandek01)

Hi, how does it perform? Is it good? Did you fix the 5ghz? Thinking of using this firmware on both my ra80 and ra81

htbuddha, 2024-10-29:
> Hi, I successfully managed to convert AX3000 RA80 to openwrt using this firmware. Brief instructions are available on this [link](https://forum.openwrt.org/t/adding-openwrt-support-for-xiaomi-redmi-router-ax6s-xiaomi-router-ax3200/111085/1875?u=dandek01)

How does it perform wirelessly?