hzyitc/openwrt-redmi-ax3000 #3: Can't interrupt the U-boot

ohn0n0n0, 2022-12-23:
i'm stuck on step 7. Send any key stroke to UART to interrupt the U-boot.

it shows bootwait is on, bootdelay=2. but i can't send anything to uart.
[image]

hzyitc, 2022-12-26:
Please make sure you connect to RX instand of VCC. Both of them are 1.8V.

ohn0n0n0, 2022-12-27:
[image]
Is this right?

hzyitc, 2022-12-29:
Corrent for CR8806. For AX3000, I'm not sure. Because I don't have one.

ohn0n0n0, 2022-12-29:
> Corrent for CR8806. For AX3000, I'm not sure. Because I don't have one.

somehow i still can't interrupt the U-boot.
i have cr8806 too. what type of usb to ttl are you using? i'm using cp2102.

hzyitc, 2022-12-29:
FT4232. Really expensive.

But i also try CH340. It works.

hzyitc, 2022-12-29:
You can try to disconnect the RX. And use something to short RX and GND to interrupt

ohn0n0n0, 2022-12-30:
> FT4232. Really expensive.
> 
> But i also try CH340. It works.

I bought a ch340g last night and it works. You just saved me a lot of time.

closed 2022-12-30 by hzyitc

hzyitc, 2022-12-30:
When I use STC microcontroller, CP2102 sometime fail to burn. CH340 always work well. Don't known why. But I never use it anymore.

cyy5358, 2023-01-29:
> [image] Is this right?

Hi, may I ask how do you connect pins to the UART hole? Using soldering or articulated arm or something ... ?
I hold pins swingingly and connection isn't steady.

ohn0n0n0, 2023-01-29:
> > [image] Is this right?
> 
> Hi, may I ask how do you connect pins to the UART hole? Using soldering or articulated arm or something ... ? I hold pins swingingly and connection isn't steady.
> 
> @hzyitc @ohn0n0n0
[image]

no need to soldering or articulated. 
[image]

cyy5358, 2023-01-30:
> swingingly

I see. cool

cyy5358, 2023-01-30:
> > > [image] Is this right?
> > 
> > 
> > Hi, may I ask how do you connect pins to the UART hole? Using soldering or articulated arm or something ... ? I hold pins swingingly and connection isn't steady.
> > @hzyitc @ohn0n0n0
> > [image]
> 
> no need to soldering or articulated. [image]

I see. cool!