hzyitc/openwrt-redmi-ax3000 #4: cr8808 fireware problem

NakanoSanku, 2023-01-08:
cr8808 
produced in 2021 use 6.2.12
produced in 2022 use 6.2.18
[image]
Because the amplifier they use has been changed
I tried to grab bags with violence，Unfortunately, its naming format seems very unique，Ihope to give you some help

NakanoSanku, 2023-01-09:
- [x] 6.2.18~cr8806
- [x] 6.2.12~ax3000

hzyitc, 2023-01-09:
There is no firmware for CR8808 in Xiaomi CDN. So what we can do is just to wait for the first upgrade.

NakanoSanku, 2023-01-09:
> There is no firmware for CR8808 in Xiaomi CDN. So what we can do is just to wait for the first upgrade.

No！I seem already solve this problem，use [miwifi flash tool](https://bigota.miwifi.com/xiaoqiang/tools/MIWIFIRepairTool.x86.zip) flash cr8806 fireware ！but must sure your cr8808 fireware version is 6.2.18 ，if not you need flash ax3000 fireware[image]

closed 2023-01-10 by NakanoSanku

hzyitc, 2023-01-10:
I check the uboot of CR8808 again. And yes, it use the same platform magic as CR8806. I will update the README. Thank you.

HapiLemon, 2023-01-11:
@hzyitc 
@NakanoSanku 
BTW, I have tried flashing firmware of CR8809 to my CR8808 and it works well.