hzyitc/openwrt-redmi-ax3000 #5: Extend rootfs?

NakanoSanku, 2023-01-14:
Maybe we can try to change the mibib file and merge overlay and rootfs_1 into a larger partition to break the firmware size that can be flash , like ax6:[reference link](https://b23.tv/mcsxu8d)

hzyitc, 2023-01-15:
It should be able to do it. What's more, `overlay` and `data` can also be merged into rootfs.

**But I haven't tried.**

Here is the definition of partition table:

https://git.codelinaro.org/clo/qsdk/oss/boot/u-boot-2016/-/blob/caf_migration/11.4.0.5/arch/arm/cpu/armv7/qca/common/smem.c#L116-L138

Elkowak, 2023-01-17:
doesn't need to touch mibib partition table, this can be done directly on nand dts mapping like on robimarko ax6 build

i already done some changes to dts based on kernel 5.14, with this code, kernel will ignore default partition table and use this partition table map from dts. idk why xiaomi did use rootfs_1 as first partition so the name on dts is rootfs_1
**this is experimental**

```
		nand: qpic-nand@79b0000 {
			compatible = "qcom,ebi2-nandc-bam-v2.1.1";
			reg = <0x79b0000 0x10000>;
			#address-cells = <1>;
			#size-cells = <0>;
			clocks = <&gcc GCC_QPIC_CLK>,
				<&gcc GCC_QPIC_AHB_CLK>,
				<&gcc GCC_QPIC_IO_MACRO_CLK>;
			clock-names = "core", "aon", "io_macro";

			dmas = <&qpic_bam 0>,
				<&qpic_bam 1>,
				<&qpic_bam 2>,
				<&qpic_bam 3>;
			dma-names = "tx", "rx", "cmd", "sts";
			pinctrl-0 = <&qspi_nand_pins>;
			pinctrl-names = "default";
			status = "ok";

			partitions {
			status = "disabled";
			};

			nandcs@0 {
				compatible = "qcom,nandcs";
				reg = <0>;
				#address-cells = <1>;
				#size-cells = <1>;

				nand-ecc-strength = <4>;
				nand-ecc-step-size = <512>;
				nand-bus-width = <8>;
					partitions {
					compatible = "fixed-partitions";
					#address-cells = <1>;
					#size-cells = <1>;

						partition@0 {
							label = "0:sbl1";
							reg = <0x0 0x80000>;
							read-only;
						};

						partition@80000 {
							label = "0:mibib";
							reg = <0x80000 0x80000>;
							read-only;
						};

						partition@100000 {
							label = "0:BOOTCONFIG";
							reg = <0x100000 0x40000>;
							read-only;
						};

						partition@140000 {
							label = "0:BOOTCONFIG1";
							reg = <0x140000 0x40000>;
							read-only;
						};

						partition@180000 {
							label = "0:QSEE";
							reg = <0x180000 0x100000>;
							read-only;
						};

						partition@280000 {
							label = "0:QSEE_1";
							reg = <0x280000 0x100000>;
							read-only;
						};

						partition@380000 {
							label = "0:DEVCFG";
							reg = <0x380000 0x40000>;
							read-only;
						};

						partition@3c0000 {
							label = "0:DEVCFG_1";
							reg = <0x3c0000 0x40000>;
							read-only;
						};

						partition@400000 {
							label = "0:CDT";
							reg = <0x400000 0x40000>;
							read-only;
						};

						partition@440000 {
							label = "0:CDT_1";
							reg = <0x440000 0x40000>;
							read-only;
						};

						partition@480000 {
							label = "0:APPSBLENV";
							reg = <0x480000 0x20000>;
						};

						partition@500000 {
							label = "0:APPSBL";
							reg = <0x500000 0x140000>;
							read-only;
						};

						partition@640000 {
							label = "0:APPSBL_1";
							reg = <0x640000 0x140000>;
							read-only;
						};

						partition@780000 {
							label = "0:art";
							reg = <0x800000 0x100000>;
							read-only;

							compatible = "nvmem-cells";
							#address-cells = <1>;
							#size-cells = <1>;

							macaddr_dp1: macaddr@0 {
								reg = <0x0 0x6>;
							};

							macaddr_dp2: macaddr@6 {
								reg = <0x6 0x6>;
							};

							caldata_qca5018: caldata@10000 {
								reg = <0x1000 0x20000>;
							};
							caldata_qcn6122: caldata_1@2680 {
								reg = <0x26800 0x20000>;
							};
						};

						partition@880000 {
							label = "0:TRAINING";
							reg = <0x880000 0x80000>;
						};

						partition@900000 {
							label = "bdata";
							reg = <0x880000 0x80000>;
						};

						partition@980000 {
							/* This is crash + crash_syslog parts combined */
							label = "pstore";
							reg = <0x900000 0x100000>;
						};

						rootfs_1: partition@A80000 {
							label = "rootfs_1";
							reg = <0xA80000 0x4800000>;
						};
					};
				};
			};
```
```
root@OpenWrt:/# cat /proc/mtd
dev:    size   erasesize  name
mtd0: 00080000 00020000 "0:sbl1"
mtd1: 00080000 00020000 "0:mibib"
mtd2: 00040000 00020000 "0:BOOTCONFIG"
mtd3: 00040000 00020000 "0:BOOTCONFIG1"
mtd4: 00100000 00020000 "0:QSEE"
mtd5: 00100000 00020000 "0:QSEE_1"
mtd6: 00040000 00020000 "0:DEVCFG"
mtd7: 00040000 00020000 "0:DEVCFG_1"
mtd8: 00040000 00020000 "0:CDT"
mtd9: 00040000 00020000 "0:CDT_1"
mtd10: 00020000 00020000 "0:APPSBLENV"
mtd11: 00140000 00020000 "0:APPSBL"
mtd12: 00140000 00020000 "0:APPSBL_1"
mtd13: 00100000 00020000 "0:art"
mtd14: 00000006 00020000 "macaddr"
mtd15: 00000006 00020000 "macaddr"
mtd16: 00020000 00020000 "caldata"
mtd17: 00020000 00020000 "caldata_1"
mtd18: 00080000 00020000 "0:TRAINING"
mtd19: 00080000 00020000 "bdata"
mtd20: 00100000 00020000 "pstore"
mtd21: 04800000 00020000 "rootfs_1"
mtd22: 003a2000 0001f000 "kernel"
mtd23: 0101b000 0001f000 "ubi_rootfs"
mtd24: 00c3b000 0001f000 "rootfs_data"
```
Still need more work to expand overlay/rootfs_data to use all space left and check if it is really stable.
```
root@OpenWrt:/# df -h
Filesystem                Size      Used Available Use% Mounted on
/dev/mtdblock23          16.0M     16.0M         0 100% /rom
tmpfs                    88.6M    364.0K     88.2M   0% /tmp
/dev/ubi0_2               9.5M    124.0K      8.9M   1% /overlay
overlayfs:/overlay        9.5M    124.0K      8.9M   1% /
tmpfs                   512.0K         0    512.0K   0% /dev
root@OpenWrt:/# df
Filesystem           1K-blocks      Used Available Use% Mounted on
/dev/mtdblock23          16384     16384         0 100% /rom
tmpfs                    90676       364     90312   0% /tmp
/dev/ubi0_2               9748       124      9092   1% /overlay
overlayfs:/overlay        9748       124      9092   1% /
tmpfs                      512         0       512   0% /dev
root@OpenWrt:/#
```
and need to manually erase boot partitions with this:

```
flash rootfs_1 0 0
flash rootfs 0 0
```

hzyitc, 2023-01-17:
> idk why xiaomi did use rootfs_1 as first partition so the name on dts is rootfs_1

Maybe to keep dual boot.

hzyitc, 2023-01-17:
But due to the small memory, so it make little sense.

closed 2023-03-27 by NakanoSanku