hzyitc/openwrt-redmi-ax3000 #6: 802.11k and 802.11v issues.

ohn0n0n0, 2023-01-29:
i try to enable 802.11k and 802.11v for band-steering. >https://openwrt.org/docs/guide-user/network/wifi/usteer.

i install wpad full ([wpad-wolfssl](https://openwrt.org/packages/pkgdata/wpad-wolfssl) or [wpad-openssl](https://openwrt.org/packages/pkgdata/wpad-openssl))

when i add there line into /etc/config/wireless. the radio down.

option bss_transition '1'
option wnm_sleep_mode '1'
option time_advertisement '2'
option time_zone 'GMT0'
option ieee80211k '1'
option rrm_neighbor_report '1'
option rrm_beacon_report '1'

[image]

[image]

system log:

Sun Jan 29 14:38:56 2023 daemon.notice netifd: radio0 (5180): Command failed: Request timed out
Sun Jan 29 14:38:56 2023 daemon.notice netifd: radio0 (5180): Command failed: Not found
Sun Jan 29 14:38:56 2023 daemon.notice netifd: radio0 (5180): Device setup failed: HOSTAPD_START_FAILED
Sun Jan 29 14:38:56 2023 daemon.notice netifd: Wireless device 'radio0' set retry=0
Sun Jan 29 14:38:56 2023 daemon.crit netifd: Wireless device 'radio0' setup failed, retry=0
Sun Jan 29 14:38:56 2023 daemon.notice netifd: Wireless device 'radio0' is now down

ohn0n0n0, 2023-01-31:
i flash old version and 802.11k and 802.11v work without issues
[openwrt-ipq50xx-generic-redmi_ax3000-squashfs-nand-factory.zip](https://github.com/hzyitc/openwrt-redmi-ax3000/files/10549074/openwrt-ipq50xx-generic-redmi_ax3000-squashfs-nand-factory.zip)

hzyitc, 2023-05-26:
It seem like that this issue don't exist now?

ohn0n0n0, 2023-05-28:
> It seem like that this issue don't exist now?

with lastest version i thinks it working now.

closed 2023-05-30 by ohn0n0n0