hzyitc/openwrt-redmi-ax3000 #8: Missing some kmod files when compiling the firmware

shinwa, 2023-02-21:
Followed the instruction, when do 'make download -j1', show lots of error messages related to qca-nss module.
`WARNING: Makefile 'package/kernel/qca/qca-nss-clients/Makefile' has a dependency on 'kmod-qca-nss-cfi-cryptoapi', which does not exist
WARNING: Makefile 'package/kernel/qca/qca-nss-clients/Makefile' has a dependency on 'kmod-qca-nss-cfi-cryptoapi', which does not exist
WARNING: Makefile 'package/kernel/qca/qca-nss-clients/Makefile' has a dependency on 'kmod-qca-nss-cfi-ocf', which does not exist
WARNING: Makefile 'package/kernel/qca/qca-nss-clients/Makefile' has a dependency on 'kmod-qca-nss-cfi-ocf', which does not exist
WARNING: Makefile 'package/kernel/qca/qca-nss-clients/Makefile' has a dependency on 'kmod-qca-ovsmgr', which does not exist
WARNING: Makefile 'package/kernel/qca/qca-nss-clients/Makefile' has a dependency on 'kmod-qca-nss-cfi-cryptoapi', which does not exist
WARNING: Makefile 'package/kernel/qca/qca-nss-clients/Makefile' has a dependency on 'kmod-qca-nss-cfi-cryptoapi', which does not exist
WARNING: Makefile 'package/kernel/qca/qca-nss-clients/Makefile' has a dependency on 'kmod-qca-wifi', which does not exist
WARNING: Makefile 'package/kernel/qca/qca-nss-clients/Makefile' has a build dependency on 'kmod-qca-nss-crypto', which does not exist`

[image]

hzyitc, 2023-02-21:
Just warnings. Not need to care about them.

I just import some necessary modules from qsdk.

PhoenixTrees, 2023-03-07:
> 
I want to install open clash but missing kmod-ipt-ipset

closed 2023-05-13 by hzyitc