hzyitc/openwrt-redmi-ax3000 #12: mac80211 build failed

zhoufuli, 2023-04-17:
make[2] -C package/utils/mtd-utils download
    ERROR: package/kernel/mac80211 failed to build (build variant: regular).
make package/download: build failed. Please re-run make with -j1 V=s or V=sc for a higher verbosity level to see what's going on

Excuse me, does this affect the final compiled data?

hzyitc, 2023-04-18:
Yes, ofc. Please make sure you have followed the instrations and not select any other `Wireless Drivers` because it's hacked.

zhoufuli, 2023-04-18:
OK, thanks

zhoufuli, 2023-04-18:
You have unfulfilled binary requirements
make[3]: *** [Makefile:634: /workdir/openwrt/dl/backports-20210222-4.4-qsdk-11.4.0.5.tar.xz] Error 1
make[3]: Leaving directory '/workdir/openwrt/package/kernel/mac80211'
time: package/kernel/mac80211/regular/compile#10.04#4.39#11.95
    ERROR: package/kernel/mac80211 failed to build (build variant: regular).
make[2]: *** [package/Makefile:116: package/kernel/mac80211/compile] Error 1
make[2]: Leaving directory '/workdir/openwrt'
make[1]: *** [package/Makefile:110: /workdir/openwrt/staging_dir/target-arm_cortex-a7_musl_eabi/stamp/.package_compile] Error 2
make[1]: Leaving directory '/workdir/openwrt'
make: *** [/workdir/openwrt/include/toplevel.mk:230: world] Error 2

Compilation still reports an error, what may be the reason this time?

zhoufuli, 2023-04-20:
The problem has been solved

closed 2023-04-20 by zhoufuli