hzyitc/openwrt-redmi-ax3000 #13: Firmware loading failed and CPU interrupt was too high

zhoufuli, 2023-04-18:
Hello, I found that after brushing, dmesg prompts the following message:

[   20.755383] Subsystem error monitoring/handling services are up
[   20.880266] ath11k c000000.wifi: Direct firmware load for ath11k/IPQ5018/hw1.0/qdss_trace_config.bin failed with error -2
[   20.945644] Subsystem error monitoring/handling services are up
[   21.222900] ath11k c000000.wifi: Direct firmware load for qdss_trace_config.bin failed with error -2
[   42.948063] Subsystem error monitoring/handling services are up
[   43.258920] ath11k soc:wifi1@c000000: Direct firmware load for ath11k/qcn6122/hw1.0/qdss_trace_config.bin failed with error -2
[   43.604336] ath11k soc:wifi1@c000000: Direct firmware load for qdss_trace_config.bin failed with error -2

During the test, it was found that a large number of calls were CPU interrupts

 89:    1569863          0     GIC-0 431 Edge      nss_queue0
 90:          0     227402     GIC-0 430 Edge      nss_queue1

Is this forwarding by NPU?

hzyitc, 2023-04-19:
> Hello, I found that after brushing, dmesg prompts the following message:
>
> [ 20.755383] Subsystem error monitoring/handling services are up
> [ 20.880266] ath11k c000000.wifi: Direct firmware load for [...]

No need to care about them.

> During the test, it was found that a large number of calls were CPU interrupts
> 
> 89: 1569863 0 GIC-0 431 Edge nss_queue0
> 90: 0 227402 GIC-0 430 Edge nss_queue1
> 
> Is this forwarding by NPU?

Possibly, no. [Branch ipq50xx-qsdk-kernel-5.4-openwrt-21.02-qsdk-11.5.05.841.1029](https://github.com/hzyitc/openwrt-redmi-ax3000/tree/ipq50xx-qsdk-kernel-5.4-openwrt-21.02-qsdk-11.5.05.841.1029) don't support `NPU`. If you need it, please use [branch ipq50xx-qsdk-kernel-5.4-openwrt-21.02-qsdk-11.5.05.841.1029](https://github.com/hzyitc/openwrt-redmi-ax3000/tree/ipq50xx-qsdk-kernel-5.4-openwrt-21.02-qsdk-11.5.05.841.1029)

zhoufuli, 2023-04-19:
OK, thanks

closed 2023-05-13 by hzyitc