hzyitc/openwrt-redmi-ax3000 #14: 重启后不同vlan被dhcp分配ip问题？

2955748358, 2023-04-18:
我为eth0划分了两个vlan，一个作为lan一个用来看iptv。我的iptv是使用dhcp方式获取的ip，lan使用的openwrt自带的dhcp分配ip。但是当我重启之后lan下的有线设备会获取到iptv的ip，需要手动重启接口才能正常获取ip。

hzyitc, 2023-04-21:
怎么接线的?

u-boot开机时会划分默认vlan。就是wan一个，lan一个。启动后才能应用系统的vlan。

2955748358, 2023-04-21:
我的VLAN配置如下
[image]
[image]

我划分了一个vlan3配合udpxy看iptv 我的iptv使用的dhcp
当路由器重启之后 两个vlan1下的有线设备被iptv的dhcp分配了ip
[image]
[image]
每次需要重启接口后才获取正常的由路由DHCP分配的IP网段192.168.0.x
目前我把所有有线设备配置静态ip来解决的此问题

hzyitc, 2023-04-21:
两个解决方案：
- 重新编译`u-boot`。`u-boot`期间，禁止交换机端口间通讯。
- 系统启动后重启所有。即让所有端口下线，让对端识别到断开，后重上线。

暂时不确定别的target会不会在启动后重启所有端口。

closed 2023-05-25 by 2955748358