hzyitc/openwrt-redmi-ax3000 #15: Zerorier build failed

zhoufuli, 2023-04-20:
In file included from service/OneService.cpp:96:0:
ext/inja/inja.hpp:63:23: fatal error: string_view: No such file or directory
compilation terminated.
make[4]: *** [<builtin>: service/OneService.o] Error 1
make[3]: *** [Makefile:84: /workdir/openwrt/build_dir/target-arm_cortex-a7_musl_eabi/ZeroTierOne-1.10.2/.built] Error 2
make[4]: Leaving directory '/workdir/openwrt/build_dir/target-arm_cortex-a7_musl_eabi/ZeroTierOne-1.10.2'
make[3]: Leaving directory '/workdir/openwrt/feeds/packages/net/zerotier'
time: package/feeds/packages/zerotier/compile#0.33#0.07#0.37
    ERROR: package/feeds/packages/zerotier failed to build.
make[2]: *** [package/Makefile:116: package/feeds/packages/zerotier/compile] Error 1
make[2]: Leaving directory 

Hello, I encountered the above problem in my compilation. Can you help me see why? Thank you.

hzyitc, 2023-04-21:
Which branch ?

I build successful.

zhoufuli, 2023-04-21:
It has been successful. Thank you. Can you add shortcut-fe-cm?

hzyitc, 2023-05-13:
> Can you add shortcut-fe-cm?

TODO. but very low priority

closed 2023-05-13 by hzyitc