hzyitc/openwrt-redmi-ax3000 #16: 编译的时候遇到点问题，求大佬解答下，谢谢

Nicolas-Gong, 2023-04-24:
在运行 ```./scripts/feeds install -a``` 时警告
```
WARNING: Makefile 'package/kernel/qca/qca-nss-clients/Makefile' has a dependency on 'kmod-qca-nss-cfi-cryptoapi', which does not exist
WARNING: Makefile 'package/kernel/qca/qca-nss-clients/Makefile' has a dependency on 'kmod-qca-nss-cfi-cryptoapi', which does not exist
WARNING: Makefile 'package/kernel/qca/qca-nss-clients/Makefile' has a dependency on 'kmod-qca-nss-cfi-ocf', which does not exist
WARNING: Makefile 'package/kernel/qca/qca-nss-clients/Makefile' has a dependency on 'kmod-qca-nss-ecm', which does not exist
WARNING: Makefile 'package/kernel/qca/qca-nss-clients/Makefile' has a dependency on 'kmod-qca-nss-cfi-ocf', which does not exist
WARNING: Makefile 'package/kernel/qca/qca-nss-clients/Makefile' has a dependency on 'kmod-qca-nss-ecm', which does not exist
WARNING: Makefile 'package/kernel/qca/qca-nss-clients/Makefile' has a dependency on 'kmod-qca-ovsmgr', which does not exist
WARNING: Makefile 'package/kernel/qca/qca-nss-clients/Makefile' has a dependency on 'kmod-qca-nss-cfi-cryptoapi', which does not exist
WARNING: Makefile 'package/kernel/qca/qca-nss-clients/Makefile' has a dependency on 'kmod-qca-nss-ecm-premium', which does not exist
WARNING: Makefile 'package/kernel/qca/qca-nss-clients/Makefile' has a dependency on 'kmod-qca-nss-cfi-cryptoapi', which does not exist
WARNING: Makefile 'package/kernel/qca/qca-nss-clients/Makefile' has a dependency on 'kmod-qca-nss-ecm-premium', which does not exist
WARNING: Makefile 'package/kernel/qca/qca-nss-clients/Makefile' has a dependency on 'kmod-qca-wifi', which does not exist
WARNING: Makefile 'package/kernel/qca/qca-nss-clients/Makefile' has a build dependency on 'kmod-qca-nss-crypto', which does not exist
WARNING: Makefile 'package/kernel/qca/qca-nss-drv/Makefile' has a dependency on 'kmod-qca-nss-gmac', which does not exist
WARNING: Makefile 'package/kernel/qca/qca-ssdk/Makefile' has a dependency on 'kmod-qca-rfs', which does not exist
```
这个有啥影响吗？

另外在编译时候选择openwrt 配置时架构是用IPQ50xx吧？我选择完后直接选择下方的保存，执行下一步了，然后报错了
```
make[2] -C package/system/uci download
 make[2] -C package/system/urandom-seed download
 make[2] -C package/system/urngd download
 make[2] -C package/system/usign download
 make[2] -C package/utils/jsonfilter download
 make[2] -C package/utils/busybox download
 make[2] -C package/utils/mtd-utils download
    ERROR: package/kernel/mac80211 failed to build (build variant: regular).
make package/download: build failed. Please re-run make with -j1 V=s or V=sc for a higher verbosity level to see what's going on
 make[1] target/download
 make[2] -C target/linux download
```
新人小白第一次尝试自己编译，不是太明白。
编译环境: deepin20
编译需要的都安装了，
```
sudo apt install build-essential clang flex g++ gawk gcc-multilib gettext \
  git libncurses5-dev libssl-dev python3-distutils rsync unzip zlib1g-dev \
  coccinelle
```
不过不是走的 `ppa:npalix/coccinelle` 源，用的是deepin默认的源。

hzyitc, 2023-05-01:
大概率是`coccinelle`版本问题，好像新版的会运行失败。

feed的警告不用理会，没有移植所有的包。

3wking, 2023-05-10:
我也遇到了，如何解决这个问题啊？？？
    ERROR: package/kernel/mac80211 failed to build (build variant: regular).
make -r world: build failed. Please re-run make with -j1 V=s or V=sc for a higher verbosity level to see what's going on

3wking, 2023-05-10:
我用的是 ppa:npalix/coccinelle 源
coccinelle版本1.0.8

Tongrens, 2023-05-13:
同问，解决了踢我以下

hzyitc, 2023-05-13:
`make V=s`看看在哪出错。

或者可以直接下载Release里面的`dl.tar`，里面的`backports-20210222-5.4.225.tar.xz `就是生成后的文件。

hy12297889, 2023-05-16:
> `make V=s`看看在哪出错。
> 
> 或者可以直接下载Release里面的`dl.tar`，里面的`backports-20210222-5.4.225.tar.xz `就是生成后的文件。

同样的报错信息
SHELL= flock /data/cr8808/tmp/.backports-20210222-5.4-qsdk-11.5.0.5.tar.xz.flock -c '  [ -f "/data/cr8808/dl/backports-20210222-5.4-qsdk-11.5.0.5.tar.xz" ] || (         mkdir -p /data/cr8808/tmp/dl && cd /data/cr8808/tmp/dl && rm -rf backports-20210222-5.4-qsdk-11.5.0.5 backports-source ath-kernel && mkdir -p backports-20210222-5.4-qsdk-11.5.0.5 backports-source ath-kernel && tar -C backports-source --strip-components=1 -xJf /data/cr8808/dl/backports-source-v5.12.19-1.tar.xz && tar -C ath-kernel --strip-components=1 -xJf /data/cr8808/dl/ath-kernel-20210222.tar.xz && ( cd /data/cr8808/tmp/dl/backports-source && PATH="/sbin:/data/cr8808/staging_dir/toolchain-arm_cortex-a7_gcc-8.4.0_musl_eabi/bin:/data/cr8808/staging_dir/toolchain-arm_cortex-a7_gcc-8.4.0_musl_eabi/bin:/data/cr8808/staging_dir/host/bin:/data/cr8808/staging_dir/host/bin:/opt/jdk1.8.0_181/bin:/opt/jdk1.8.0_181/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin" ./gentree.py --clean /data/cr8808/tmp/dl/ath-kernel /data/cr8808/tmp/dl/backports-20210222-5.4-qsdk-11.5.0.5 ) && (cd /data/cr8808/tmp/dl;  tar --numeric-owner --owner=0 --group=0 --mode=a-s --sort=name ${TAR_TIMESTAMP:+--mtime="$TAR_TIMESTAMP"} -c backports-20210222-5.4-qsdk-11.5.0.5 |      xz -T0 -zc -7e > /data/cr8808/tmp/dl/backports-20210222-5.4-qsdk-11.5.0.5.tar.xz) && mv /data/cr8808/tmp/dl/backports-20210222-5.4-qsdk-11.5.0.5.tar.xz /data/cr8808/dl && rm -rf backports-20210222-5.4-qsdk-11.5.0.5 backports-source ath-kernel )'
Copy original source files ...
fatal: not a git repository (or any parent up to mount point /)
Stopping at filesystem boundary (GIT_DISCOVERY_ACROSS_FILESYSTEM not set).
Traceback (most recent call last):
  File "./gentree.py", line 1135, in <module>
    ret = _main()
  File "./gentree.py", line 721, in _main
    retv = process(args.kerneldir, args.copy_list,
  File "./gentree.py", line 893, in process
    backports_version = git.describe(tree=source_dir, extra_args=['--long'])
  File "/data/cr8808/tmp/dl/backports-source/lib/bpgit.py", line 93, in describe
    _check(process)
  File "/data/cr8808/tmp/dl/backports-source/lib/bpgit.py", line 13, in _check
    raise ExecutionError(process.returncode)
lib.bpgit.ExecutionError: 128
make[2]: *** [Makefile:652: /data/cr8808/dl/backports-20210222-5.4-qsdk-11.5.0.5.tar.xz] Error 1
make[2]: Leaving directory '/data/cr8808/package/kernel/mac80211'
time: package/kernel/mac80211/regular/download#1.13#10.08#8.08
    ERROR: package/kernel/mac80211 failed to build (build variant: regular).
make[1]: *** [package/Makefile:116: package/kernel/mac80211/download] Error 1
make[1]: Leaving directory '/data/cr8808'
make[1]: Entering directory '/data/cr8808'
make[2]: Entering directory '/data/cr8808/target/linux'
make[3]: Entering directory '/data/cr8808/target/linux/ipq50xx'
make[3]: Nothing to be done for 'download'.
make[3]: Leaving directory '/data/cr8808/target/linux/ipq50xx'
make[2]: Leaving directory '/data/cr8808/target/linux'
time: target/linux/download#0.03#0.00#0.03
make[1]: Leaving directory '/data/cr8808'

有没有可能是少了#src-link custom /usr/src/openwrt/custom-feed ?
[image]

IITII, 2023-05-17:
这个我遇到过很多次 `ERROR: package/kernel/mac80211 failed to build (build variant: regular).` 反正最后是 make -j1 V=s 的时候自动 clone 了一个 2.5G 的仓库, 后面就编译成功了. 可以试试上面说的替换 `backports-20210222-5.4.225.tar.xz`

* ~~按 CI 来的, 然后 mac80211 一直编译不过, 很烦~~

hy12297889, 2023-05-17:
> 这个我遇到过很多次 `ERROR: package/kernel/mac80211 failed to build (build variant: regular).` 反正最后是 make -j1 V=s 的时候自动 clone 了一个 2.5G 的仓库, 后面就编译成功了. 可以试试上面说的替换 `backports-20210222-5.4.225.tar.xz`
> 
> * ~按 CI 来的, 然后 mac80211 一直编译不过, 很烦~

把`backports-20210222-5.4.225.tar.xz` 文件放到dl目录成功编译,应该就是少了这个 `2.5G的仓库`

hzyitc, 2023-05-17:
> fatal: not a git repository (or any parent up to mount point /)
Stopping at filesystem boundary (GIT_DISCOVERY_ACROSS_FILESYSTEM not set).

这错误确实没见过。不确定是啥问题。

可能是下载的ath-kernel有问题。把dl里面的ath-kernel和backports-source删了重下载试试

3wking, 2023-05-17:
HEAD is now at 3a9d70a3 backports: Remove crypto/akcipher.h
Packing checkout...
mkdir -p /openwrt-redmi-ax3000/dl), 45.75 MiB | 63.00 KiB/s
SHELL= flock /openwrt-redmi-ax3000/tmp/.ath-kernel-20210222.tar.xz.flock -c '  [ -f "/openwrt-redmi-ax3000/dl/ath-kernel-20210222.tar.xz" ] || (          ( echo "Downloading source code from codelinaro..."; mkdir -p /openwrt-redmi-ax3000/tmp/dl && cd /openwrt-redmi-ax3000/tmp/dl && rm -rf ath-kernel-20210222 && mkdir ath-kernel-20210222 && curl "https://git.codelinaro.org/clo/qsdk/kvalo/ath/-/archive/b157d227636eb4fd9854218c4802e8dab2eadf1b/archive.tar.gz" | tar -C ath-kernel-20210222 --strip-components 1 -xzf - && echo "Repacking..." && export TAR_TIMESTAMP="" &&  tar --numeric-owner --owner=0 --group=0 --mode=a-s --sort=name ${TAR_TIMESTAMP:+--mtime="$TAR_TIMESTAMP"} -c ath-kernel-20210222 |       xz -T0 -zc -7e > /openwrt-redmi-ax3000/tmp/dl/ath-kernel-20210222.tar.xz && mv /openwrt-redmi-ax3000/tmp/dl/ath-kernel-20210222.tar.xz /openwrt-redmi-ax3000/dl/ && rm -rf ath-kernel-20210222; ) || (   echo "Checking out files from the git repository..."; mkdir -p /openwrt-redmi-ax3000/tmp/dl && cd /openwrt-redmi-ax3000/tmp/dl && rm -rf ath-kernel-20210222 && [ \! -d ath-kernel-20210222 ] && git clone  https://git.codelinaro.org/clo/qsdk/kvalo/ath.git ath-kernel-20210222 && (cd ath-kernel-20210222 && git checkout b157d227636eb4fd9854218c4802e8dab2eadf1b && git submodule update --init --recursive) && echo "Packing checkout..." && export TAR_TIMESTAMP=`cd ath-kernel-20210222 && git log -1 --format='\''@%ct'\''` && rm -rf ath-kernel-20210222/.git &&  tar --numeric-owner --owner=0 --group=0 --mode=a-s --sort=name ${TAR_TIMESTAMP:+--mtime="$TAR_TIMESTAMP"} -c ath-kernel-20210222 |       xz -T0 -zc -7e > /openwrt-redmi-ax3000/tmp/dl/ath-kernel-20210222.tar.xz && mv /openwrt-redmi-ax3000/tmp/dl/ath-kernel-20210222.tar.xz /openwrt-redmi-ax3000/dl/ && rm -rf ath-kernel-20210222; );      )'
Downloading source code from codelinaro...
/bin/sh: 1: curl: not found

gzip: stdin: unexpected end of file
tar: Child returned status 1
tar: Error is not recoverable: exiting now
Checking out files from the git repository...
Cloning into 'ath-kernel-20210222'...
remote: Enumerating objects: 9518649, done.
remote: Counting objects: 100% (100950/100950), done.
remote: Compressing objects: 100% (36678/36678), done.
error: RPC failed; curl 56 GnuTLS recv error (-9): Error decoding the received TLS packet.
fatal: the remote end hung up unexpectedly
fatal: early EOF
fatal: index-pack failed
make[2]: *** [Makefile:520: /openwrt-redmi-ax3000/dl/ath-kernel-20210222.tar.xz] Error 128
make[2]: Leaving directory '/openwrt-redmi-ax3000/package/kernel/mac80211'
time: package/kernel/mac80211/regular/download#2.30#0.79#2460.46
    ERROR: package/kernel/mac80211 failed to build (build variant: regular).
make[1]: *** [package/Makefile:116: package/kernel/mac80211/download] Error 1
make[1]: *** Waiting for unfinished jobs....
remote: Total 1858 (delta 158), reused 276 (delta 157), pack-reused 1580
Receiving objects: 100% (1858/1858), 148.77 MiB | 61.00 KiB/s, done.
Resolving deltas: 100% (1154/1154), done.
Updating files: 100% (2098/2098), done.
Already on 'main'
Your branch is up to date with 'origin/main'.
Packing checkout...
make[2]: Leaving directory '/openwrt-redmi-ax3000/package/firmware/ath11k-firmware'
time: package/firmware/ath11k-firmware/download#388.90#9.76#2543.71
make[1]: Leaving directory '/openwrt-redmi-ax3000'
make[1]: Entering directory '/openwrt-redmi-ax3000'
make[2]: Entering directory '/openwrt-redmi-ax3000/target/linux'
make[3]: Entering directory '/openwrt-redmi-ax3000/target/linux/ipq50xx'
make[3]: Nothing to be done for 'download'.
make[3]: Leaving directory '/openwrt-redmi-ax3000/target/linux/ipq50xx'
make[2]: Leaving directory '/openwrt-redmi-ax3000/target/linux'
time: target/linux/download#0.06#0.03#0.08
make[1]: Leaving directory '/openwrt-redmi-ax3000'

hzyitc, 2023-05-17:
> /bin/sh: 1: curl: not found

curl没装，请按照官方要求安装软件

> error: RPC failed; curl 56 GnuTLS recv error (-9): Error decoding the received TLS packet.
> fatal: the remote end hung up unexpectedly
> fatal: early EOF
> fatal: index-pack failed

网络问题

3wking, 2023-05-18:
我一直以为linux系统默认带有 curl

3wking, 2023-05-18:
安装 curl 后报这个啊：
make[2]: Leaving directory '/openwrt-redmi-ax3000/package/utils/lua'
time: package/utils/lua/download#0.22#0.06#0.26
mkdir -p /openwrt-redmi-ax3000/dl
SHELL= flock /openwrt-redmi-ax3000/tmp/.jsonfilter-2018-02-04-c7e938d6.tar.xz.flock -c '  [ -f "/openwrt-redmi-ax3000/dl/jsonfilter-2018-02-04-c7e938d6.tar.xz" ] || (  /openwrt-redmi-ax3000/scripts/download.pl "/openwrt-redmi-ax3000/dl" "jsonfilter-2018-02-04-c7e938d6.tar.xz" "0601b4d7aa5ee096e99388a57cb0701673ab58fccd6ed2984a2abbd4f846e045" "" || (         echo "Checking out files from the git repository..."; mkdir -p /openwrt-redmi-ax3000/tmp/dl && cd /openwrt-redmi-ax3000/tmp/dl && rm -rf jsonfilter-2018-02-04-c7e938d6 && [ \! -d jsonfilter-2018-02-04-c7e938d6 ] && git clone  https://git.openwrt.org/project/jsonpath.git jsonfilter-2018-02-04-c7e938d6 && (cd jsonfilter-2018-02-04-c7e938d6 && git checkout c7e938d6582a436dddc938539e72dd1320625c54 && git submodule update --init --recursive) && echo "Packing checkout..." && export TAR_TIMESTAMP=`cd jsonfilter-2018-02-04-c7e938d6 && git log -1 --format='\''@%ct'\''` && rm -rf jsonfilter-2018-02-04-c7e938d6/.git &&         tar --numeric-owner --owner=0 --group=0 --mode=a-s --sort=name ${TAR_TIMESTAMP:+--mtime="$TAR_TIMESTAMP"} -c jsonfilter-2018-02-04-c7e938d6 |   xz -T0 -zc -7e > /openwrt-redmi-ax3000/tmp/dl/jsonfilter-2018-02-04-c7e938d6.tar.xz && mv /openwrt-redmi-ax3000/tmp/dl/jsonfilter-2018-02-04-c7e938d6.tar.xz /openwrt-redmi-ax3000/dl/ && rm -rf jsonfilter-2018-02-04-c7e938d6;  )     )'
make[2]: Leaving directory '/openwrt-redmi-ax3000/package/utils/jsonfilter'
time: package/utils/jsonfilter/download#0.37#0.04#0.40
make[2]: Nothing to be done for 'download'.
make[2]: Leaving directory '/openwrt-redmi-ax3000/package/kernel/linux'
time: package/kernel/linux/download#2.15#0.06#2.27
Copy original source files ...
Applying patches from patches to /openwrt-redmi-ax3000/tmp/dl/backports-20210222-5.4.225 ...
Failed to process SmPL patch 0001-group-attr/0001-group_attr_class.cocci with 1
make[2]: *** [Makefile:657: /openwrt-redmi-ax3000/dl/backports-20210222-5.4.225.tar.xz] Error 2
make[2]: Leaving directory '/openwrt-redmi-ax3000/package/kernel/mac80211'
time: package/kernel/mac80211/regular/download#14.06#8.04#15.36
    ERROR: package/kernel/mac80211 failed to build (build variant: regular).
make[1]: *** [package/Makefile:116: package/kernel/mac80211/download] Error 1
make[1]: Leaving directory '/openwrt-redmi-ax3000'
make[1]: Entering directory '/openwrt-redmi-ax3000'
make[2]: Entering directory '/openwrt-redmi-ax3000/target/linux'
make[3]: Entering directory '/openwrt-redmi-ax3000/target/linux/ipq50xx'
make[3]: Nothing to be done for 'download'.
make[3]: Leaving directory '/openwrt-redmi-ax3000/target/linux/ipq50xx'
make[2]: Leaving directory '/openwrt-redmi-ax3000/target/linux'
time: target/linux/download#0.07#0.03#0.09
make[1]: Leaving directory '/openwrt-redmi-ax3000'

3wking, 2023-05-18:
加了：backports-20210222-5.4.225.tar.xz
make[3]: Leaving directory '/openwrt-redmi-ax3000/feeds/luci/collections/luci'
time: package/feeds/luci/luci/compile#0.34#0.07#0.40
Copy original source files ...
Applying patches from patches to /openwrt-redmi-ax3000/tmp/dl/backports-20210222-5.4.225 ...
Failed to process SmPL patch 0001-group-attr/0001-group_attr_class.cocci with 1
make[3]: *** [Makefile:657: /openwrt-redmi-ax3000/dl/backports-20210222-5.4.225.tar.xz] Error 2
make[3]: Leaving directory '/openwrt-redmi-ax3000/package/kernel/mac80211'
time: package/kernel/mac80211/regular/compile#13.60#6.07#12.39
    ERROR: package/kernel/mac80211 failed to build (build variant: regular).
make[2]: *** [package/Makefile:116: package/kernel/mac80211/compile] Error 1
make[2]: Leaving directory '/openwrt-redmi-ax3000'
make[1]: *** [package/Makefile:110: /openwrt-redmi-ax3000/staging_dir/target-arm_cortex-a7_musl_eabi/stamp/.package_compile] Error 2
make[1]: Leaving directory '/openwrt-redmi-ax3000'
make: *** [/openwrt-redmi-ax3000/include/toplevel.mk:230: world] Error 2

hzyitc, 2023-05-18:
>  加了：backports-20210222-5.4.225.tar.xz

放错位置了吧。这还是在生成这个包。

3wking, 2023-05-19:
是 /dl目录啊

chteow, 2023-06-07:
mac80211 的错码什么哪里不对？

/backup/home/chteow/router/ax3000/build_dir/target-aarch64_cortex-a53_musl/linux-ipq50xx_aarch64/backports-5.10.157-1/net/mac80211/debugfs_netdev.c: In function 'ieee80211_if_read_bmiss_threshold':
/backup/home/chteow/router/ax3000/build_dir/target-aarch64_cortex-a53_musl/linux-ipq50xx_aarch64/backports-5.10.157-1/net/mac80211/debugfs_netdev.c:139:6: error: passing argument 5 of 'ieee80211_if_read' from incompatible pointer type [-Werror=incompatible-pointer-types]
      ieee80211_if_fmt_##name);  \
      ^~~~~~~~~~~~~~~~~
/backup/home/chteow/router/ax3000/build_dir/target-aarch64_cortex-a53_musl/linux-ipq50xx_aarch64/backports-5.10.157-1/net/mac80211/debugfs_netdev.c:160:2: note: in expansion of macro '_IEEE80211_IF_FILE_R_FN'
  _IEEE80211_IF_FILE_R_FN(name)     \
  ^~~~~~~~~~~~~~~~~~~~~~~
/backup/home/chteow/router/ax3000/build_dir/target-aarch64_cortex-a53_musl/linux-ipq50xx_aarch64/backports-5.10.157-1/net/mac80211/debugfs_netdev.c:327:1: note: in expansion of macro 'IEEE80211_IF_FILE_RW'
 IEEE80211_IF_FILE_RW(bmiss_threshold);
 ^~~~~~~~~~~~~~~~~~~~
/backup/home/chteow/router/ax3000/build_dir/target-aarch64_cortex-a53_musl/linux-ipq50xx_aarch64/backports-5.10.157-1/net/mac80211/debugfs_netdev.c:29:12: note: expected 'ssize_t (*)(const struct ieee80211_sub_if_data *, char *, int)' {aka 'long int (*)(const struct ieee80211_sub_if_data *, char *, int)'} but argument is of type 'int (*)(const struct ieee80211_sub_if_data *, char *, int)'
  ssize_t (*format)(const struct ieee80211_sub_if_data *, char *, int))
  ~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
cc1: some warnings being treated as errors
make[9]: *** [scripts/Makefile.build:262: /backup/home/chteow/router/ax3000/build_dir/target-aarch64_cortex-a53_musl/linux-ipq50xx_aarch64/backports-5.10.157-1/net/mac80211/debugfs_netdev.o] Error 1

hzyitc, 2023-06-29:
> mac80211 的错码什么哪里不对？
>
> /backup/home/chteow/router/ax3000/build_dir/target-aarch64_cortex-a53_musl/linux-ipq50xx_aarch64/backports-5.10.157-1/net/mac80211/debugfs_netdev.c: In function [...]

应该是代码错了，我测试的时候没开debug，你把mac80211的debug关了吧。

hzyitc, 2023-06-29:
已经rebase到主线了，这个问题不存在了。

closed 2023-06-29 by hzyitc

legale, 2024-07-22:
> It is most likely `coccinelle`a version problem, it seems that the new version will fail to run.
> 
> Ignore the warnings from the feed, not all packages have been ported.

Is there any chance to fix cocci patches to make them work with actual version of the coccinelle 1.1?