hzyitc/openwrt-redmi-ax3000 #17: R81 (AX3000 works but not tested at all)

SlimRG, 2023-05-06:


SlimRG, 2023-05-06:
But I have only 15 MB for Apps

baysonfox, 2023-06-23:
> But I have only 15 MB for Apps

需要刷入扩容分区表(MIBIB).

baysonfox, 2023-06-23:
补充一个兼容性情况：

- 正常启动
- 交换机正常
- CPU Ethernet 正常
- NSS NAT无法使用(具体表现为iperf3测速时最好只能跑到620Mbps，且 CPU 满载)
- 2.4G/5G WiFi正常，但 5G 160Mhz需要同 README 相同的设置.

启动过程中dmesg: https://gist.github.com/baysonfox/454873fb47d4276241b92487c231a632
(刷了自定义的扩容分区表，不知有无影响)

hzyitc, 2023-06-29:
> NSS NAT无法使用(具体表现为iperf3测速时最好只能跑到620Mbps，且 CPU 满载)

单纯的nat还是PPPoE?

soapmancn, 2023-06-30:
> > NSS NAT无法使用(具体表现为iperf3测速时最好只能跑到620Mbps，且 CPU 满载)
> 
> 
> 
> 单纯的nat还是PPPoE? 路由上跑的iperf3还是外部设备?如果是外部设备，连的是wifi还是有线。

R81可以刷了吗！

baysonfox, 2023-06-30:
> > NSS NAT无法使用(具体表现为iperf3测速时最好只能跑到620Mbps，且 CPU 满载)
> 
> 单纯的nat还是PPPoE? 路由上跑的iperf3还是外部设备?如果是外部设备，连的是wifi还是有线。

目前是纯 NAT (PPPoE还未进行测试), 路由上用iperf3作为服务端和客户端都进行了测试, 作为服务端时速度在 825Mbps 左右, 且比较稳定. 
但作为客户端时速度会有较大波动(869Mbps - 602Mbps), 对端设备是一台J4125的机器, 使用1Gbps的网卡进行有线连接.
(感觉有种类似“暖机”的情况？在部分情况下某次测试最开始的速度可能只有300-400Mbps, 之后速度会回到正常状态.)

作为服务端进行测试:
<pre>-----------------------------------------------------------
Server listening on 5201 (test #1)
-----------------------------------------------------------
Accepted connection from 192.168.1.3, port 55666
[  5] local 192.168.1.1 port 5201 connected to 192.168.1.3 port 55676
[ ID] Interval           Transfer     Bitrate
[  5]   0.00-1.00   sec   104 MBytes   874 Mbits/sec
[  5]   1.00-2.00   sec   108 MBytes   909 Mbits/sec
[  5]   2.00-3.00   sec  76.8 MBytes   642 Mbits/sec
[  5]   3.00-4.00   sec  78.3 MBytes   658 Mbits/sec
[  5]   4.00-5.00   sec   108 MBytes   905 Mbits/sec
[  5]   5.00-6.00   sec   106 MBytes   891 Mbits/sec
[  5]   6.00-7.00   sec   105 MBytes   881 Mbits/sec
[  5]   7.00-8.00   sec   109 MBytes   912 Mbits/sec
[  5]   8.00-9.00   sec  97.9 MBytes   822 Mbits/sec
[  5]   9.00-10.00  sec  96.7 MBytes   811 Mbits/sec
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate
[  5]   0.00-10.00  sec   990 MBytes   830 Mbits/sec                  receiver
-----------------------------------------------------------
Server listening on 5201 (test #2)
-----------------------------------------------------------
Accepted connection from 192.168.1.3, port 53598
[  5] local 192.168.1.1 port 5201 connected to 192.168.1.3 port 53608
[ ID] Interval           Transfer     Bitrate
[  5]   0.00-1.00   sec  99.4 MBytes   832 Mbits/sec
[  5]   1.00-2.00   sec   104 MBytes   875 Mbits/sec
[  5]   2.00-3.00   sec  98.6 MBytes   827 Mbits/sec
[  5]   3.00-4.00   sec  97.3 MBytes   816 Mbits/sec
[  5]   4.00-5.00   sec  97.0 MBytes   814 Mbits/sec
[  5]   5.00-6.01   sec  96.9 MBytes   807 Mbits/sec
[  5]   6.01-7.01   sec  85.5 MBytes   719 Mbits/sec
[  5]   7.01-8.00   sec   103 MBytes   870 Mbits/sec
[  5]   8.00-9.00   sec  98.6 MBytes   827 Mbits/sec
[  5]   9.00-10.00  sec  98.3 MBytes   825 Mbits/sec
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate
[  5]   0.00-10.00  sec   979 MBytes   821 Mbits/sec                  receiver
-----------------------------------------------------------
Server listening on 5201 (test #3)
-----------------------------------------------------------
</pre>

作为客户端进行测试:
<pre>Connecting to host 192.168.1.3, port 5201
[  5] local 192.168.1.1 port 49646 connected to 192.168.1.3 port 5201
[ ID] Interval           Transfer     Bitrate         Retr  Cwnd
[  5]   0.00-1.00   sec  93.1 MBytes   780 Mbits/sec    0    618 KBytes
[  5]   1.00-2.00   sec   103 MBytes   861 Mbits/sec    0    648 KBytes
[  5]   2.00-3.00   sec   104 MBytes   869 Mbits/sec    0    718 KBytes
[  5]   3.00-4.02   sec  98.2 MBytes   807 Mbits/sec    0    836 KBytes
[  5]   4.02-5.01   sec  74.9 MBytes   639 Mbits/sec    0    836 KBytes
[  5]   5.01-6.03   sec  73.9 MBytes   608 Mbits/sec    0    977 KBytes
[  5]   6.03-7.03   sec  72.0 MBytes   602 Mbits/sec    0    977 KBytes
[  5]   7.03-8.00   sec  74.8 MBytes   645 Mbits/sec    0    977 KBytes
[  5]   8.00-9.00   sec  91.7 MBytes   768 Mbits/sec    0    977 KBytes
[  5]   9.00-10.00  sec   101 MBytes   854 Mbits/sec    0   1.21 MBytes
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-10.00  sec   886 MBytes   743 Mbits/sec    0             sender
[  5]   0.00-10.01  sec   886 MBytes   742 Mbits/sec                  receiver
[... 85 lines cut ...]
 1600 root       840 S    odhcp6c -s /lib/netifd/dhcpv6.script -P0 -t120 eth1
 2113 root      1380 S    /usr/sbin/miniupnpd -f /var/etc/miniupnpd.conf
 2526 root      1104 S<   /usr/sbin/ntpd -n -N -S /usr/sbin/ntpd-hotplug -p 0.openwrt.pool.ntp.org -p 1.openwrt.pool.ntp.org -p 2.openwrt.pool.ntp.org -p 3.openwrt.pool.ntp.org -p ntp.aliyun.com -p time1.cloud.te
 3714 root         0 IW   [kworker/u4:2-ev]
 5205 dnsmasq   3360 S    /usr/sbin/dnsmasq -C /var/etc/dnsmasq.conf.cfg01411c -k -x /var/run/dnsmasq/dnsmasq.cfg01411c.pid
 5620 root         0 DW   [kworker/u4:3+ec]
 6259 root         0 IW   [kworker/0:1-ubi]
 6262 root         0 IW   [kworker/u4:1-ev]
 6521 root       888 R    /usr/sbin/dropbear -F -P /var/run/dropbear.1.pid -p 22 -K 300 -T 3
 6522 root      1108 S    -ash
 6667 root         0 IW   [kworker/1:2-mm_]
 6668 root         0 IW   [kworker/1:5-eve]
 6761 root         0 IW   [kworker/0:0-eve]
 7101 root      1108 S    -ash
 7784 root      1104 S    sleep 600
 7874 root      1104 S    sleep 600
 7876 root         0 IW   [kworker/u4:4-ev]
 7878 root      1100 R    ps
12489 root      1548 S    /usr/sbin/uhttpd -f -h /www -r OpenWrt -x /cgi-bin -u /ubus -t 3600 -T 30 -k 20 -A 1 -n 50 -N 100 -R -p 0.0.0.0:80 -p [::]:80
17259 root      1376 S    /bin/sh /usr/lib/ddns/dynamic_dns_updater.sh -v 0 -S myddns_ipv6 -- start
20336 root      1080 S    /usr/sbin/pppd nodetach ipparam wan ifname pppoe-wan lcp-echo-interval 1 lcp-echo-failure 5 lcp-echo-adaptive +ipv6 set AUTOIPV6=1 nodefaultroute usepeerdns maxfail 1 user **REDACTED**
20581 root       840 S    odhcp6c -s /lib/netifd/dhcpv6.script -P0 -t120 pppoe-wan
21061 root      1368 S    /bin/sh /usr/lib/ddns/dynamic_dns_updater.sh -v 0 -S myddns_ipv4 -- start
32049 root         0 DW   [kworker/u4:0+ec]
</pre>

woyeshizuileye, 2023-11-06:
> > But I have only 15 MB for Apps
> 
> 需要刷入扩容分区表(MIBIB).

能分享一个吗，找了好久了，一直找不到

dnx-xy, 2024-03-15:
hi, do you mind share what you have progress with Xiaomi AX3000. Would love to see how it perform