hzyitc/openwrt-redmi-ax3000 #20: target 镜像如何选择?

IITII, 2023-05-17:
## 目的
* 通过 luci 网页更新 op
## 编译后的文件
* 目前是根据下面的 config 配置编译出来了以下文件, 但是还是不知道用哪个镜像可以通过 网页来更新...
* 尝试过直接用 ubi 文件进行更新, 但是貌似更新失败了
* 以前一般用 bin/img/trx 就可以在 luci web 上面进行更新, 但是都没有,,,
https://openwrt.org/docs/techref/image.format
```
bin/targets/ipq50xx/arm/
├── config.buildinfo
├── feeds.buildinfo
├── openwrt-ipq50xx-arm-redmi_ax3000-ext4-nand-factory.ubi.gz
├── openwrt-ipq50xx-arm-redmi_ax3000-initramfs-fit-uImage.itb
├── openwrt-ipq50xx-arm-redmi_ax3000.manifest
├── openwrt-ipq50xx-arm-redmi_ax3000-rootfs.tar.gz
├── openwrt-ipq50xx-arm-redmi_ax3000-squashfs-nand-factory.ubi
├── packages
├── sha256sums
└── version.buildinfo
```
## config 配置
[image]

```bash
#❯ grep TARGET .config | grep -v ^#
CONFIG_TARGET_ipq50xx=y
CONFIG_TARGET_ipq50xx_arm=y
CONFIG_TARGET_ipq50xx_arm_DEVICE_redmi_ax3000=y
CONFIG_HAS_SUBTARGETS=y
CONFIG_TARGET_BOARD="ipq50xx"
CONFIG_TARGET_SUBTARGET="arm"
CONFIG_TARGET_PROFILE="DEVICE_redmi_ax3000"
CONFIG_TARGET_ARCH_PACKAGES="arm_cortex-a7"
CONFIG_DEFAULT_TARGET_OPTIMIZATION="-Os -pipe"
CONFIG_TARGET_ROOTFS_INITRAMFS=y
CONFIG_TARGET_INITRAMFS_COMPRESSION_NONE=y
CONFIG_TARGET_ROOTFS_TARGZ=y
CONFIG_TARGET_ROOTFS_EXT4FS=y
CONFIG_TARGET_EXT4_RESERVED_PCT=0
CONFIG_TARGET_EXT4_BLOCKSIZE_4K=y
CONFIG_TARGET_EXT4_BLOCKSIZE=4096
CONFIG_TARGET_ROOTFS_SQUASHFS=y
CONFIG_TARGET_SQUASHFS_BLOCK_SIZE=256
CONFIG_TARGET_UBIFS_FREE_SPACE_FIXUP=y
CONFIG_TARGET_UBIFS_JOURNAL_SIZE=""
CONFIG_TARGET_IMAGES_GZIP=y
CONFIG_TARGET_ROOTFS_PARTSIZE=104
CONFIG_TARGET_ROOTFS_DIR=""
CONFIG_TARGET_OPTIMIZATION="-Os -pipe"
CONFIG_TARGET_SUFFIX="muslgnueabi"
CONFIG_TARGET_PREINIT_SUPPRESS_STDERR=y
CONFIG_TARGET_PREINIT_TIMEOUT=2
CONFIG_TARGET_PREINIT_IFNAME=""
CONFIG_TARGET_PREINIT_IP="192.168.1.1"
CONFIG_TARGET_PREINIT_NETMASK="255.255.255.0"
CONFIG_TARGET_PREINIT_BROADCAST="192.168.1.255"
CONFIG_TARGET_INIT_PATH="/usr/sbin:/usr/bin:/sbin:/bin"
CONFIG_TARGET_INIT_ENV=""
CONFIG_TARGET_INIT_CMD="/sbin/init"
CONFIG_TARGET_INIT_SUPPRESS_STDERR=y
```

* 提前谢谢大佬

hzyitc, 2023-05-17:
可以直接用ubi更新啊。原生uboot就行，我做的是AB启动的。你要是改过分区表的话就可能会出错。

IITII, 2023-05-18:
我这边是不行的， 看起来貌似是少了分区（然而我并没有修改相关的配置

luci 点击更新后， 串口日志
```
[  193.280023] wlan1: Destroyed NSS virtual interface
Sun May 14 22:17:29 CST 2023 upgrade: Sending KILL to remaining processes ...sh: S: out of range

[  197.034758] sh (3419): drop_caches: 3
Sun May 14 22:17:31 CST 2023 upgrade: Switching to ramdisk...
[  199.035949] UBIFS (ubi0:2): background thread "ubifs_bgt0_2" stops
[  199.045267] UBIFS (ubi0:2): un-mount UBI device 0
Sun May 14 14:17:33 UTC 2023 upgrade: Performing system upgrade...
libmtd: error!: cannot get information about "/dev/mtd18 mtd23"
        error 2 (No such file or directory)
ubiformat: error!: cannot get information about "/dev/mtd18 mtd23"
           error 2 (No such file or directory)
Sun May 14 14:17:33 UTC 2023 upgrade: Upgrade completed
Sun May 14 14:17:34 UTC 2023 upgrade: Rebooting system...
umount: can't unmount /dev: Resource busy
umount: can't unmount /tmp: Resource

U-Boot 2016.01 (Nov 23 2021 - 05:24:42 +0000), Build: jenkins-common_router_openwrt_ota_publish-2222

DRAM:  smem ram ptable found: ver: 1 len: 4
256 MiB
USB: Node Not found, skipping initia▒*9▒  QPIC controller support serial NAND
ID = 41c841c8
Vendor = c8
Device = 41
Serial Nand Device Found With ID : 0xc8 0x41
Serial NAND device Manufacturer:GD5F1GQ5REYIG
Device Size:256 MiB, Page size:2048, Spare Size:128, ECC:8-bit
qpic_nand: changing oobsize to 80 from 128 bytes
SF: Unsupported flash IDs: manuf 00, jedec 0000, ext_jedec 0000
ipq_spi: SPI Flash not found (bus/cs/speed/mode) = (0/0/48000000/0)
256 MiB
MMC:   sdhci: Node Not found, skipping initialization
```

IITII, 2023-05-19:
```
root@OpenWrt:~# cat /proc/mtd
dev:    size   erasesize  name
mtd0: 00080000 00020000 "0:SBL1"
mtd1: 00080000 00020000 "0:MIBIB"
mtd2: 00040000 00020000 "0:BOOTCONFIG"
mtd3: 00040000 00020000 "0:BOOTCONFIG1"
mtd4: 00100000 00020000 "0:QSEE"
mtd5: 00100000 00020000 "0:QSEE_1"
mtd6: 00040000 00020000 "0:DEVCFG"
mtd7: 00040000 00020000 "0:DEVCFG_1"
mtd8: 00040000 00020000 "0:CDT"
mtd9: 00040000 00020000 "0:CDT_1"
mtd10: 00080000 00020000 "0:APPSBLENV"
mtd11: 00140000 00020000 "0:APPSBL"
mtd12: 00140000 00020000 "0:APPSBL_1"
mtd13: 00100000 00020000 "0:ART"
mtd14: 00080000 00020000 "0:TRAINING"
mtd15: 00080000 00020000 "bdata"
mtd16: 00080000 00020000 "crash"
mtd17: 00080000 00020000 "crash_log"
mtd18: 02400000 00020000 "rootfs"
mtd19: 02400000 00020000 "rootfs_1"
mtd20: 01f00000 00020000 "overlay"
mtd21: 00d80000 00020000 "data"
mtd22: 0043d000 0001f000 "kernel"
mtd23: 00991000 0001f000 "rootfs"
mtd24: 00fbe000 0001f000 "rootfs_data"
```

hzyitc, 2023-05-26:
确认了，确实存在这个问题。待修复。

```
root@OpenWrt:/# find_mtd_index "rootfs"
18 mtd23
```

hzyitc, 2023-05-27:
已修复。

closed 2023-05-27 by hzyitc