hzyitc/openwrt-redmi-ax3000 #24: AX3000 编译时如何添加ssr-plus

wilimcn, 2023-06-13:
我通过sed -i '$a src-git helloworld https://github.com/fw876/helloworld.git' ./feeds.conf.default
添加 CONFIG_PACKAGE_luci-app-ssr-plus=y
编译出错   能指点下吗？

wilimcn, 2023-06-13:
Build
警告
WARNING: Makefile 'package/kernel/qca/qca-nss-clients/Makefile' has a dependency on 'kmod-qca-nss-cfi-cryptoapi', which does not exist
WARNING: Makefile 'package/kernel/qca/qca-nss-clients/Makefile' has a dependency on 'kmod-qca-wifi', which does not exist
WARNING: Makefile 'package/kernel/qca/qca-nss-clients/Makefile' has a build dependency on 'kmod-qca-nss-crypto', which does not exist
WARNING: Makefile 'package/kernel/qca/qca-nss-drv/Makefile' has a dependency on 'kmod-qca-nss-gmac', which does not exist
WARNING: Makefile 'package/kernel/qca/qca-nss-ecm/Makefile' has a dependency on 'kmod-qca-ovsmgr', which does not exist
WARNING: Makefile 'package/kernel/qca/qca-nss-ecm/Makefile' has a dependency on 'kmod-shortcut-fe-drv', which does not exist
WARNING: Makefile 'package/kernel/qca/qca-nss-ecm/Makefile' has a dependency on 'kmod-qca-mcs', which does not exist
WARNING: Makefile 'package/kernel/qca/qca-nss-ecm/Makefile' has a dependency on 'kmod-qca-ovsmgr', which does not exist
WARNING: Makefile 'package/kernel/qca/qca-nss-ecm/Makefile' has a dependency on 'kmod-qca-mcs', which does not exist
WARNING: Makefile 'package/kernel/qca/qca-nss-ecm/Makefile' has a dependency on 'kmod-qca-ovsmgr', which does not exist
WARNING: Makefile 'package/kernel/qca/qca-nss-ecm/Makefile' has a dependency on 'kmod-shortcut-fe-drv', which does not exist
WARNING: Makefile 'package/kernel/qca/qca-nss-ecm/Makefile' has a dependency on 'kmod-qca-mcs', which does not exist
WARNING: Makefile 'package/kernel/qca/qca-nss-ecm/Makefile' has a dependency on 'kmod-qca-hyfi-bridge', which does not exist
WARNING: Makefile 'package/kernel/qca/qca-nss-ecm/Makefile' has a dependency on 'kmod-emesh-sp', which does not exist
WARNING: Makefile 'package/kernel/qca/qca-nss-ecm/Makefile' has a dependency on 'kmod-qca-ovsmgr', which does not exist
WARNING: Makefile 'package/kernel/qca/qca-nss-ecm/Makefile' has a dependency on 'kmod-shortcut-fe-drv', which does not exist
WARNING: Makefile 'package/kernel/qca/qca-nss-ecm/Makefile' has a dependency on 'kmod-qca-mcs', which does not exist
WARNING: Makefile 'package/kernel/qca/qca-nss-ecm/Makefile' has a dependency on 'kmod-qca-hyfi-bridge', which does not exist
 
出错
make[3] -C package/packages/openwrt-passwall/xray-core compile
    ERROR: package/packages/openwrt-passwall/xray-core failed to build.
make -r world: build failed. Please re-run make with -j1 V=s or V=sc for a higher verbosity level to see what's going on
make: *** [/home/runner/work/openwrt-redmi-ax3000/openwrt-redmi-ax3000/include/toplevel.mk:230: world] Error 1
Error: Process completed with exit code 2.

hzyitc, 2023-06-15:
这看起来是这个软件的问题啊。你问问包作者吧。我用的分支比较老，得确定这个包支持21.02。23.05的包及5.15内核我还在做。

hzyitc, 2023-06-23:
> 你好，23.05的包大概什么时间能出来呢？多谢了。

23.05目前只支持2.4G WiFi。5G WiFi我还在研究。

WhaleFell, 2024-12-03:
u need to update feeds golang version to compile xray-core, follow above:

```shell
rm -rf feeds/packages/lang/golang
git clone https://github.com/sbwml/packages_lang_golang -b 22.x feeds/packages/lang/golang
```