hzyitc/openwrt-redmi-ax3000 #25: [New device request] RAX3000Q

zclkkk, 2023-06-16:
MTD info:

dev:    size   erasesize  name
mtd0: 00080000 00020000 "0:SBL1"
mtd1: 00080000 00020000 "0:MIBIB"
mtd2: 00040000 00020000 "0:BOOTCONFIG"
mtd3: 00040000 00020000 "0:BOOTCONFIG1"
mtd4: 00100000 00020000 "0:QSEE_1"
mtd5: 00100000 00020000 "0:QSEE"
mtd6: 00040000 00020000 "0:DEVCFG_1"
mtd7: 00040000 00020000 "0:DEVCFG"
mtd8: 00040000 00020000 "0:CDT"
mtd9: 00040000 00020000 "0:CDT_1"
mtd10: 00080000 00020000 "0:APPSBLENV"
mtd11: 00140000 00020000 "0:APPSBL"
mtd12: 00140000 00020000 "0:APPSBL_1"
mtd13: 00100000 00020000 "0:ART"
mtd14: 00080000 00020000 "0:TRAINING"
mtd15: 03000000 00020000 "rootfs_1"
mtd16: 03000000 00020000 "rootfs"
mtd17: 00a40000 00020000 "TZPARAM"
mtd18: 00a40000 00020000 "TZBAK"
mtd19: 002fce60 0001f000 "kernel"
mtd20: 00367000 0001f000 "wifi_fw"
mtd21: 00004000 0001f000 "bt_fw"
mtd22: 01303000 0001f000 "ubi_rootfs"
mtd23: 00c1c000 0001f000 "rootfs_data"
mtd24: 00706000 0001f000 "tzparam"

hzyitc, 2023-06-23:
Please provide the boot log and the origin firmware. And make sure you known how to recover firmware after boot fail.

viqbgrg, 2023-08-05:
How to submit these things, I am willing to provide

hzyitc, 2023-08-10:
> How to submit these things, I am willing to provide

Just paste here.

viqbgrg, 2023-11-13:
不知道为什么ttl,我要晚一会儿接gnd线才能显示东西,而且一开始可能还有乱码,有没有别的办法获取启动日志

hzyitc, 2023-11-18:
> 不知道为什么ttl,我要晚一会儿接gnd线才能显示东西,而且一开始可能还有乱码,有没有别的办法获取启动日志

那就是你的ttl模块电压不对，5018采用1.8V电平，且串口部分与配置引脚复用。电平过高的话有可能导致启动项配置错误。

viqbgrg, 2023-11-19:
我买的就是普通的ch341a,可能调节电压吗? 我之前是插电源孔开机,会出现那个问题,我现在开插座电源就不会那个问题,但是我看日志偶尔还是有乱码,不过我把日志,还有分区备份弄好了,发给你看一下链接：https://pan.baidu.com/s/1-vKgttrrRgJ-AB6V1fPR9g 
提取码：2efj

hzyitc, 2023-11-19:
> 我买的就是普通的ch341a,可能调节电压吗? 我之前是插电源孔开机,会出现那个问题,我现在开插座电源就不会那个问题,但是我看日志偶尔还是有乱码,不过我把日志,还有分区备份弄好了,发给你看一下链接：https://pan.baidu.com/s/1-vKgttrrRgJ-AB6V1fPR9g 提取码：2efj

打成压缩包发到github就行，百度盘下载太麻烦了。

viqbgrg, 2023-11-19:
好的,重新发送给你https://github.com/viqbgrg/rax3000q/raw/main/%E6%97%A5%E5%BF%97%E5%92%8C%E5%A4%87%E4%BB%BD.zip

viqbgrg, 2023-11-24:
@zclkkk 你好,你是不是修改了BOOTCONFIG 分区的内容了,把rootfs和rootfs_1的优先顺序给改了,现在我的rootfs分区坏了,我想使用rootfs_1分区,也想改一个这个顺序

lilaoye88, 2024-10-12:
这个是超频到1.3G了吗

closed 2024-10-18 by zclkkk