hzyitc/openwrt-redmi-ax3000 #27: 编译错误

Tongrens, 2023-06-30:
make的时候会有关于qca的警告，如下：
```info
WARNING: Makefile 'package/kernel/qca/qca-nss-clients/Makefile' has a build dependency on 'kmod-qca-nss-crypto', which does not exist
WARNING: Makefile 'package/kernel/qca/qca-nss-drv/Makefile' has a dependency on 'kmod-qca-nss-gmac', which does not exist
WARNING: Makefile 'package/kernel/qca/qca-ssdk/Makefile' has a dependency on 'kmod-qca-rfs', which does not exist
```

编译到qca-nss-drv的时候提示错误：
```error
    ERROR: package/kernel/qca/qca-nss-drv failed to build.
make -r world: build failed. Please re-run make with -j1 V=s or V=sc for a higher verbosity level to see what's going on
make: *** [/home/kali/openwrt-redmi-ax3000/include/toplevel.mk:230：world] 错误 1
```

使用make -j1 V=s进行编译提示'No such file or directory'：
```error
rm -f /home/kali/openwrt-redmi-ax3000/build_dir/target-mips_24kc_musl/linux-ath79_generic/qca-nss-drv-NHSS.QSDK.11.5.0.5/.built
touch /home/kali/openwrt-redmi-ax3000/build_dir/target-mips_24kc_musl/linux-ath79_generic/qca-nss-drv-NHSS.QSDK.11.5.0.5/.built_check
make -C "/home/kali/openwrt-redmi-ax3000/build_dir/target-mips_24kc_musl/linux-ath79_generic/linux-5.4.225" NSS_DRV_C2C_ENABLE=n NSS_DRV_CAPWAP_ENABLE=n NSS_DRV_CLMAP_ENABLE=n NSS_DRV_CRYPTO_ENABLE=n NSS_DRV_DTLS_ENABLE=n NSS_DRV_GRE_ENABLE=n NSS_DRV_GRE_REDIR_ENABLE=n NSS_DRV_GRE_TUNNEL_ENABLE=n NSS_DRV_IGS_ENABLE=n NSS_DRV_IPSEC_ENABLE=n NSS_DRV_LAG_ENABLE=n NSS_DRV_L2TP_ENABLE=n NSS_DRV_MAPT_ENABLE=n NSS_DRV_OAM_ENABLE=n NSS_DRV_PPTP_ENABLE=n NSS_DRV_PORTID_ENABLE=n NSS_DRV_PVXLAN_ENABLE=n NSS_DRV_QRFS_ENABLE=n NSS_DRV_QVPN_ENABLE=n NSS_DRV_RMNET_ENABLE=n NSS_DRV_SHAPER_ENABLE=n NSS_DRV_SJACK_ENABLE=n NSS_DRV_TLS_ENABLE=n NSS_DRV_TRUSTSEC_ENABLE=n NSS_DRV_TSTAMP_ENABLE=n NSS_DRV_TUN6RD_ENABLE=n NSS_DRV_TUNIPIP6_ENABLE=n NSS_DRV_VXLAN_ENABLE=n NSS_DRV_MATCH_ENABLE=n NSS_DRV_MIRROR_ENABLE=n NSS_DRV_IPV4_REASM_ENABLE=n NSS_DRV_IPV6_REASM_ENABLE=n CROSS_COMPILE="mips-openwrt-linux-musl-" ARCH="mips" M="/home/kali/openwrt-redmi-ax3000/build_dir/target-mips_24kc_musl/linux-ath79_generic/qca-nss-drv-NHSS.QSDK.11.5.0.5" EXTRA_CFLAGS="-I/home/kali/openwrt-redmi-ax3000/staging_dir/target-mips_24kc_musl/usr/include/qca-nss-gmac -I/home/kali/openwrt-redmi-ax3000/staging_dir/target-mips_24kc_musl/usr/include/qca-nss-dp -I/home/kali/openwrt-redmi-ax3000/staging_dir/target-mips_24kc_musl/usr/include/qca-ssdk -DNSS_MEM_PROFILE_LOW" SoC=""ath79"" modules
make[4]: Entering directory '/home/kali/openwrt-redmi-ax3000/build_dir/target-mips_24kc_musl/linux-ath79_generic/linux-5.4.225'
  CC [M]  /home/kali/openwrt-redmi-ax3000/build_dir/target-mips_24kc_musl/linux-ath79_generic/qca-nss-drv-NHSS.QSDK.11.5.0.5/nss_cmn.o
In file included from /home/kali/openwrt-redmi-ax3000/build_dir/target-mips_24kc_musl/linux-ath79_generic/qca-nss-drv-NHSS.QSDK.11.5.0.5/nss_core.h:37,
                 from /home/kali/openwrt-redmi-ax3000/build_dir/target-mips_24kc_musl/linux-ath79_generic/qca-nss-drv-NHSS.QSDK.11.5.0.5/nss_hal/include/nss_hal.h:26,
                 from /home/kali/openwrt-redmi-ax3000/build_dir/target-mips_24kc_musl/linux-ath79_generic/qca-nss-drv-NHSS.QSDK.11.5.0.5/nss_tx_rx_common.h:25,
                 from /home/kali/openwrt-redmi-ax3000/build_dir/target-mips_24kc_musl/linux-ath79_generic/qca-nss-drv-NHSS.QSDK.11.5.0.5/nss_cmn.c:26:
/home/kali/openwrt-redmi-ax3000/build_dir/target-mips_24kc_musl/linux-ath79_generic/qca-nss-drv-NHSS.QSDK.11.5.0.5/exports/nss_api_if.h:33:10: fatal error: nss_arch.h: No such file or directory
 #include "nss_arch.h"
          ^~~~~~~~~~~~
compilation terminated.
make[5]: *** [scripts/Makefile.build:262: /home/kali/openwrt-redmi-ax3000/build_dir/target-mips_24kc_musl/linux-ath79_generic/qca-nss-drv-NHSS.QSDK.11.5.0.5/nss_cmn.o] Error 1
make[4]: *** [Makefile:1739: /home/kali/openwrt-redmi-ax3000/build_dir/target-mips_24kc_musl/linux-ath79_generic/qca-nss-drv-NHSS.QSDK.11.5.0.5] Error 2
make[4]: Leaving directory '/home/kali/openwrt-redmi-ax3000/build_dir/target-mips_24kc_musl/linux-ath79_generic/linux-5.4.225'
make[3]: *** [Makefile:143: /home/kali/openwrt-redmi-ax3000/build_dir/target-mips_24kc_musl/linux-ath79_generic/qca-nss-drv-NHSS.QSDK.11.5.0.5/.built] Error 2
make[3]: Leaving directory '/home/kali/openwrt-redmi-ax3000/package/kernel/qca/qca-nss-drv'
time: package/kernel/qca/qca-nss-drv/compile#0.70#0.46#1.01
    ERROR: package/kernel/qca/qca-nss-drv failed to build.
make[2]: *** [package/Makefile:116: package/kernel/qca/qca-nss-drv/compile] Error 1
make[2]: Leaving directory '/home/kali/openwrt-redmi-ax3000'
make[1]: *** [package/Makefile:110: /home/kali/openwrt-redmi-ax3000/staging_dir/target-mips_24kc_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/kali/openwrt-redmi-ax3000'
make: *** [/home/kali/openwrt-redmi-ax3000/include/toplevel.mk:230：world] 错误 2
```

小白编译求教0.0

hzyitc, 2023-07-01:
> `target-mips_24kc_musl/linux-ath79_generic/`

选错架构了

closed 2023-07-01 by hzyitc

kettly1260, 2024-02-20:
@hzyitc  time: target/linux/prereq#0.05#0.00#0.06
make[1]: Entering directory '/home/ying/openwrt-redmi-ax3000'
make[1]: *** No rule to make target 'with'.  Stop.
make[1]: Leaving directory '/home/ying/openwrt-redmi-ax3000'
make: *** [/home/ying/openwrt-redmi-ax3000/include/toplevel.mk:230: with] Error 2

我也遇到了这个问题

hzyitc, 2024-03-03:
> @hzyitc time: target/linux/prereq#0.05#0.00#0.06 make[1]: Entering directory '/home/ying/openwrt-redmi-ax3000' make[1]: *** No rule to make target 'with'. Stop. make[1]: Leaving directory [...]

看起来像是编译命令错了