hzyitc/openwrt-redmi-ax3000 #34: 缺少librt库

zhangbaichang, 2023-08-07:
安装软件包，发现缺少了librt库，发现默认源中指向了一个不存在的文件夹https://archive.openwrt.org/releases/21.02.0/targets/ipq50xx
貌似openwrt不支持此设备，请问如何解决

zhangbaichang, 2023-08-07:
更换作者github上的源后，提示架构不同无法安装:
`Package kmod-usb-core version 5.4-qsdk-11.5.0.5-1 has no valid architecture, ignoring.
Package libgcc1 version 8.4.0-3 has no valid architecture, ignoring.
Package librt version 1.1.24-3 has no valid architecture, ignoring.
Package libxtables12 version 1.8.7-1 has no valid architecture, ignoring.
Package ath10k-firmware-qca4019-ct-htt version 2020-11-08-1 has no valid architecture, ignoring.
Package ath10k-firmware-qca4019-ct version 2020-11-08-1 has no valid architecture, ignoring.
Package ath10k-firmware-qca4019 version 20211216-1 has no valid architecture, ignoring.
Package ath10k-firmware-qca9887-ct version 2020-11-08-1 has no valid architecture, ignoring.
Package ath10k-firmware-qca9887 version 20211216-1 has no valid architecture, ignoring.
Package ath10k-firmware-qca9888-ct-htt version 2020-11-08-1 has no valid architecture, ignoring.
Package ath10k-firmware-qca9888-ct version 2020-11-08-1 has no valid architecture, ignoring.
Package ath10k-firmware-qca9888 version 20211216-1 has no valid architecture, ignoring.
Package ath10k-firmware-qca988x-ct version 2020-11-08-1 has no valid architecture, ignoring.
Package ath10k-firmware-qca988x version 20211216-1 has no valid architecture, ignoring.
Package ath10k-firmware-qca9984-ct-htt version 2020-11-08-1 has no valid architecture, ignoring.
Package ath10k-firmware-qca9984-ct version 2020-11-08-1 has no valid architecture, ignoring.
Package ath10k-firmware-qca9984 version 20211216-1 has no valid architecture, ignoring.
Package ath10k-firmware-qca99x0-ct-htt version 2020-11-08-1 has no valid architecture, ignoring.
Package ath10k-firmware-qca99x0-ct version 2020-11-08-1 has no valid architecture, ignoring.
Package ath10k-firmware-qca99x0 version 20211216-1 has no valid architecture, ignoring.
Package eapol-test-wolfssl version 2020-06-08-5a8b3662-41 has no valid architecture, ignoring.
Package eapol-test version 2020-06-08-5a8b3662-41 has no valid architecture, ignoring.
Package hostapd-basic version 2020-06-08-5a8b3662-41 has no valid architecture, ignoring.
Package hostapd-common version 2020-06-08-5a8b3662-41 has no valid architecture, ignoring.
Package hostapd version 2020-06-08-5a8b3662-41 has no valid architecture, ignoring.
Package libbfd version 2.35.1-2 has no valid architecture, ignoring.
Package libbpf0 version 5.10.10-2 has no valid architecture, ignoring.
Package libdw1 version 0.180-1 has no valid architecture, ignoring.
Package libelf1 version 0.180-1 has no valid architecture, ignoring.
Package libjson-c5 version 0.15-2 has no valid architecture, ignoring.
Package libmnl0 version 1.0.4-2 has no valid architecture, ignoring.
Package libnetfilter-conntrack3 version 1.0.8-1 has no valid architecture, ignoring.
Package libnettle8 version 3.6-1 has no valid architecture, ignoring.
Package libnfnetlink0 version 1.0.1-4 has no valid architecture, ignoring.
Package libnl-route200 version 3.5.0-1 has no valid architecture, ignoring.
Package libnl-tiny1 version 2020-08-05-c291088f-2 has no valid architecture, ignoring.
Package libopcodes version 2.35.1-2 has no valid architecture, ignoring.
Package libopenssl1.1 version 1.1.1t-2 has no valid architecture, ignoring.
Package libselinux version 3.2-1 has no valid architecture, ignoring.
Package libss2 version 1.45.6-2 has no valid architecture, ignoring.
Package libubox20210516 version 2021-05-16-b14c4688-2 has no valid architecture, ignoring.
Package libubus20210630 version 2021-06-30-4fc532c8-2 has no valid architecture, ignoring.
Package libuci20130104 version 2021-04-14-4b3db117-5 has no valid architecture, ignoring.
Package libuuid1 version 2.36.1-2 has no valid architecture, ignoring.
Package libwolfssl5.5.3.99a5b54a version 5.5.3-stable-1 has no valid architecture, ignoring.
Package resolveip version 2 has no valid architecture, ignoring.
Package tc-mod-iptables version 5.11.0-3 has no valid architecture, ignoring.
Package terminfo version 6.2-3 has no valid architecture, ignoring.
Package ubox version 2020-10-25-9ef88681-2 has no valid architecture, ignoring.
Package ubus version 2021-06-30-4fc532c8-2 has no valid architecture, ignoring.
Package ubusd version 2021-06-30-4fc532c8-2 has no valid architecture, ignoring.
Package usign version 2020-05-23-f1f65026-1 has no valid architecture, ignoring.
Package wpa-supplicant-mesh-openssl version 2020-06-08-5a8b3662-41 has no valid architecture, ignoring.
Package wpa-supplicant-mesh-wolfssl version 2020-06-08-5a8b3662-41 has no valid architecture, ignoring.
Package wpa-supplicant-p2p version 2020-06-08-5a8b3662-41 has no valid architecture, ignoring.
Package wpa-supplicant version 2020-06-08-5a8b3662-41 has no valid architecture, ignoring.
Package wpad-basic-openssl version 2020-06-08-5a8b3662-41 has no valid architecture, ignoring.
Package wpad-basic-wolfssl version 2020-06-08-5a8b3662-41 has no valid architecture, ignoring.
Package wpad-basic version 2020-06-08-5a8b3662-41 has no valid architecture, ignoring.
Package wpad-mesh-openssl version 2020-06-08-5a8b3662-41 has no valid architecture, ignoring.
Package wpad-mesh-wolfssl version 2020-06-08-5a8b3662-41 has no valid architecture, ignoring.
Package wpad-mini version 2020-06-08-5a8b3662-41 has no valid architecture, ignoring.
Package wpad-openssl version 2020-06-08-5a8b3662-41 has no valid architecture, ignoring.
Package wpad-wolfssl version 2020-06-08-5a8b3662-41 has no valid architecture, ignoring.
[... cut ...]
Package vim-full version 8.2-3 has no valid architecture, ignoring.
Package vim-fuller version 8.2-3 has no valid architecture, ignoring.
Package vim version 8.2-3 has no valid architecture, ignoring.
Package yajl version 2.1.0-2 has no valid architecture, ignoring.
Package zoneinfo-simple version 2023c-1 has no valid architecture, ignoring.
Package libks1 version 1.7.0-1 has no valid architecture, ignoring.
Unknown package 'librt'.
Collected errors:
 * opkg_install_cmd: Cannot install package librt.`

zhangbaichang, 2023-08-07:
修改opkg配置文件后问题已解决

closed 2023-08-07 by zhangbaichang

zhangbaichang, 2025-05-25:
> [@zhangbaichang](https://github.com/zhangbaichang) 请问怎么解决的啊？我安装ttyd也提示找不到librt

这个路由已经卖了，换了rax3000m，稳定一年了。这机子固件少的可怜，能用的固件还有一些小问题，受不了

fly263, 2025-05-25:
> > [@zhangbaichang](https://github.com/zhangbaichang) 请问怎么解决的啊？我安装ttyd也提示找不到librt
> 
> 这个路由已经卖了，换了rax3000m，稳定一年了。这机子固件少的可怜，能用的固件还有一些小问题，受不了

感谢回复，昨晚我重新安装又能安装上了。AX3000闪存是有点小，只能小折腾。