hzyitc/openwrt-redmi-ax3000 #35: WiFi拒绝加入

zhangbaichang, 2023-08-07:
红米AX3000与小米AX3000硬件配置相同，而我的小米AX3000刷入固件后2，4g与5g发射正常，但无法连接，手机显示拒绝连接
使用其他作者编译的固件，大部分WiFi可用但版本太低，这个固件版本新，但WiFi异常

hzyitc, 2023-08-07:
贴个日志吧。

而且硬件配置相同不代表就一定通用，特别是wifi这种配置信息跟硬件关系极大的。

自己检查下bdf文件和校准文件吧。

zhangbaichang, 2023-08-08:
这是系统日志:
```
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 5.4.164 (runner@fv-az457-764) (gcc version 8.4.0 (OpenWrt GCC 8.4.0 ci-20230716-155109-a41da3c9)) #0 SMP PREEMPT Sun Jul 16 15:48:41 2023
[    0.000000] CPU: ARMv7 Processor [51af8014] revision 4 (ARMv7), cr=10c0383d
[    0.000000] CPU: div instructions available: patching division code
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
[    0.000000] OF: fdt: Machine model: Redmi AX3000
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] On node 0 totalpages: 46336
[    0.000000]   Normal zone: 576 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 46336 pages, LIFO batch:15
[    0.000000] psci: probing for conduit method from DT.
[    0.000000] psci: PSCIv1.0 detected in firmware.
[    0.000000] psci: Using standard PSCI v0.2 function IDs
[    0.000000] psci: MIGRATE_INFO_TYPE not supported.
[    0.000000] psci: SMC Calling Convention v1.0
[    0.000000] psci: OSI mode supported.
[    0.000000] percpu: Embedded 15 pages/cpu s32204 r8192 d21044 u61440
[    0.000000] pcpu-alloc: s32204 r8192 d21044 u61440 alloc=15*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 45760
[    0.000000] Kernel command line: console=ttyMSM0,115200n8 ubi.mtd=rootfs root=mtd:ubi_rootfs rootfstype=squashfs rootwait root=/dev/ubiblock0_1 coherent_pool=2M
[    0.000000] Dentry cache hash table entries: 32768 (order: 5, 131072 bytes, linear)
[    0.000000] Inode-cache hash table entries: 16384 (order: 4, 65536 bytes, linear)
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory: 172204K/185344K available (6894K kernel code, 262K rwdata, 2088K rodata, 1024K init, 360K bss, 13140K reserved, 0K cma-reserved, 0K highmem)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=2, Nodes=1
[    0.000000] rcu: Preemptible hierarchical RCU implementation.
[    0.000000] 	Tasks RCU enabled.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 10 jiffies.
[    0.000000] NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
[    0.000000] GICv2m: range[mem 0x0b00a000-0x0b00affc], SPI[448:479]
[    0.000000] GICv2m: range[mem 0x0b00b000-0x0b00bffc], SPI[480:511]
[    0.000000] random: get_random_bytes called from start_kernel+0x344/0x4f4 with crng_init=0
[    0.000000] arch_timer: cp15 and mmio timer(s) running at 24.00MHz (virt/virt).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0x588fe9dc0, max_idle_ns: 440795202592 ns
[    0.000006] sched_clock: 56 bits at 24MHz, resolution 41ns, wraps every 4398046511097ns
[    0.000019] Switching to timer-based delay loop, resolution 41ns
[    0.000417] Console: colour dummy device 80x30
[    0.000449] Calibrating delay loop (skipped), value calculated using timer frequency.. 48.00 BogoMIPS (lpj=240000)
[    0.000465] pid_max: default: 32768 minimum: 301
[    0.000616] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.000630] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.001511] CPU: Testing write buffer coherency: ok
[    0.001943] CPU0: thread -1, cpu 0, socket 0, mpidr 80000000
[    0.002766] Setting up static identity map for 0x41300000 - 0x41300060
[    0.002944] rcu: Hierarchical SRCU implementation.
[    0.003432] smp: Bringing up secondary CPUs ...
[    0.004405] CPU1: thread -1, cpu 1, socket 0, mpidr 80000001
[    0.004600] smp: Brought up 1 node, 2 CPUs
[    0.004623] SMP: Total of 2 processors activated (96.00 BogoMIPS).
[    0.004630] CPU: All CPU(s) started in SVC mode.
[    0.012819] VFP support v0.3: implementor 51 architecture 3 part 40 variant 3 rev 4
[    0.013225] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[    0.013253] futex hash table entries: 512 (order: 3, 32768 bytes, linear)
[    0.013418] pinctrl core: initialized pinctrl subsystem
[    0.014533] NET: Registered protocol family 16
[    0.016018] DMA: preallocated 2048 KiB pool for atomic coherent allocations
[    0.017257] cpuidle: using governor ladder
[    0.017313] cpuidle: using governor menu
[    0.017544] NET: Registered protocol family 42
[    0.017896] hw-breakpoint: found 5 (+1 reserved) breakpoint and 4 watchpoint registers.
[    0.017909] hw-breakpoint: maximum watchpoint size is 8 bytes.
[    0.060054] qcom,gcc-ipq5018 1800000.gcc: Registered ipq5018 GCC clocks provider
[    0.074314] msm_sps_probe: sps:sps is ready
[    0.103037] pps_core: LinuxPPS API ver. 1 registered
[    0.103049] pps_core: Software ver. 5.3.6 - Copyright 2005-2007 Rodolfo Giometti <giometti@linux.it>
[    0.103070] PTP clock support registered
[... 305 lines cut ...]
[   48.755555] br-lan: port 1(eth1.1) entered blocking state
[   48.755595] br-lan: port 1(eth1.1) entered forwarding state
[   49.493142] pppoe-wan: renamed from ppp0
[   49.766020] IPv6: ADDRCONF(NETDEV_CHANGE): br-lan: link becomes ready
[   50.889977] wlan0: Created a NSS virtual interface
[   50.898534] br-lan: port 2(wlan0) entered blocking state
[   50.898569] br-lan: port 2(wlan0) entered disabled state
[   50.916641] device wlan0 entered promiscuous mode
[   50.917891] br-lan: port 2(wlan0) entered blocking state
[   50.920379] br-lan: port 2(wlan0) entered forwarding state
[... cut ...]
[   51.068137] br-lan: port 3(wlan1) entered blocking state
[   51.068179] br-lan: port 3(wlan1) entered forwarding state
[   51.526603] ath11k soc:wifi1@c000000: bss color param 0xb000000 set on vdev 0
[   51.526744] IPv6: ADDRCONF(NETDEV_CHANGE): wlan1: link becomes ready
[   52.856086] ath11k c000000.wifi: bss color param 0xe000000 set on vdev 0
[   52.856223] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
[   52.862976] br-lan: port 2(wlan0) entered blocking state
[   52.868274] br-lan: port 2(wlan0) entered forwarding state
```

zhangbaichang, 2023-08-08:
这是系统日志:
```
Mon Aug  7 06:59:53 2023 kern.info kernel: [    0.000000] Booting Linux on physical CPU 0x0
Mon Aug  7 06:59:53 2023 kern.notice kernel: [    0.000000] Linux version 5.4.164 (runner@fv-az457-764) (gcc version 8.4.0 (OpenWrt GCC 8.4.0 ci-20230716-155109-a41da3c9)) #0 SMP PREEMPT Sun Jul 16 15:48:41 2023
Mon Aug  7 06:59:53 2023 kern.info kernel: [    0.000000] CPU: ARMv7 Processor [51af8014] revision 4 (ARMv7), cr=10c0383d
Mon Aug  7 06:59:53 2023 kern.info kernel: [    0.000000] CPU: div instructions available: patching division code
Mon Aug  7 06:59:53 2023 kern.info kernel: [    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
Mon Aug  7 06:59:53 2023 kern.info kernel: [    0.000000] OF: fdt: Machine model: Redmi AX3000
Mon Aug  7 06:59:53 2023 kern.info kernel: [    0.000000] Memory policy: Data cache writealloc
Mon Aug  7 06:59:53 2023 kern.debug kernel: [    0.000000] On node 0 totalpages: 46336
Mon Aug  7 06:59:53 2023 kern.debug kernel: [    0.000000]   Normal zone: 576 pages used for memmap
Mon Aug  7 06:59:53 2023 kern.debug kernel: [    0.000000]   Normal zone: 0 pages reserved
Mon Aug  7 06:59:53 2023 kern.debug kernel: [    0.000000]   Normal zone: 46336 pages, LIFO batch:15
Mon Aug  7 06:59:53 2023 kern.info kernel: [    0.000000] psci: probing for conduit method from DT.
Mon Aug  7 06:59:53 2023 kern.info kernel: [    0.000000] psci: PSCIv1.0 detected in firmware.
Mon Aug  7 06:59:53 2023 kern.info kernel: [    0.000000] psci: Using standard PSCI v0.2 function IDs
Mon Aug  7 06:59:53 2023 kern.info kernel: [    0.000000] psci: MIGRATE_INFO_TYPE not supported.
Mon Aug  7 06:59:53 2023 kern.info kernel: [    0.000000] psci: SMC Calling Convention v1.0
Mon Aug  7 06:59:53 2023 kern.info kernel: [    0.000000] psci: OSI mode supported.
Mon Aug  7 06:59:53 2023 kern.info kernel: [    0.000000] percpu: Embedded 15 pages/cpu s32204 r8192 d21044 u61440
Mon Aug  7 06:59:53 2023 kern.debug kernel: [    0.000000] pcpu-alloc: s32204 r8192 d21044 u61440 alloc=15*4096
Mon Aug  7 06:59:53 2023 kern.debug kernel: [    0.000000] pcpu-alloc: [0] 0 [0] 1
Mon Aug  7 06:59:53 2023 kern.info kernel: [    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 45760
Mon Aug  7 06:59:53 2023 kern.notice kernel: [    0.000000] Kernel command line: console=ttyMSM0,115200n8 ubi.mtd=rootfs root=mtd:ubi_rootfs rootfstype=squashfs rootwait root=/dev/ubiblock0_1 coherent_pool=2M
Mon Aug  7 06:59:53 2023 kern.info kernel: [    0.000000] Dentry cache hash table entries: 32768 (order: 5, 131072 bytes, linear)
Mon Aug  7 06:59:53 2023 kern.info kernel: [    0.000000] Inode-cache hash table entries: 16384 (order: 4, 65536 bytes, linear)
Mon Aug  7 06:59:53 2023 kern.info kernel: [    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
Mon Aug  7 06:59:53 2023 kern.info kernel: [    0.000000] Memory: 172204K/185344K available (6894K kernel code, 262K rwdata, 2088K rodata, 1024K init, 360K bss, 13140K reserved, 0K cma-reserved, 0K highmem)
Mon Aug  7 06:59:53 2023 kern.info kernel: [    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=2, Nodes=1
Mon Aug  7 06:59:53 2023 kern.info kernel: [    0.000000] rcu: Preemptible hierarchical RCU implementation.
Mon Aug  7 06:59:53 2023 kern.info kernel: [    0.000000] 	Tasks RCU enabled.
Mon Aug  7 06:59:53 2023 kern.info kernel: [    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 10 jiffies.
Mon Aug  7 06:59:53 2023 kern.info kernel: [    0.000000] NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
Mon Aug  7 06:59:53 2023 kern.info kernel: [    0.000000] GICv2m: range[mem 0x0b00a000-0x0b00affc], SPI[448:479]
Mon Aug  7 06:59:53 2023 kern.info kernel: [    0.000000] GICv2m: range[mem 0x0b00b000-0x0b00bffc], SPI[480:511]
Mon Aug  7 06:59:53 2023 kern.notice kernel: [    0.000000] random: get_random_bytes called from start_kernel+0x344/0x4f4 with crng_init=0
Mon Aug  7 06:59:53 2023 kern.info kernel: [    0.000000] arch_timer: cp15 and mmio timer(s) running at 24.00MHz (virt/virt).
Mon Aug  7 06:59:53 2023 kern.info kernel: [    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0x588fe9dc0, max_idle_ns: 440795202592 ns
Mon Aug  7 06:59:53 2023 kern.info kernel: [    0.000006] sched_clock: 56 bits at 24MHz, resolution 41ns, wraps every 4398046511097ns
Mon Aug  7 06:59:53 2023 kern.info kernel: [    0.000019] Switching to timer-based delay loop, resolution 41ns
Mon Aug  7 06:59:53 2023 kern.info kernel: [    0.000417] Console: colour dummy device 80x30
Mon Aug  7 06:59:53 2023 kern.info kernel: [    0.000449] Calibrating delay loop (skipped), value calculated using timer frequency.. 48.00 BogoMIPS (lpj=240000)
Mon Aug  7 06:59:53 2023 kern.info kernel: [    0.000465] pid_max: default: 32768 minimum: 301
Mon Aug  7 06:59:53 2023 kern.info kernel: [    0.000616] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
Mon Aug  7 06:59:53 2023 kern.info kernel: [    0.000630] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
Mon Aug  7 06:59:53 2023 kern.info kernel: [    0.001511] CPU: Testing write buffer coherency: ok
Mon Aug  7 06:59:53 2023 kern.info kernel: [    0.001943] CPU0: thread -1, cpu 0, socket 0, mpidr 80000000
Mon Aug  7 06:59:53 2023 kern.info kernel: [    0.002766] Setting up static identity map for 0x41300000 - 0x41300060
[... cut ...]
Tue Aug  8 16:40:49 2023 daemon.notice hostapd: wlan0: STA 60:7e:a4:42:df:24 IEEE 802.11: did not acknowledge authentication response
Tue Aug  8 16:40:53 2023 daemon.notice hostapd: wlan0: STA 60:7e:a4:42:df:24 IEEE 802.11: did not acknowledge authentication response
Tue Aug  8 16:40:54 2023 user.info : luci: accepted login on /admin/system/flash for root from 192.168.123.142
Tue Aug  8 16:40:59 2023 daemon.notice hostapd: wlan0: STA 60:7e:a4:42:df:24 IEEE 802.11: did not acknowledge authentication response
```

zhangbaichang, 2023-08-08:
目前似乎还没有人适配小米ax3000的WiFi的bdf及校准文件，看您为红米ax3000的老版本(2022.12左右)是可以在小米ax3000上正常运行的，还请作者看看是修改哪些文件造成的WiFi不可用

zhangbaichang, 2023-08-08:
WiFi拒绝加入的情况具体是手机尝试加入5G频段WiFi，在未连接成功时断开，多次循环后显示拒绝连接；而2.4G频段可连接，但经过约20秒后会丢失连接

zhangbaichang, 2023-08-09:
发现问题了，负载爆了
[image]
top输出如下：
[image]
没看到哪个进程占用了大量资源，请问如何解决

hzyitc, 2023-10-23:
抱歉，之前有点忙，给忘了。

用最新的包试试吧

897180673, 2023-10-30:
@zhangbaichang  老哥 你是拆机 串口刷的机么?