hzyitc/openwrt-redmi-ax3000 #41: wi-fi not working on my black cylinder AX3000, please help

beelabcloud, 2023-10-27:
The Dec 30, 2022 version "openwrt-ipq50xx-generic-redmi_ax3000-squashfs-nand-factory.ubi" does. None others could drive wi-fi properly. The kernel-4.4 version released a few days ago cannot even boot up. It's possible to get a copy of source code of the Dec 30 2022. Thanks for the great project. By the way, what is bdf files, it's possible to get it from miwifi firmware, I could roll it back to the stock.

beelabcloud, 2023-10-27:
[mtd18_rootfs.zip](https://github.com/hzyitc/openwrt-redmi-ax3000/files/13189938/mtd18_rootfs.zip) Here is backup of the stock rootfs.

beelabcloud, 2023-10-28:
Here is the log of working one (openwrt-ipq50xx-generic-redmi_ax3000-squashfs-nand-factory.ubi)

[   19.199492] ath11k c000000.wifi: qmi failed to load QDSS config: qdss_trace_config.bin
[   19.206759] ath11k c000000.wifi: Failed to download QDSS config to FW: -11
[   45.645931] remoteproc remoteproc2: stopped remote processor q6v5_wcss_userpd2
[   45.645971] remoteproc remoteproc2: powering up q6v5_wcss_userpd2
[   45.667942] Subsystem error monitoring/handling services are up
[   45.668013] remoteproc remoteproc2: remote processor q6v5_wcss_userpd2 is now up
[   45.676436] kmodloader: done loading kernel modules from /etc/modules.d/*
[   45.745105] ath11k soc:wifi1@c000000: chip_id 0x0 chip_family 0x4 board_id 0x60 soc_id 0xffffffff
[   45.745146] ath11k soc:wifi1@c000000: fw_version 0x250a04b8 fw_build_timestamp 2021-12-20 07:51 fw_build_id QC_IMAGE_VERSION_STRING=WLAN.HK.2.5.0.1-01208-QCAHKSWPL_SILICONZ-1
[   45.757552] ath11k soc:wifi1@c000000: qmi caldata downloaded: type: 2
[   45.991713] ieee80211 phy1: copying sband (band 1) due to VHT EXT NSS BW flag
[   45.994504] ath11k soc:wifi1@c000000: Direct firmware load for ath11k/qcn6122/hw1.0/qdss_trace_config.bin failed with error -2
[   45.994544] ath11k soc:wifi1@c000000: Falling back to user helper
[   46.372115] firmware ath11k!qcn6122!hw1.0!qdss_trace_config.bin: firmware_loading_store: map pages failed
[   46.372502] ath11k soc:wifi1@c000000: Direct firmware load for qdss_trace_config.bin failed with error -2
[   46.381473] ath11k soc:wifi1@c000000: Falling back to user helper
[   46.531194] firmware qdss_trace_config.bin: firmware_loading_store: map pages failed
[   46.531527] ath11k soc:wifi1@c000000: qmi failed to load QDSS config: qdss_trace_config.bin
[   46.540834] ath11k soc:wifi1@c000000: Failed to download QDSS config to FW: -11

hzyitc, 2023-10-28:
2022? Maybe what you use is not my firmware.

All of my releases have commit id records in the Github release.

beelabcloud, 2023-10-28:
[mistock.zip](https://github.com/hzyitc/openwrt-redmi-ax3000/files/13195502/mistock.zip) Here is a copy of  Mi stock /lib/firmware, /ini, /etc

beelabcloud, 2023-10-28:
I'm pretty sure that "openwrt-ipq50xx-generic-redmi_ax3000-squashfs-nand-factory.ubi" was downloaded from this project somewhere (not in the releases), the mod time of the file is Dec 30, 2022.

beelabcloud, 2023-10-28:
@hzyitc, Thank you very much, I wish I would have a chance to buy you a beer. Where are you, I may deliver a black Mi ax3000 to you.

hzyitc, 2023-10-28:
You can try to copy the BDF file.

My firmware: /lib/firmware/ath11k/*/board-2.bin
QSDK: /lib/firmware/*/bdwlan.b*

The extension is the board id, which you can find in boot log.

hzyitc, 2023-10-28:
> @hzyitc, Thank you very much, I wish I would have a chance to buy you a beer. Where are you, I may deliver a black Mi ax3000 to you.

Oh, thank you. But it's unnecessary for now. The different between Xiaomi AX3000 and Redmi AX3000 should be very small. Maybe just GPIO and BDF. The GPIO can be got by the origin dts. The BDF need some tests, but I can send it to you.

beelabcloud, 2023-10-28:
Here is the dmesg log of MI stock rom.
[  121.818845] cnss[41]: INFO: Waiting for FW ready. Device: 0xfffb, FW ready timeout: 15 seconds
[  121.870978] cnss[41]: INFO: QMI WLFW service connected, state: 0x1
[  121.871295] cnss[41]: INFO: device_id : 65531 mem mode : [2]
[  121.876051] cnss[41]: INFO: platform name: Qualcomm Technologies, Inc. IPQ5018/AP-MP02.1
[  121.883130] cnss[41]: INFO: Target capability: chip_id: 0x0, chip_family: 0x4, board_id: 0xff, soc_id: 0xffffffff, fw_version: 0x2504ffff, fw_build_timestamp: 2021-07-09 07:01, otp_version: 0x0 eeprom_caldata_read_timeout 0s
[  121.891071] cnss[41]: INFO: Device BAR Info pa: 0x81e00000, va: 0x94f80000, size: 0x200000
[  121.910606] cnss[41]: INFO: Using Boardid from bootargs:0x60
[  121.929379] cnss[41]: INFO: BDF location : 0x4d300000
[  121.929409] cnss[41]: INFO: BDF qcn6122/bdwlan.b60 size 131072
[  121.934702] cnss[41]: INFO: per device BDF location : 0x4d320000
[  121.939655] cnss[41]: INFO: CALDATA qcn6122/caldata_1.bin size 131072 offset 0x20000
[  122.038083] cnss[41]: INFO: FW ready received for device 0xfffb
[  122.038276] wlan: [2631:I:ANY] pld_ahb_wlan_enable: cnss_mode: Mission mode
[  122.043227] cnss[41]: INFO: Sending mode message, mode: MISSION(0), state: 0x7
[  122.060942] wlan: [2631:I:ANY] qca_napi_create: qca_napi_create: napi instance 0 created on pipe 4
[  122.061010] wlan: [2631:I:ANY] htc_wmi_init: HT Create . 86354000
[  122.070007] wlan: [2631:I:ANY] htc_wmi_init: Skipping BMI Done
[  122.074953] wlan: [2631:I:ANY] htc_wmi_init: host_enable 0
[  122.090506] wlan: [2631:I:ANY] ol_target_init_complete: CDP soc attach success
[  122.090542] wlan: [2631:I:ANY] ol_target_init_complete: Soc attach success NSS config 0
[  122.096742] wlan: [2631:I:ANY] htc_set_async_ep: htc_set_async_ep: htc_handle 86354000, ep 2, value 1
[  122.106374] wlan: [2390:I:TIF] HE_CAP Info = 9a08010d
[  122.115160] wlan: [2390:I:TIF] Num HW modes = 1
[  122.119177] wlan: [2390:I:TIF] Num PHY = 1
[  122.123761] wlan: [2390:I:TIF] FW Build Version (Ext.) = 0
[  122.127853] wlan: [2390:I:TIF] HDL Version: MajorVersion.MinorVersion = 0.0
[  122.134873] wlan: [2390:I:TIF] Preferred HW Mode = 0 Num Radios = 1
[  122.140087] wlan: [2390:W:TIF] add_11ax_mode_flags: 11AX wireless modes are not populated by the target in reg capabilities, adding 11AX wireless modes
[  122.146107] wlan: [2390:I:TIF] init_deinit_add_11ax_modes: hw_mode_id = 0 phy_id = 0 wireless modes = 0
[  122.159905] wlan: [2390:I:TIF] init_deinit_get_total_num_ema_vaps: num_radios: 1
[  122.169201] wlan: [2390:I:TIF] init_deinit_get_total_num_ema_vaps: ema_max_vap_cnt: 0

hzyitc, 2023-10-28:
As you can see: qcn6122/bdwlan.b60

And there is an other one for IPQ5018, which is not shown in your log.

beelabcloud, 2023-10-28:
Yes! 
2021-07-15T20:24:55+08:00 XiaoQiang kernel: [  113.331421] cnss[2]: INFO: QMI WLFW service connected, state: 0x1
2021-07-15T20:24:55+08:00 XiaoQiang kernel: [  113.331716] cnss[2]: INFO: device_id : 65532 mem mode : [2]
2021-07-15T20:24:55+08:00 XiaoQiang kernel: [  113.332863] cnss[2]: INFO: platform name: Qualcomm Technologies, Inc. IPQ5018/AP-MP02.1
2021-07-15T20:24:55+08:00 XiaoQiang kernel: [  113.332864] cnss[2]: INFO: Target capability: chip_id: 0x0, chip_family: 0x4, board_id: 0xff, soc_id: 0xffffffff, fw_version: 0x2504ffff, fw_build_timestamp: 2021-07-09 07:01, otp_version: 0x0 eeprom_caldata_read_timeout 0s
2021-07-15T20:24:55+08:00 XiaoQiang kernel: [  113.332877] cnss[2]: INFO: Using Boardid from bootargs:0x24
2021-07-15T20:24:55+08:00 XiaoQiang kernel: [  113.349523] cnss[2]: INFO: BDF location : 0x4c400000
2021-07-15T20:24:55+08:00 XiaoQiang kernel: [  113.349532] cnss[2]: INFO: BDF IPQ5018/bdwlan.b24 size 131072
2021-07-15T20:24:55+08:00 XiaoQiang kernel: [  113.350557] cnss[2]: INFO: per device BDF location : 0x4c420000
2021-07-15T20:24:55+08:00 XiaoQiang kernel: [  113.350699] cnss[2]: INFO: CALDATA IPQ5018/caldata.bin size 131072 offset 0x20000
2021-07-15T20:24:55+08:00 XiaoQiang kernel: [  113.399465] remoteproc remoteproc1: remote processor q6v5_wcss_userpd1 is now up
2021-07-15T20:24:55+08:00 XiaoQiang kernel: [  113.407176] wlan: [1298:I:ANY] ol_ath_wifi_ssr: ol_ath_wifi_ssr: SSR event 3 bus_type 1
2021-07-15T20:24:55+08:00 XiaoQiang kernel: [  113.414689] wlan: [1298:I:ANY] ol_ath_target_start: subsystem_get success
2021-07-15T20:24:55+08:00 XiaoQiang kernel: [  113.422357] cnss[2]: INFO: Waiting for FW ready. Device: 0xfffc, FW ready timeout: 15 seconds
2021-07-15T20:24:55+08:00 XiaoQiang kernel: [  113.429287] cnss[2]: INFO: FW ready received for device 0xfffc
2021-07-15T20:24:55+08:00 XiaoQiang kernel: [  113.438322] wlan: [1298:I:ANY] pld_ahb_wlan_enable: cnss_mode: Mission mode 
2021-07-15T20:24:55+08:00 XiaoQiang kernel: [  113.443773] cnss[2]: INFO: Sending mode message, mode: MISSION(0), state: 0x7
2021-07-15T20:24:55+08:00 XiaoQiang kernel: [  113.455812] wlan: [1298:I:ANY] qca_napi_create: qca_napi_create: napi instance 0 created on pipe 4
2021-07-15T20:24:55+08:00 XiaoQiang kernel: [  113.457731] wlan: [1298:I:ANY] htc_wmi_init: HT Create . 876b9000
2021-07-15T20:24:55+08:00 XiaoQiang kernel: [  113.466650] wlan: [1298:I:ANY] htc_wmi_init: Skipping BMI Done
2021-07-15T20:24:55+08:00 XiaoQiang kernel: [  113.472791] wlan: [1298:I:ANY] htc_wmi_init: host_enable 0
2021-07-15T20:24:55+08:00 XiaoQiang kernel: [  113.487625] wlan: [1298:I:ANY] ol_target_init_complete: CDP soc attach success
2021-07-15T20:24:55+08:00 XiaoQiang kernel: [  113.487658] wlan: [1298:I:ANY] ol_target_init_complete: Soc attach success NSS config 0 
2021-07-15T20:24:55+08:00 XiaoQiang kernel: [  113.494062] wlan: [1298:I:ANY] htc_set_async_ep: htc_set_async_ep: htc_handle 876b9000, ep 2, value 1
2021-07-15T20:24:55+08:00 XiaoQiang kernel: [  113.503162] wlan: [408:I:TIF] HE_CAP Info = 9a08010d
2021-07-15T20:24:55+08:00 XiaoQiang kernel: [  113.511202] wlan: [408:I:TIF] Num HW modes = 1
2021-07-15T20:24:55+08:00 XiaoQiang kernel: [  113.516157] wlan: [408:I:TIF] Num PHY = 1
2021-07-15T20:24:55+08:00 XiaoQiang kernel: [  113.520472] wlan: [408:I:TIF] FW Build Version (Ext.) = 0
2021-07-15T20:24:55+08:00 XiaoQiang kernel: [  113.524512] wlan: [408:I:TIF] HDL Version: MajorVersion.MinorVersion = 0.0
2021-07-15T20:24:55+08:00 XiaoQiang kernel: [  113.529957] wlan: [408:I:TIF] Preferred HW Mode = 0 Num Radios = 1

beelabcloud, 2023-10-28:
[messages_dmesg.zip](https://github.com/hzyitc/openwrt-redmi-ax3000/files/13196182/messages_dmesg.zip)
here is full system log.

beelabcloud, 2023-10-29:
Here is the lastest stock rom dump: mtd, messages, dmesg, firmware, and caldata.bin.
[mistock1.0.46.zip](https://github.com/hzyitc/openwrt-redmi-ax3000/files/13198076/mistock1.0.46.zip)

beelabcloud, 2023-10-29:
@hzyitc, does Mi use the open driver in QSDK? Can I replace the bdf files by the stock's, would it damage the hardware?

beelabcloud, 2023-10-29:
Ok, I did bdf replace, bdwlan.b24 -->board-2.bin, caldata.bin -->caldata.bin,  bdwlan.b60 -->board-2.bin(inqcn6122), caldata_1.bin-->caldata_1.bin, system told that "invalid board magic"

   16.301197] ath11k c000000.wifi: fw_version 0x260a83ce fw_build_timestamp 2022-06-21 02:06 fw_build_id QC_IMAGE_VERSION_STRING=WLAN.HK.2.6.0.1-00974-QCAHKSWPL_SILICONZ-1
[   16.320592] ath11k c000000.wifi: **found invalid board magic**
[   16.415602] ath11k c000000.wifi: failed to fetch board-2.bin or board.bin from IPQ5018/hw1.0
[   16.415648] ath11k c000000.wifi: qmi failed to fetch bdf
[   16.423333] ath11k c000000.wifi: qmi failed to load board data file:-12
[   16.426441] remoteproc remoteproc1: remote processor 4ab000.remoteproc_pd1 is now up
[   79.846356] ath11k c000000.wifi: Coldboot Calibration timed out
[   81.926368] ath11k soc:wifi1@c000000: qcn6122
[   81.926773] remoteproc remoteproc2: powering up cd00000.remoteproc:remoteproc_pd2
[   81.929860] remoteproc remoteproc2: Booting fw image IPQ5018/q6_fw.mdt, size 1820
[   81.940044] Subsystem error monitoring/handling services are up
[   82.016167] ath11k soc:wifi1@c000000: qmi ignore invalid mem req type 10
[   82.016918] ath11k soc:wifi1@c000000: chip_id 0x0 chip_family 0x4 board_id 0x60 soc_id 0xffffffff
[   82.021967] ath11k soc:wifi1@c000000: fw_version 0x260a83ce fw_build_timestamp 2022-06-21 02:06 fw_build_id QC_IMAGE_VERSION_STRING=WLAN.HK.2.6.0.1-00974-QCAHKSWPL_SILICONZ-1
[   82.041473] ath11k soc:wifi1@c000000: found invalid board magic
[   82.056438] remoteproc remoteproc2: remote processor cd00000.remoteproc:remoteproc_pd2 is now up
[   82.094630] ath11k soc:wifi1@c000000: failed to fetch board-2.bin or board.bin from qcn6122/hw1.0
[   82.094681] ath11k soc:wifi1@c000000: qmi failed to fetch bdf
[   82.102688] ath11k soc:wifi1@c000000: qmi failed to load board data file:-12
[  146.406358] ath11k soc:wifi1@c000000: Coldboot Calibration timed out

hzyitc, 2023-10-30:
> Ok, I did bdf replace, bdwlan.b24 -->board-2.bin, caldata.bin -->caldata.bin, bdwlan.b60 -->board-2.bin(inqcn6122), caldata_1.bin-->caldata_1.bin, system told that "invalid board magic"
>
> 16.301197] [...]

Oh, you need to use https://github.com/qca/qca-swiss-army-knife to pack it as board-2.bin

hzyitc, 2023-10-30:
> @hzyitc, Wonderful, it works like a charmm using the oldest driver from mi stock, but no 2.4G. The 5G streaming is around 90MByte/S. Signal strength seems a little bit weak. I'll do more test later. And it reports lots of debug, like: [ 1140.178581] ath11k_warn: 842 callbacks suppressed [ 1140.178599] ath11k soc:wifi1@c000000: Invalid len 36 for the tag 0x1 [ 1140.182278] ath11k soc:wifi1@c000000: Invalid len 76 for the tag 0x6 How can I suppress the "842 callbacks suppressed" warning? There is no codes matching "callbacks suppressed" pattern.

oh, you don't need to rollback the firmware. There are some compatible problem between fw 2.5 and linux 5.4.

`842 callbacks suppressed` was printed by log_limit. Just because lots of `Invalid len 76 for the tag 0x6`

beelabcloud, 2023-10-30:
Mi didn't ship driver above 2.5.0.1 yet. The latest one is version 0x2509ffff. There are many gpio warnings on starting, it reports "failed on cold starting calibration" and wifi would become inactivated on a hot restart. What should I do next to make it better?.   
2023-01-30T15:21:58+08:00 XiaoQiang kernel: [   48.471081] cnss[41]: INFO: platform name: Qualcomm Technologies, Inc. IPQ5018/AP-MP02.1
2023-01-30T15:21:58+08:00 XiaoQiang kernel: [   48.479683] cnss[41]: INFO: Target capability: chip_id: 0x0, chip_family: 0x4, board_id: 0xff, soc_id: 0xffffffff, fw_version: 0x2509ffff, fw_build_timestamp: 2021-10-19 23:11, otp_version: 0x0 eeprom_caldata_read_timeout 0s

beelabcloud, 2023-10-30:
Here is the part of log,
[   15.729975] ath11k c000000.wifi: chip_id 0x0 chip_family 0x4 board_id 0x24 soc_id 0xffffffff
[   15.731303] ath11k c000000.wifi: fw_version 0x2504ffff fw_build_timestamp 2021-07-09 07:01 fw_build_id QC_IMAGE_VERSION_STRING=WLAN.HK.2.5.r4-00683-QCAHKSWPL_SILICONZ-1.414325.6
[   15.748910] ath11k c000000.wifi: failed to fetch board data for bus=ahb,qmi-chip-id=0,qmi-board-id=36 from ath11k/IPQ5018/hw1.0/board-2.bin
[   15.817779] ath11k c000000.wifi: failed to fetch board-2.bin or board.bin from IPQ5018/hw1.0
[   15.817827] ath11k c000000.wifi: qmi failed to fetch bdf
[   15.825360] ath11k c000000.wifi: qmi failed to load board data file:-12
[   15.846549] remoteproc remoteproc1: remote processor 4ab000.remoteproc_pd1 is now up
[   79.846479] ath11k c000000.wifi: Coldboot Calibration timed out
[   81.926462] ath11k soc:wifi1@c000000: qcn6122
[   81.926888] remoteproc remoteproc2: powering up cd00000.remoteproc:remoteproc_pd2
[   81.929972] remoteproc remoteproc2: Booting fw image IPQ5018/q6_fw.mdt, size 1820
[   81.940186] Subsystem error monitoring/handling services are up
[   82.016775] ath11k soc:wifi1@c000000: chip_id 0x0 chip_family 0x4 board_id 0x60 soc_id 0xffffffff
[   82.016807] ath11k soc:wifi1@c000000: fw_version 0x2504ffff fw_build_timestamp 2021-07-09 07:01 fw_build_id QC_IMAGE_VERSION_STRING=WLAN.HK.2.5.r4-00683-QCAHKSWPL_SILICONZ-1.414325.6
[   82.039380] ath11k soc:wifi1@c000000: qmi caldata downloaded: type: 2
[   82.056532] remoteproc remoteproc2: remote processor cd00000.remoteproc:remoteproc_pd2 is now up
[  106.283963] remoteproc remoteproc2: stopped remote processor cd00000.remoteproc:remoteproc_pd2
[  106.283998] remoteproc remoteproc2: powering up cd00000.remoteproc:remoteproc_pd2
[  106.291645] remoteproc remoteproc2: Booting fw image IPQ5018/q6_fw.mdt, size 1820
[  106.312001] Subsystem error monitoring/handling services are up
[  106.388689] ath11k soc:wifi1@c000000: chip_id 0x0 chip_family 0x4 board_id 0x60 soc_id 0xffffffff
[  106.388722] ath11k soc:wifi1@c000000: fw_version 0x2504ffff fw_build_timestamp 2021-07-09 07:01 fw_build_id QC_IMAGE_VERSION_STRING=WLAN.HK.2.5.r4-00683-QCAHKSWPL_SILICONZ-1.414325.6
[  106.399939] ath11k soc:wifi1@c000000: qmi caldata downloaded: type: 2
[  106.426498] remoteproc remoteproc2: remote processor cd00000.remoteproc:remoteproc_pd2 is now up
[  106.427345] kmodloader: done loading kernel modules from /etc/modules.d/*
[  106.617551] ieee80211 phy0: copying sband (band 1) due to VHT EXT NSS BW flag
[  106.620174] ath11k soc:wifi1@c000000: Direct firmware load for ath11k/qcn6122/hw1.0/qdss_trace_config.bin failed with error -2
[  106.620224] ath11k soc:wifi1@c000000: Falling back to sysfs fallback for: ath11k/qcn6122/hw1.0/qdss_trace_config.bin
[  107.061882] ath11k soc:wifi1@c000000: Direct firmware load for qdss_trace_config.bin failed with error -2
[  107.061924] ath11k soc:wifi1@c000000: Falling back to sysfs fallback for: qdss_trace_config.bin
[  107.280806] ath11k soc:wifi1@c000000: qmi failed to load QDSS config: qdss_trace_config.bin
[  107.280851] ath11k soc:wifi1@c000000: Failed to download QDSS config to FW: -12

beelabcloud, 2023-10-30:
@hzyitc, it takes about 1 minute to do coldboot calibration break out on timed out? Am I using a wrong caldata.bin?

hzyitc, 2023-10-30:
Firmware and BDF has forwards/backwards compatibility. So you may be able to use the legacy BDF with new firmware.

And BDF contains lots of board level configs, so there is no place to copy from. Just some demos here: https://git.codelinaro.org/clo/qsdk/oss/ath11k-bdf

hzyitc, 2023-10-30:
> Where can I get a copy of bdwlan.b24 for version 2.6.0.1. There is no such file in https://github.com/quic/upstream-wifi-fw. One more question, I did comment out " Invalid len" in the patches, and re-make after "make clean", "Invalid len 76 for the tag 0x6" still there. Thanks.

Tag 0x1 and Tag 0x6 use different handler. So you need to comment out two places. But I don't suggest to do it. Removing the exceed fields is a better idea.

beelabcloud, 2023-10-31:
@hzyitc, I made a wrong conculation. What I saw yestoday is magic somehow. Doing bdf files subtitution can only bring up the wi-fi barely, no matter coldboot or hotboot, or odd-even tests, it only works randomly base more tests I have done today. So it's still in wrong way. Or I might make some mistakes on assembling board-2.bin file. Do you have a working "board-2.bin". Thanks anyway.

beelabcloud, 2023-10-31:
The FW filed in MI latest stock rom (version 1.0.46) is still at 2.5.0.1. Is it possible to get FW2.6 from Redmi stock rom? Or you have one, please give me a copy.  Seriously, I own you a dinner, text me whenever you stop at Beijing.

hzyitc, 2023-11-02:
You can try the public FW. It should work. What make it bad is BDF (TX power, FEM config, ...).

https://github.com/hzyitc/openwrt-redmi-ax3000/blob/ipq50xx-mainline-kernel-5.15-openwrt-23.05/package/firmware/ath11k-firmware/Makefile

closed 2023-11-07 by beelabcloud