hzyitc/openwrt-redmi-ax3000 #53: 同一个SSID下互ping问题

xuoogle, 2024-02-04:
作者你好, 直接用你的固件刷cr8809, **连接同一个SSID的客户端之间**不能互ping
下面情况可以互通
1. 一个连接2.4G, 另一个连接5G.
2. 经快速漫游切换回同一个SSID. (手动断开重连仍然不通)
尝试自编译23.05版本没有这个问题, 但是下载速度只有百兆, 上传正常.

hzyitc, 2024-03-03:
5.4是指qsdk-5.4还是mainline-5.4？

Kissycat, 2024-03-04:
没错，很离谱，用其他AP也是正常的

kettly1260, 2024-03-04:
> 5.4是指qsdk-5.4还是mainline-5.4？

我用的也是5.4，两个问题都是存在的

hzyitc, 2024-04-05:
应该是AP客户端隔离被开启了，等有空再查吧。

nopcall, 2024-07-08:
我也遇到类似问题。我想adb连接手机时发现连接不上。然后还ping不通

yuanchanglong1997, 2024-10-11:
同遇到这个问题，都连接5gwifi频段，手机电脑无法ping通，电脑连网线又可以了，连wifi不行

Kissycat, 2024-12-17:
> [openwrt#8159](https://github.com/openwrt/openwrt/issues/8159)

好用了！
```
root@CR8808:~# cat /etc/config/network

...

config device
        option name 'br-lan'
        option type 'bridge'
        list ports 'eth0'
        option bridge_empty '1'
+++     option multicast_to_unicast '0'

...
```

Miuzarte, 2024-12-31:
pin一下这个issue吧，找了半天问题就没想过是上游的锅

dcto, 2025-03-29:
艹，干了三天，才找到原因

distance99, 2025-04-02:
我也遇到了这个问题。电脑和手机都连接的5GHz频段的 WiFi ，电脑ping 手机，使用WireShark 抓包显示手机没有没有响应ARP，以至于电脑一直重复发送。

[image]

ArkURL, 2025-07-31:
刷了旧版本的OpenWrt，今天调试时发现这个问题了，感谢