hzyitc/openwrt-redmi-ax3000 #59: ipq50xx: Support for IPQ5018 MP03.5-c1

kimocoder, 2024-03-02:
Hi there!

I wonder if anyone could assist me in adding support for "ipq50xx" ?

It's supported in "wlan-ap" as seen here
[https://github.com/Telecominfraproject/wlan-ap/blob/main/feeds/ipq807x/ipq807x/image/ipq50xx.mk#L58](https://github.com/Telecominfraproject/wlan-ap/blob/main/feeds/ipq807x/ipq807x/image/ipq50xx.mk#L58)

hzyitc, 2024-03-12:
`mp03.5-c1` is the demo board of Qualcomm. Due to the lots of hardcoding checks in the QSDK, most devices still kept this name.

So which device did you actually want to support? And the detailed specifications ?

kimocoder, 2024-03-24:
@hzyitc I've got 2 devices from PUSR dot running IPQ5018 MP03.5-c1 boards, they run OpenWRT, but its an anciend v15 (Chaos Chalmer) which I wanna get up-to-date, but dtb/dtsi is a bit over my head atm.

Give me a heads up if you got some time to throw at it. I may, externally give you full device access so most stuff should be available to put together (I guess) :+1:

hzyitc, 2024-04-05:
> @hzyitc I've got 2 devices from PUSR dot running IPQ5018 MP03.5-c1 boards, they run OpenWRT, but its an anciend v15 (Chaos Chalmer) which I wanna get up-to-date, but dtb/dtsi is a bit over my head atm.

yet, the qsdk 11.4 is based on openwrt 15

> Give me a heads up if you got some time to throw at it. I may, externally give you full device access so most stuff should be available to put together (I guess) 👍

Just provide the stock firmware and boot log.

kimocoder, 2024-04-07:
How may I pull stick firmware from a running environment?
Cheers!

hzyitc, 2024-06-23:
> How may I pull stock firmware from a running environment? Cheers!

cat /proc/mtd
dd if=/dev/mtdblockX of=/tmp/mtdX
pull them back.

TheRootED24, 2024-08-30:
Im looking into adding support for the GL-B3000 , based on the ipq5018 soc aswell. I've included a zip of what I've put together so far. I just got the unit a few days ago. Included the zip you'll find some full logs and the unpacked dtb from the unit.
[GL-B3000.zip](https://github.com/user-attachments/files/16809124/GL-B3000.zip)

kimocoder, 2024-08-30:
Sorry late answer, I'll be home in a few days, then I will provide you the logs @hzyitc  🫡

I got 4 routers with that board, now.
I made a IPQ5018 image with a Wally's IPQ5018 image, but I think I managed to brick when I installed to rootfs, so better ask to competent people who has done this before 👍 

I will get back with logs shortly, I used 'wlan-ap' which is a OpenWRT clone sort of, because I haven't seen this board inside OpenWRT yet, something I think is kinda weird, as there are quite some people who has it, and I got 2 different sets of them, 2x big boards with lots of space for modules, WiFI 6E, 5G QuecTel, and 2x mikro size boards with don't have space for anything other then those integrated 😀

kimocoder, 2024-09-03:
@hzyitc  @TheRootED24 

[dmesg.txt](https://github.com/user-attachments/files/16853528/dmesg.txt)

TheRootED24, 2024-09-06:
i've got the B3000 ready to test. I managed to get most of the qca-nss packages to compile ... after much tinkering.

```
FIT description: ARM64 OpenWrt FIT (Flattened Image Tree)
Created:         Fri Mar 22 22:09:42 2024
 Image 0 (kernel-1)
  Description:  ARM64 OpenWrt Linux-5.15.150
  Created:      Fri Mar 22 22:09:42 2024
  Type:         Kernel Image
  Compression:  lzma compressed
  Data Size:    10508292 Bytes = 10262.00 KiB = 10.02 MiB
  Architecture: AArch64
  OS:           Linux
  Load Address: 0x41000000
  Entry Point:  0x41000000
  Hash algo:    crc32
  Hash value:   e3fc7eae
  Hash algo:    sha1
  Hash value:   671ae32ab82a4841f237c24bb0735368cfdda7a3
 Image 1 (fdt-1)
  Description:  ARM64 OpenWrt glinet_gl-b3000 device tree blob
  Created:      Fri Mar 22 22:09:42 2024
  Type:         Flat Device Tree
  Compression:  uncompressed
  Data Size:    18639 Bytes = 18.20 KiB = 0.02 MiB
  Architecture: AArch64
  Hash algo:    crc32
  Hash value:   03989e54
  Hash algo:    sha1
  Hash value:   b48d28a32016a85bbafb46b5d403016eb6a94420
 Default Configuration: 'config-1'
 Configuration 0 (config-1)
  Description:  OpenWrt glinet_gl-b3000
  Kernel:       kernel-1
  FDT:          fdt-1
cp /home/hostle/openwrt-gl_inet-b3000/final/openwrt-redmi-ax3000/build_dir/target-aarch64_cortex-a53_musl/linux-ipq50xx_aarch64/tmp/openwrt-ipq50xx-aarch64-glinet_gl-b3000-initramfs-fit-uImage.itb /home/hostle/openwrt-gl_inet-b3000/final/openwrt-redmi-ax3000/bin/targets/ipq50xx/aarch64/openwrt-ipq50xx-aarch64-glinet_gl-b3000-initramfs-fit-uImage.itb
DEVICE_ID="glinet_gl-b3000" SOURCE_DATE_EPOCH=1711145382 FILE_NAME="openwrt-ipq50xx-aarch64-glinet_gl-b3000-initramfs-fit-uImage.itb" FILE_DIR="/home/hostle/openwrt-gl_inet-b3000/final/openwrt-redmi-ax3000/build_dir/target-aarch64_cortex-a53_musl/linux-ipq50xx_aarch64/tmp" FILE_TYPE="kernel" FILE_FILESYSTEM="initramfs" DEVICE_IMG_PREFIX="openwrt-ipq50xx-aarch64-glinet_gl-b3000" DEVICE_VENDOR="GL.iNET" DEVICE_MODEL="B3000" DEVICE_VARIANT="" DEVICE_ALT0_VENDOR="" DEVICE_ALT0_MODEL="" DEVICE_ALT0_VARIANT="" DEVICE_ALT1_VENDOR="" DEVICE_ALT1_MODEL="" DEVICE_ALT1_VARIANT="" DEVICE_ALT2_VENDOR="" DEVICE_ALT2_MODEL="" DEVICE_ALT2_VARIANT="" DEVICE_ALT3_VENDOR="" DEVICE_ALT3_MODEL="" DEVICE_ALT3_VARIANT="" DEVICE_ALT4_VENDOR="" DEVICE_ALT4_MODEL="" DEVICE_ALT4_VARIANT="" DEVICE_TITLE="GL.iNET B3000" DEVICE_PACKAGES="ath11k1-firmware-ipq50xx-spruce ath11k1-firmware-qcn6122 //ath11k-firmware-ipq5018 //ath11k-firmware-qcn6122 ipq-wifi-glinet_gl-b3000" TARGET="ipq50xx" SUBTARGET="aarch64" VERSION_NUMBER="23.05.3" VERSION_CODE="r23809-234f1a2efa" SUPPORTED_DEVICES="glinet,gl-b3000" /home/hostle/openwrt-gl_inet-b3000/final/openwrt-redmi-ax3000/scripts/json_add_image_info.py /home/hostle/openwrt-gl_inet-b3000/final/openwrt-redmi-ax3000/build_dir/target-aarch64_cortex-a53_musl/json_info_files/openwrt-ipq50xx-aarch64-glinet_gl-b3000-initramfs-fit-uImage.itb.json
IPKG_NO_SCRIPT=1 IPKG_INSTROOT=/home/hostle/openwrt-gl_inet-b3000/final/openwrt-redmi-ax3000/build_dir/target-aarch64_cortex-a53_musl/root.orig-ipq50xx TMPDIR=/home/hostle/openwrt-gl_inet-b3000/final/openwrt-redmi-ax3000/build_dir/target-aarch64_cortex-a53_musl/root.orig-ipq50xx/tmp /home/hostle/openwrt-gl_inet-b3000/final/openwrt-redmi-ax3000/staging_dir/host/bin/opkg --offline-root /home/hostle/openwrt-gl_inet-b3000/final/openwrt-redmi-ax3000/build_dir/target-aarch64_cortex-a53_musl/root.orig-ipq50xx --force-postinstall --add-dest root:/ --add-arch all:100 --add-arch aarch64_cortex-a53:200 list-installed > /home/hostle/openwrt-gl_inet-b3000/final/openwrt-redmi-ax3000/bin/targets/ipq50xx/aarch64/openwrt-ipq50xx-aarch64-glinet_gl-b3000.manifest
make[5]: Leaving directory '/home/hostle/openwrt-gl_inet-b3000/final/openwrt-redmi-ax3000/target/linux/ipq50xx/image'
make[4]: Leaving directory '/home/hostle/openwrt-gl_inet-b3000/final/openwrt-redmi-ax3000/target/linux/ipq50xx'
make[3]: Leaving directory '/home/hostle/openwrt-gl_inet-b3000/final/openwrt-redmi-ax3000/target/linux'
time: target/linux/install#89.68#23.18#73.47
make[2]: Leaving directory '/home/hostle/openwrt-gl_inet-b3000/final/openwrt-redmi-ax3000'
export MAKEFLAGS= ;make -w -r package/index
make[2]: Entering directory '/home/hostle/openwrt-gl_inet-b3000/final/openwrt-redmi-ax3000'
Generating package index...
Generating index for package ./base-files_1554-r23809-234f1a2efa_aarch64_cortex-a53.ipk
Generating index for package ./dropbear_2022.82-6_aarch64_cortex-a53.ipk
Generating index for package ./fstools_2023-02-28-bfe882d5-1_aarch64_cortex-a53.ipk
Generating index for package ./fwtool_2019-11-12-8f7fe925-1_aarch64_cortex-a53.ipk
Generating index for package ./ipq-wifi-glinet_gl-b3000_2023-11-10-0c2e810e-1_aarch64_cortex-a53.ipk
Generating index for package ./kmod-ath11k-ahb_5.15.150+6.1.24-3_aarch64_cortex-a53.ipk
Generating index for package ./kmod-ath11k-pci_5.15.150+6.1.24-3_aarch64_cortex-a53.ipk
Generating index for package ./kmod-ath11k_5.15.150+6.1.24-3_aarch64_cortex-a53.ipk
Generating index for package ./kmod-ath_5.15.150+6.1.24-3_aarch64_cortex-a53.ipk
Generating index for package ./kmod-bonding_5.15.150-1_aarch64_cortex-a53.ipk
Generating index for package ./kmod-cfg80211_5.15.150+6.1.24-3_aarch64_cortex-a53.ipk
Generating index for package ./kmod-coresight-stream-sock_5.15.150+11.5-1_aarch64_cortex-a53.ipk
Generating index for package ./kmod-crypto-aead_5.15.150-1_aarch64_cortex-a53.ipk
[... cut ...] iles.json 
make[2]: Leaving directory '/home/hostle/openwrt-gl_inet-b3000/final/openwrt-redmi-ax3000'
export MAKEFLAGS= ;make -w -r checksum
make[2]: Entering directory '/home/hostle/openwrt-gl_inet-b3000/final/openwrt-redmi-ax3000'
make[2]: Leaving directory '/home/hostle/openwrt-gl_inet-b3000/final/openwrt-redmi-ax3000'
make[1]: Leaving directory '/home/hostle/openwrt-gl_inet-b3000/final/openwrt-redmi-ax3000'
hostle@hostle-Satellite-C670:~/openwrt-gl_inet-b3000/final/openwrt-redmi-ax3000$ 
```

kimocoder, 2024-09-09:
My IPQ5018 is running now, and most seem to work, but the Q6 remoteproc controlling ath11k / coldboot crashes 👎

kimocoder, 2024-09-09:
@TheRootED24 You know your device is supported already? On openwrt-23.05 in facts...

But it's a OpenWRT clone, but you got lots of stuff there..

Look;
[https://github.com/search?q=repo%3ATelecominfraproject%2Fwlan-ap%20glinet&type=code](https://github.com/search?q=repo%3ATelecominfraproject%2Fwlan-ap%20glinet&type=code)

Rigi-1976, 2025-10-31:
i have a device using ipq5018-mp03.6-c1 and its a clone of old opewrt v 15 i think need to upgrade it i have full accses uart shell and all mtd blocks but cant figure out how to install new openwrt im stuck with dts mapping