hzyitc/openwrt-redmi-ax3000 #61: CR8808关于5.4qsdk11.5.05.841.1029固件2.4G wifi加密问题

Livxx, 2024-03-13:
由于智能家居需802.11b only及wpa加密才能连上，目前这个固件设置2.4G wifi为legacy，设置wpa加密方式，手机无法连接，wpa/wpa2混合加密可以连接。求教是不是我设置问题。

Livxx, 2024-03-13:
升级了5.15新内核，也是如此。找到了如下日志：
Wed Mar 13 19:58:22 2024 daemon.info hostapd: wlan0: STA fc:02:96:80:10:0f IEEE 802.11: authenticated
Wed Mar 13 19:58:22 2024 daemon.info hostapd: wlan0: STA fc:02:96:80:10:0f IEEE 802.11: associated (aid 1)
Wed Mar 13 19:58:23 2024 daemon.notice hostapd: wlan0: AP-STA-CONNECTED fc:02:96:80:10:0f
Wed Mar 13 19:58:23 2024 daemon.info hostapd: wlan0: STA fc:02:96:80:10:0f WPA: pairwise key handshake completed (WPA)
Wed Mar 13 19:58:26 2024 daemon.info hostapd: wlan0: STA fc:02:96:80:10:0f WPA: group key handshake failed (WPA) after 4 tries

Livxx, 2024-03-14:
已收到，谢谢！

Livxx, 2024-03-15:
> > 由于智能家居需802.11b only及wpa加密才能连上，目前这个固件设置2.4G wifi为legacy，设置wpa加密方式，手机无法连接，wpa/wpa2混合加密可以连接。求教是不是我设置问题。
> 
> 你手机是什么设备，这边测试小米的可以连接

两台小米手机，一台电脑均无法连接。CR8808为M79，IPQ5018。
5.4qsdk还存在负载高的问题，基本在1以上，5.15为基本0.5以下，没有问题。

Livxx, 2024-03-15:
> > > > 由于智能家居需802.11b only及wpa加密才能连上，目前这个固件设置2.4G wifi为legacy，设置wpa加密方式，手机无法连接，wpa/wpa2混合加密可以连接。求教是不是我设置问题。
> > > 
> > > 
> > > 你手机是什么设备，这边测试小米的可以连接
> > 
> > 
> > 两台小米手机，一台电脑均无法连接。CR8808为M79，IPQ5018。 5.4qsdk还存在负载高的问题，基本在1以上，5.15为基本0.5以下，没有问题。
> 
> 手头几台5000测试没问题，待5018样机过去到了做进一步测试（这边说5018需要重新适配，可能是这个问题导致的，其他人若是有5018也可以帮忙测试下

非常感谢大佬的付出！

hzyitc, 2024-03-15:
> 5.4qsdk还存在负载高的问题，基本在1以上，5.15为基本0.5以下，没有问题。

qsdk kernel 的 uptime数据不准，看top数据

Livxx, 2024-03-18:
> 这边动手看了下，新版的8818官方并没有发布固件，你确定你是5018嘛，目前5018没有固件是没法解锁ttl刷机的

原固件没试，刷成8806固件可以中断，输入命令界面显示ipq5018，也刷进openwrt了

Livxx, 2024-03-22:
> > > 这边动手看了下，新版的8818官方并没有发布固件，你确定你是5018嘛，目前5018没有固件是没法解锁ttl刷机的
> > 
> > 
> > 原固件没试，刷成8806固件可以中断，输入命令界面显示ipq5018，也刷进openwrt了
> 
> 5000也是显示5018的，看你ttl位置，如果是在右上角就是5000，5018是在下面的
> 
> 另外已经抓到881x的三套固件了，而且目前原始固件有免拆解锁telnet的方法，实测880x和881x均能开启telnet
> 
> 原始固件放在一号贴里

按照你说的TTL位置，那应该是5000，不好意思搞错了。5000应该适配好了，不应该有wpa加密问题的，很无奈