hzyitc/openwrt-redmi-ax3000 #62: 一些关于remoteproc的疑惑

UzLung1435, 2024-03-13:
前些日子拉了联通家宽送了台VS010，几乎一样的硬件参数让我萌生了想移植的想法。遂对ax3000的dts进行修改试图适配vs010。
使用[ipq50xx-mainline-kernel-5.15-openwrt-23.05](https://github.com/hzyitc/openwrt-redmi-ax3000/tree/ipq50xx-mainline-kernel-5.15-openwrt-23.05)中的AX3000tree修改后进行编译，在启动时出现以下错误：
```
[   16.495910] ath11k soc:wifi1@c000000: qcn6122
[   16.500757] ath11k soc:wifi1@c000000: FW memory mode: 2
[   16.505355] remoteproc remoteproc2: powering up cd00000.remoteproc:remoteproc_pd2
[   16.510198] remoteproc remoteproc2: Booting fw image IPQ5018/q6_fw.mdt, size 1820
[   16.520559] remoteproc remoteproc2: remote processor cd00000.remoteproc:remoteproc_pd2 is now up
[   16.532118] ath11k c000000.wifi: qmi fail to get qcom,m3-dump-addr, ignore m3 dump mem req
[   16.538388] kmodloader: done loading kernel modules from /etc/modules.d/*
[   16.604091] ath11k soc:wifi1@c000000: qmi ignore invalid mem req type 10
[   16.615491] qcom-q6-mpd cd00000.remoteproc: fatal error received: err_smem_ver.2.1:
[   16.615491] QC Image Version : QC_IMAGE_VERSION_STRING=WLAN.HK.2.7.0.1-01744-QCAHKSWPL_SILICONZ-1
[   16.615491] Image Variant : IMAGE_VARIANT_STRING=5018.wlanfw2.map_spr_spr_evalQ
[   16.615491] platform.c:965 Assertion remote_ddr_mem_req->mem_seg[bdf_mem_idx].addr != 0x0 && remote_ddr_mem_req->mem_seg[bdf_             mem_idx].size == (Q6_DDR_BDF_MAX_SIZE * 2) failed param0 :zero,param1 :zero,param2 :zero
[   16.615491] Thread ID : 0x00000046 Thread name : wlan_platform Process ID : 0x00000002 Process name :wlan1
[   16.615491]
[   16.615491] Registers:
[   16.615491] SP : 0x4bfdbe58
[   16.615491] FP : 0x4bfdbea0
[   16.615491] PC : 0xd025bc70
[   16.615491] SSR : 0x00000000
[   16.615491] BADVA : 0x00000000
[   16.615491] LR : 0xd024a1d0
[   16.615491]
[   16.615491] StackDump
[   16.615491] from:0x4bfdbe58
[   16.615491] to: 0x00000000:
[   16.615491]
[   16.635253] mtdblock: MTD device 'rootfs' is NAND, please consider using UBI block devices instead.
[   16.676793] remoteproc remoteproc0: crash detected in cd00000.remoteproc: type fatal error
[   16.676939] remoteproc remoteproc0: handling crash #1 in cd00000.remoteproc
[   16.716294] remoteproc remoteproc0: recovering cd00000.remoteproc
[   16.729362] remoteproc remoteproc0: stopped remote processor cd00000.remoteproc
[   16.844279] remoteproc remoteproc0: remote processor cd00000.remoteproc is now up
[   26.729220] ath11k soc:wifi1@c000000: failed to wait qmi memory request: -110
[   26.729282] ath11k soc:wifi1@c000000: qmi failed to respond fw mem req: -110
[   33.389770] br-lan: port 1(eth0) entered blocking state
[   33.389818] br-lan: port 1(eth0) entered disabled state
[   33.394428] device eth0 entered promiscuous mode
[   48.170659] ath11k c000000.wifi: chip_id 0x0 chip_family 0x4 board_id 0x10 soc_id 0xffffffff
[   48.170736] ath11k c000000.wifi: fw_version 0x270206d0 fw_build_timestamp 2022-08-04 13:28 fw_build_id WLAN.HK.2.7.0.1-01744-             QCAHKSWPL_SILICONZ-1
```
此时，有线连接无网络，无法dhcp，无线也不工作。

能否在百忙之中赐教🥺十分感谢><

~~感觉像是我dts没写好的原因（x~~

hzyitc, 2024-03-15:
dts里面保留内存和wifi的内存地址配置有改动吗?跟我dts一样吗?

UzLung1435, 2024-03-15:
> dts里面保留内存和wifi的内存地址配置有改动吗?跟我dts一样吗?

`reserved-memory`下我看你的tree里只有两项所以只改了那两项的。从某个闭源固件里捞出来的如下所示：
```
reserved-memory {
	#address-cells = <0x02>;
	#size-cells = <0x02>;
	ranges;

	bt@7000000 {
		no-map;
		phandle = <0x2b>;
		reg = <0x00 0x7000000 0x00 0x58000>;
	};

	m3_dump@4D100000 {
		no-map;
		phandle = <0x24>;
		reg = <0x00 0x4d100000 0x00 0x100000>;
	};

	m3_dump_qcn6122_1@4E200000 {
		no-map;
		phandle = <0x26>;
		reg = <0x00 0x4e200000 0x00 0x100000>;
	};

	nss@40000000 {
		no-map;
		reg = <0x00 0x40000000 0x00 0x1000000>;
	};

	q6_code_data@4B000000 {
		no-map;
		reg = <0x00 0x4b000000 0x00 0x60000>;
	};

	q6_etr_dump@4D200000 {
		no-map;
		phandle = <0x21>;
		reg = <0x00 0x4d200000 0x00 0x100000>;
	};

	q6_ipq5018_data@4C400000 {
		no-map;
		phandle = <0x23>;
		reg = <0x00 0x4c400000 0x00 0xd00000>;
	};

	q6_mem_regions@4B000000 {
		no-map;
		phandle = <0x20>;
		reg = <0x00 0x4b000000 0x00 0x3400000>;
	};

	q6_qcn6122_data1@4D300000 {
		no-map;
		phandle = <0x25>;
		reg = <0x00 0x4d300000 0x00 0xf00000>;
	};

	q6_qcn6122_etr_1@4E300000 {
		no-map;
		phandle = <0x27>;
		reg = <0x00 0x4e300000 0x00 0x100000>;
	};

	sbl@4aa00000 {
		no-map;
		reg = <0x00 0x4aa00000 0x00 0x100000>;
	};

	smem@4ab00000 {
		no-map;
		phandle = <0x0a>;
		reg = <0x00 0x4ab00000 0x00 0x100000>;
	};

	tz@4ac00000 {
		no-map;
		reg = <0x00 0x4ac00000 0x00 0x400000>;
	};

	tzapp@4a400000 {
		no-map;
		phandle = <0x07>;
	};

	tzapp_data@4a700000 {
		compatible = "shared-dma-pool";
		no-map;
		phandle = <0x08>;
	};

	uboot@4a800000 {
		no-map;
		reg = <0x00 0x4a800000 0x00 0x200000>;
	};
};
```

wifi的内存配置我看起来是跟你的一样的。
从固件中提取出来的如下所示：
```
wifi1@c000000 {
	compatible = "qcom,cnss-qcn6122\0qcom,cnss-qcn9100";
	interrupts = <0x00 0x1a0 0x01>;
	mem-region = <0x25>;
	msi-parent = <0x1b>;
	qcom,bdf-addr = <0x00 0x00 0x4d300000 0x00 0x00>;
	qcom,board_id = <0x60>;
	qcom,caldb-addr = <0x00 0x00 0x00 0x00 0x00>;
	qcom,caldb-size = <0x00>;
	qcom,multipd_arch;
	qcom,rproc = <0x29>;
	qcom,tgt-mem-mode = <0x02>;
	qcom,userpd-subsys-name = "q6v5_wcss_userpd2";
	status = "ok";
};

wifi2@c000000 {
	compatible = "qcom,cnss-qcn6122";
	interrupts = <0x00 0x1c0 0x01>;
	msi-parent = <0x1b>;
	qcom,rproc = <0x22>;
	status = "disabled";
};

wifi3@f00000 {
	compatible = "qcom,cnss-qcn9000";
	qca,auto-restart;
	qcom,wlan-ramdump-dynamic = <0x400000>;
	qrtr_node_id = <0x20>;
	status = "disabled";
};

wifi4@f00000 {
	compatible = "qcom,cnss-qcn9000";
	qca,auto-restart;
	qcom,wlan-ramdump-dynamic = <0x400000>;
	qrtr_node_id = <0x21>;
	status = "disabled";
};

wifi5@f00000 {
	compatible = "qcom,cnss-qcn9224";
	qca,auto-restart;
	qcom,wlan-ramdump-dynamic = <0x600000>;
	status = "disabled";
};

wifi@c000000 {
	compatible = "qcom,cnss-qca5018\0qcom,ipq5018-wifi";
	interrupt-names = "misc-pulse1\0misc-latch\0sw-exception\0ce0\0ce1\0ce2\0ce3\0ce4\0ce5\0ce6\0ce7\0ce8\0ce9\0ce10\0ce11\0host2wbm-desc-feed\0host2reo-re-injection\0host2reo-command\0host2rxdma-monitor-ring3\0host2rxdma-monitor-ring2\0host2rxdma-monitor-ring1\0reo2ost-exception\0wbm2host-rx-release\0reo2host-status\0reo2host-destination-ring4\0reo2host-destination-ring3\0reo2host-destination-ring2\0reo2host-destination-ring1\0rxdma2host-monitor-destination-mac3\0rxdma2host-monitor-destination-mac2\0rxdma2host-monitor-destination-mac1\0ppdu-end-interrupts-mac3\0ppdu-end-interrupts-mac2\0ppdu-end-interrupts-mac1\0rxdma2host-monitor-status-ring-mac3\0rxdma2host-monitor-status-ring-mac2\0rxdma2host-monitor-status-ring-mac1\0host2rxdma-host-buf-ring-mac3\0host2rxdma-host-buf-ring-mac2\0host2rxdma-host-buf-ring-mac1\0rxdma2host-destination-ring-mac3\0rxdma2host-destination-ring-mac2\0rxdma2host-destination-ring-mac1\0host2tcl-input-ring4\0host2tcl-input-ring3\0host2tcl-input-ring2\0host2tcl-input-ring1\0wbm2host-tx-completions-ring4\0wbm2host-tx-completions-ring3\0wbm2host-tx-completions-ring2\0wbm2host-tx-completions-ring1\0tcl2host-status-ring";
	interrupts = <0x00 0x120 0x01 0x00 0x121 0x01 0x00 0x122 0x01 0x00 0x124 0x01 0x00 0x125 0x01 0x00 0x126 0x01 0x00 0x127 0x01 0x00 0x128 0x01 0x00 0x129 0x01 0x00 0x12a 0x01 0x00 0x12b 0x01 0x00 0x12c 0x01 0x00 0x12d 0x01 0x00 0x12e 0x01 0x00 0x12f 0x01 0x00 0x130 0x01 0x00 0x131 0x01 0x00 0x132 0x01 0x00 0x133 0x01 0x00 0x134 0x01 0x00 0x135 0x01 0x00 0x136 0x01 0x00 0x137 0x01 0x00 0x14e 0x01 0x00 0x139 0x01 0x00 0x13a 0x01 0x00 0x13b 0x01 0x00 0x13c 0x01 0x00 0x13d 0x01 0x00 0x13e 0x01 0x00 0x13f 0x01 0x00 0x140 0x01 0x00 0x141 0x01 0x00 0x142 0x01 0x00 0x143 0x01 0x00 0x144 0x01 0x00 0x145 0x01 0x00 0x146 0x01 0x00 0x147 0x01 0x00 0x148 0x01 0x00 0x149 0x01 0x00 0x14a 0x01 0x00 0x14b 0x01 0x00 0x14c 0x01 0x00 0x14d 0x01 0x00 0x138 0x01 0x00 0x14f 0x01 0x00 0x156 0x01 0x00 0x150 0x01 0x00 0x151 0x01 0x00 0x152 0x01 0x00 0x153 0x01>;
	mem-region = <0x23>;
	qcom,bdf-addr = <0x00 0x00 0x4c400000 0x00 0x00>;
	qcom,board_id = <0x10>;
	qcom,caldb-addr = <0x00 0x00 0x00 0x00 0x00>;
	qcom,caldb-size = <0x00>;
	qcom,multipd_arch;
	qcom,rproc = <0x28>;
	qcom,tgt-mem-mode = <0x02>;
	qcom,userpd-subsys-name = "q6v5_wcss_userpd1";
	reg = <0xc000000 0x1000000>;
	status = "ok";
};
```
🥺请您赐教，万分感谢！~

hzyitc, 2024-03-15:
保持我的配置不动试一下。

hzyitc, 2024-03-16:
> > 保持我的配置不动试一下。
> 
> 保持配置不动，会直接重启（ ![image](https://private-user-images.githubusercontent.com/27557591/313273813-6d8e6a62-cc50-40d9-8200-16570ed89864.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTA1ODA0MjAsIm5iZiI6MTcxMDU4MDEyMCwicGF0aCI6Ii8yNzU1NzU5MS8zMTMyNzM4MTMtNmQ4ZTZhNjItY2M1MC00MGQ5LTgyMDAtMTY1NzBlZDg5ODY0LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAzMTYlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMzE2VDA5MDg0MFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTcwZTEzOWRjYjBkYTk5OTljZmVkNmJjZjE5ZDczZjcwMTMzNWNhYTE0NGJmYWFkNjc1MjQ4OTkyOTNiMmI1OTAmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.EwffbSm1ANH-_fVpm0wubhFMew6G-VdJk8SaNXHfVoo) 大概如上图所示，具体的log我没保存TT

你这错误日志还在上面，这部分不是

UzLung1435, 2024-03-28:
如果直接使用您编译的AX3000固件，switch和wlan都不工作，下面附上完整的启动日志：
```

Starting kernel ...

[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 5.4.164 (runner@fv-az1114-505) (gcc version 8.4.0 (OpenWrt GCC 8.4.0 ci-20240308-145228-ee2c84b6)) #0 SMP PREEMPT Fri Mar 8 14:47:37 2024
[    0.000000] CPU: ARMv7 Processor [51af8014] revision 4 (ARMv7), cr=10c0383d
[    0.000000] CPU: div instructions available: patching division code
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
[    0.000000] OF: fdt: Machine model: Redmi AX3000
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] psci: probing for conduit method from DT.
[    0.000000] psci: PSCIv1.0 detected in firmware.
[    0.000000] psci: Using standard PSCI v0.2 function IDs
[    0.000000] psci: MIGRATE_INFO_TYPE not supported.
[    0.000000] psci: SMC Calling Convention v1.0
[    0.000000] psci: OSI mode supported.
[    0.000000] percpu: Embedded 15 pages/cpu s32204 r8192 d21044 u61440
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 48576
[    0.000000] Kernel command line: console=ttyMSM0,115200n8 ubi.mtd=rootfs root=mtd:ubi_rootfs rootfstype=squashfs rootwait root=/dev/ubiblock0_1 coherent_pool=2M
[    0.000000] Dentry cache hash table entries: 32768 (order: 5, 131072 bytes, linear)
[    0.000000] Inode-cache hash table entries: 16384 (order: 4, 65536 bytes, linear)
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory: 183396K/196608K available (6894K kernel code, 262K rwdata, 2088K rodata, 1024K init, 360K bss, 13212K reserved, 0K cma-reserved, 0K highmem)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=2, Nodes=1
[    0.000000] rcu: Preemptible hierarchical RCU implementation.
[    0.000000]  Tasks RCU enabled.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 10 jiffies.
[    0.000000] NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
[    0.000000] GICv2m: range[mem 0x0b00a000-0x0b00affc], SPI[448:479]
[    0.000000] GICv2m: range[mem 0x0b00b000-0x0b00bffc], SPI[480:511]
[    0.000000] random: get_random_bytes called from start_kernel+0x344/0x4f4 with crng_init=0
[    0.000000] arch_timer: cp15 and mmio timer(s) running at 24.00MHz (virt/virt).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0x588fe9dc0, max_idle_ns: 440795202592 ns
[    0.000007] sched_clock: 56 bits at 24MHz, resolution 41ns, wraps every 4398046511097ns
[    0.000018] Switching to timer-based delay loop, resolution 41ns
[    0.000423] Console: colour dummy device 80x30
[    0.000456] Calibrating delay loop (skipped), value calculated using timer frequency.. 48.00 BogoMIPS (lpj=240000)
[    0.000471] pid_max: default: 32768 minimum: 301
[    0.000624] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.000637] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.001531] CPU: Testing write buffer coherency: ok
[    0.001968] CPU0: thread -1, cpu 0, socket 0, mpidr 80000000
[    0.002779] Setting up static identity map for 0x41300000 - 0x41300060
[    0.002958] rcu: Hierarchical SRCU implementation.
[    0.003445] smp: Bringing up secondary CPUs ...
[    0.004416] CPU1: thread -1, cpu 1, socket 0, mpidr 80000001
[    0.004600] smp: Brought up 1 node, 2 CPUs
[    0.004621] SMP: Total of 2 processors activated (96.00 BogoMIPS).
[    0.004627] CPU: All CPU(s) started in SVC mode.
[    0.012815] VFP support v0.3: implementor 51 architecture 3 part 40 variant 3 rev 4
[    0.013237] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[    0.013265] futex hash table entries: 512 (order: 3, 32768 bytes, linear)
[    0.013424] pinctrl core: initialized pinctrl subsystem
[    0.014542] NET: Registered protocol family 16
[    0.016010] DMA: preallocated 2048 KiB pool for atomic coherent allocations
[    0.017276] cpuidle: using governor ladder
[    0.017332] cpuidle: using governor menu
[    0.017565] NET: Registered protocol family 42
[    0.017909] hw-breakpoint: found 5 (+1 reserved) breakpoint and 4 watchpoint registers.
[    0.017922] hw-breakpoint: maximum watchpoint size is 8 bytes.
[    0.065116] qcom,gcc-ipq5018 1800000.gcc: Registered ipq5018 GCC clocks provider
[    0.079787] msm_sps_probe: sps:sps is ready
[    0.108171] pps_core: LinuxPPS API ver. 1 registered
[    0.108183] pps_core: Software ver. 5.3.6 - Copyright 2005-2007 Rodolfo Giometti <giometti@linux.it>
[    0.108207] PTP clock support registered
[    0.109047] Advanced Linux Sound Architecture Driver Initialized.
[    0.109154] workqueue: max_active 576 requested for napi_workq is out of range, clamping between 1 and 512
[    0.110534] clocksource: Switched to clocksource arch_sys_counter
[... 256 lines cut ...]
[   34.967605] ath11k soc:wifi1@c000000: chip_id 0x0 chip_family 0x4 board_id 0x60 soc_id 0xffffffff
[   34.967653] ath11k soc:wifi1@c000000: fw_version 0x270206d0 fw_build_timestamp 2022-08-04 13:28 fw_build_id QC_IMAGE_VERSION_STRING=WLAN.HK.2.7.0.1-01744-QCAHKSWPL_SILICONZ-1
[   35.141540] ath11k c000000.wifi: qmi caldata downloaded: type: 2
[   35.180282] ath11k c000000.wifi: No qcom,pta_num entry in dev-tree.
[   35.180324] ath11k c000000.wifi: No qcom,coex_mode entry in dev-tree.
[   35.186624] ath11k c000000.wifi: No qcom,bt-active-time entry in dev-tree.
[... cut ...]
[   36.257557] ath11k c000000.wifi: Direct firmware load for qdss_trace_config.bin failed with error -2
[   36.257592] ath11k c000000.wifi: Falling back to sysfs fallback for: qdss_trace_config.bin
[   36.949869] ath11k c000000.wifi: qmi failed to load QDSS config: qdss_trace_config.bin
[   36.949914] ath11k c000000.wifi: Failed to download QDSS config to FW: -12
[   38.132576] br-lan: port 1(eth0) entered blocking state
[   38.132604] br-lan: port 1(eth0) entered disabled state
[   38.138867] device eth0 entered promiscuous mode
```

UzLung1435, 2024-03-28:
同样的，使用某个闭源固件所取得的开机log如下所示：
```
Starting kernel ...

[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 5.4.164 (Author@qsdk12) (gcc version 7.5.0 (OpenWrt GCC 7.5.0 r0-d63f1527e)) #0 SMP PREEMPT Wed Jan 3 10:38:32 2024
[    0.000000] CPU: ARMv7 Processor [51af8014] revision 4 (ARMv7), cr=10c0383d
[    0.000000] CPU: div instructions available: patching division code
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
[    0.000000] OF: fdt: Machine model: CUCC VS010
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] OF: reserved mem: OVERLAP DETECTED!
[    0.000000] q6_code_data@4B000000 (0x4b000000--0x4b060000) overlaps with q6_mem_regions@4B000000 (0x4b000000--0x4e400000)
[    0.000000] OF: reserved mem: OVERLAP DETECTED!
[    0.000000] q6_mem_regions@4B000000 (0x4b000000--0x4e400000) overlaps with q6_ipq5018_data@4C400000 (0x4c400000--0x4d100000)
[    0.000000] psci: probing for conduit method from DT.
[    0.000000] psci: PSCIv1.0 detected in firmware.
[    0.000000] psci: Using standard PSCI v0.2 function IDs
[    0.000000] psci: MIGRATE_INFO_TYPE not supported.
[    0.000000] psci: SMC Calling Convention v1.0
[    0.000000] psci: OSI mode supported.
[    0.000000] percpu: Embedded 15 pages/cpu s32076 r8192 d21172 u61440
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 45504
[    0.000000] Kernel command line: console=ttyMSM0,115200n8 ubi.mtd=rootfs root=mtd:ubi_rootfs rootfstype=squashfs rootwait ubi.mtd=rootfs root=/dev/ubiblock0_1 swiotlb=1 coherent_pool=2M
[    0.000000] Dentry cache hash table entries: 32768 (order: 5, 131072 bytes, linear)
[    0.000000] Inode-cache hash table entries: 16384 (order: 4, 65536 bytes, linear)
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory: 174024K/184320K available (5733K kernel code, 228K rwdata, 1228K rodata, 264K init, 340K bss, 10296K reserved, 0K cma-reserved, 0K highmem)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=2, Nodes=1
[    0.000000] rcu: Preemptible hierarchical RCU implementation.
[    0.000000]  Tasks RCU enabled.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 10 jiffies.
[    0.000000] NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
[    0.000000] GICv2m: range[mem 0x0b00a000-0x0b00affc], SPI[448:479]
[    0.000000] GICv2m: range[mem 0x0b00b000-0x0b00bffc], SPI[480:511]
[    0.000000] arch_timer: cp15 and mmio timer(s) running at 24.00MHz (virt/virt).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0x588fe9dc0, max_idle_ns: 440795202592 ns
[    0.000006] sched_clock: 56 bits at 24MHz, resolution 41ns, wraps every 4398046511097ns
[    0.000018] Switching to timer-based delay loop, resolution 41ns
[    0.000377] Calibrating delay loop (skipped), value calculated using timer frequency.. 48.00 BogoMIPS (lpj=240000)
[    0.000394] pid_max: default: 32768 minimum: 301
[    0.000561] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.000576] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.001558] CPU: Testing write buffer coherency: ok
[    0.002005] CPU0: thread -1, cpu 0, socket 0, mpidr 80000000
[    0.002917] Setting up static identity map for 0x41208280 - 0x412082e0
[    0.003106] rcu: Hierarchical SRCU implementation.
[    0.003633] smp: Bringing up secondary CPUs ...
[    0.004682] CPU1: thread -1, cpu 1, socket 0, mpidr 80000001
[    0.004883] smp: Brought up 1 node, 2 CPUs
[    0.004901] SMP: Total of 2 processors activated (96.00 BogoMIPS).
[    0.004907] CPU: All CPU(s) started in SVC mode.
[    0.013993] random: get_random_u32 called from 0x81489934 with crng_init=0
[    0.014171] VFP support v0.3: implementor 51 architecture 3 part 40 variant 3 rev 4
[    0.014662] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[    0.014700] futex hash table entries: 512 (order: 3, 32768 bytes, linear)
[    0.014892] pinctrl core: initialized pinctrl subsystem
[    0.016023] NET: Registered protocol family 16
[    0.017544] DMA: preallocated 2048 KiB pool for atomic coherent allocations
[    0.018756] cpuidle: using governor ladder
[    0.018807] cpuidle: using governor menu
[    0.019053] NET: Registered protocol family 42
[    0.019442] hw-breakpoint: found 5 (+1 reserved) breakpoint and 4 watchpoint registers.
[    0.019454] hw-breakpoint: maximum watchpoint size is 8 bytes.
[    0.069681] qcom,gcc-ipq5018 1800000.gcc: Registered ipq5018 GCC clocks provider
[    0.086403] msm_sps_probe: sps:sps is ready
[    3.082928] pps_core: LinuxPPS API ver. 1 registered
[    3.082940] pps_core: Software ver. 5.3.6 - Copyright 2005-2007 Rodolfo Giometti <giometti@linux.it>
[    3.082962] PTP clock support registered
[    3.083906] Advanced Linux Sound Architecture Driver Initialized.
[... 702 lines cut ...]
[   37.272989] wlan: [1209:I:ANY] wlan_cfg80211_chan_to_phymode: band:1 width: 5 channel_cfreq: 5220 center_freq1: 5250 chandef.center_freq2: 0 flags: 512
[   37.277189] wlan: [1209:I:ANY] HT Support:81dfac98 VHT Support:f034f67f HE Support:9d01aa0b
[   37.290620] wlan: [1209:I:ANY] Max supported wireless mode: 3
[   37.298792] wlan: [1209:I:ANY] Phymode: 31
[   37.298792]
[   37.305179] wlan: [1209:I:MBSSIE] wlan_cfg80211_start_ap: Set tx-vap ath1: vdev_id:0
[... cut ...]
Custom Service Class Count: 1
SLA Count: 7
***** No coldboot or daemon support for ap******
device: wifi0 vifs: cfg023579
device: wifi1 vifs: cfg043579 cfg043579
[   39.020667] ***** Fast boot VAP is completed... *****
[   39.570874] qca-wifi loaded
[   41.384241] wlan: [895:I:ANY] ieee80211_acs_post_event: [EXT] Channel number: 33a6665b
[   41.384282] wlan: [895:I:ANY] wlan_mbss_done_ht40: Channel: 33a6665b
[   41.391705] wlan: [895:I:ANY] ieee80211_ht40intol_evhandler: [EXT] Lock held duration (8ms)
```

hzyitc, 2024-04-05:
看你这日志，wifi好像正常启动了

islwx, 2026-02-15:
请问现在解决了么，我也有一样的路由器，想刷openwrt