hzyitc/openwrt-redmi-ax3000 #63: 不知为何自动关机

114514thD, 2024-03-23:
刷入最新的几个固件，全都是一开无线，就过一会自己熄灯断网，ttl输命令无反应，测电压txrx1v，vcc1.8v

114514thD, 2024-03-24:
经过实验，发现具体现象：2.4G中继+无线，一会儿自动关机、5G中继+2.4G无线，正常，重启正常、5G中继+2.4G无线+5G无线，正常，但是重启后关机，进后台开关5G无线后能用，但是重启后还是关机
所以关键现象是同一网卡同时开中继和无线广播的情况下不能正常启动
关机也不是真正的关机，是电压txrx1v，vcc1.8v，而正常工作时txrxvcc都是1.8v

baysonfox, 2024-04-04:
我也遇到了类似情况，不过只发生在2.4G频段，且与部分小米家居的智能设备有关.
目前似乎能定位到发射功率，可能在比较高的功率会崩溃（？

114514thD, 2024-04-04:
我试过了降低发射功率，但是过一段时间还是会自动关机，5G中继的话久了也会自动关机，但是一重启很快就会关机，我感觉可能是固件哪里的问题。
换了恩山论坛某团队制作的Nwrt（用了QSDK）后一切正常。

baysonfox, 2024-04-04:
我也在继续尝试，似乎还是会不太稳定. 有点难定位原因.

baysonfox, 2024-04-04:
>  换了恩山论坛某团队制作的Nwrt（用了QSDK）后一切正常。
介意发个论坛的链接吗？

114514thD, 2024-04-04:
https://www.right.com.cn/forum/forum.php
固件团队看起来没打算开源，付费进他们的Q群才可以下载

hzyitc, 2024-04-05:
可能是oom？

中继我确实没测试过

114514thD, 2024-04-05:
应该不是，我观察过，内存占用一直很正常，nwrt的固件上试过爆内存，也只会卡住而不会关机

ne00, 2024-07-11:
遇到了同样情况

charlestamz, 2024-07-22:
我也有这个情况

CRT-HAO, 2024-07-24:
实测有相同问题
使用版本：`[CI ipq50xx-mainline-kernel-5.4-openwrt-21.02-qsdk-wifi-rebase 2024-07-19 15:47:59 UTC(+00:00)](https://github.com/hzyitc/openwrt-redmi-ax3000/releases/tag/ci-ipq50xx-mainline-kernel-5.4-openwrt-21.02-qsdk-wifi-rebase-20240719-154759-UTC)`

I also encountered the same problem
Version: `[CI ipq50xx-mainline-kernel-5.4-openwrt-21.02-qsdk-wifi-rebase 2024-07-19 15:47:59 UTC(+00:00)](https://github.com/hzyitc/openwrt-redmi-ax3000/releases/tag/ci-ipq50xx-mainline-kernel-5.4-openwrt-21.02-qsdk-wifi-rebase-20240719-154759-UTC)`

CRT-HAO, 2024-07-24:
> 我试过了降低发射功率，但是过一段时间还是会自动关机，5G中继的话久了也会自动关机，但是一重启很快就会关机，我感觉可能是固件哪里的问题。 换了恩山论坛某团队制作的Nwrt（用了QSDK）后一切正常。 可能确实和小米的智能硬件有点关系，因为在后台输入ifconfig的话，可以看到有miot的网段，但是我不确定关系几何。

因为我来自台湾，没办法付费进群下载nwrt固件，你是否愿意私下分享一下nwrt固件用于研究？
我的邮箱是m831718@gmail.com

danzhewuju, 2025-03-09:
> > 换了恩山论坛某团队制作的Nwrt（用了QSDK）后一切正常。
> > 介意发个论坛的链接吗？

可以给我发一份吗？我也遇到一样的问题，邮箱：1668926294@qq.com