hzyitc/openwrt-redmi-ax3000 #67: 小米AX3000刷入红米AX3000固件后5GHz极不稳定且自动被踢掉线

AmasakiKiha, 2024-04-18:
日志：是真的无。检查了控制台发现系统日志并没有输出任何报错，就好像设备自己断开了连接一样。
路由器界面：5Ghz信号弱
[image]
设备：联想笔记本（网卡：英特尔AX201）。现象：可以正常连接，贴着路由器信号只有三格，一段时间后掉线重连，一段更久的时间后（约2~5小时）掉线且无法自动连接，手动连接正常。
设备：小米11Pro（澎湃OS）、小米平板5Pro（澎湃OS）。现象：设备连接2.4G正常，5G在切换频段后约两分钟内可以连接。之后设备断开连接，日志里什么报错都没有，就好像设备自己断开了一样。之后5G的wifi正常搜得到，但是尝试连接会经过正在连接-正在获取IP地址-未连接，已保存（设置静态ip后直接为正在连接-已保存）。之后无论什么时候都是已保存的状态，再连不上了。
[image]
排查过DHCP，静态地址问题依旧，排查过网段，更换ip网段问题依旧
排查过信道，64区域信道和149区域信道全试过
排查过带宽，40,80,160试过
排查过国家代码，00，CN，PL试过
排查过低ACK应答，调整检测时间长短，开关各种选项
问题依旧

AmasakiKiha, 2024-04-18:
加密方式也改过，PSK，PSK2，PSK3，1/2Mix，2/3Mix都改过，问题依旧

AmasakiKiha, 2024-04-18:
[image]
看了#35，好像我的路由器也是负载爆了，固件是最新的3月的固件
top同样是啥都没
[image]

AmasakiKiha, 2024-04-19:
[image]
可以看到一段时间后路由器抽风一样负载突然升高，然后设备掉线，不知道发生什么了

AmasakiKiha, 2024-04-19:
感觉就像有什么东西在循环慢慢加负载一样……很怪，重启也不消失，刷写固件到可以……

jiuin1104, 2024-05-01:
我也存在同样问题，一样是小米路由器AX3000，5GHZ提示无法加入网络，设置都已经试过了。希望大大看到能修复问题，现在只能作为交换机使用了。。。

KaranocaVe, 2024-09-01:
感谢避雷！今天刚解锁了ssh，准备刷一个红米的固件到小米里面去的，现在还是决定用原装系统

httpsTeabagexe, 2024-09-01:
> Thanks for lightning protection! I just unlocked SSH today.,I'm going to flash a Redmi's firmware to Xiaomi.,Now I still decide to use the original system.

I just flashed my router and i cant use it normally. I recommend you using the original firmware

KaranocaVe, 2024-09-28:
所以现在市面上是没有能给小米ax3000跑的固件吗

ASTPoseidon, 2024-10-21:
找遍了也没找到小米 ax3000 合适的固件

KaranocaVe, 2024-11-17:
希望有大佬能把RA80适用的固件搞出来，买了这么久一直不能刷openwrt

hzyitc, 2025-02-08:
wifi的bdf需要适配。

bbaa-bbaa, 2026-11-12:
用 AI 在 [kuncy7/openwrt-nss-edma](https://github.com/kuncy7/openwrt-nss-edma) 上面移植了 AX3000 的 bdf

https://github.com/bbaa-bbaa/openwrt-nss-xiaomi-ra80-ax3000

可以试试，~~不保证好使~~