hzyitc/openwrt-redmi-ax3000 #73: 你好 hzyitc,

TonyTeng22, 2024-05-23:
你好 hzyitc,
非常感谢你关于红米AX3000的贡献， 以及上个月向openwrt官方仓库PR尝试. 让IPQ50xx WIFI6用户看到了曙光和希望。
RAX3000Q/QY 的拆机文章在这里，[拆机](https://www.acwifi.net/18224.html)

打开telnet: ping 测试 输入：$(telnetd&) 开始； 然后输入$(echo${IFS}root::0:0:::/bin/sh${IFS}>/etc/passwd)
[image]
ps:
[image]
flash:
[image]
RAX300QY_dmesg:
[RAX300QY_dmesg.txt](https://github.com/hzyitc/openwrt-redmi-ax3000/files/15413231/RAX300QY_dmesg.txt)

ini:
[RAX3000QY_ini.txt](https://github.com/hzyitc/openwrt-redmi-ax3000/files/15413230/RAX3000QY_ini.txt)

facto
[factory_param_config.def.txt](https://github.com/hzyitc/openwrt-redmi-ax3000/files/15413236/factory_param_config.def.txt)
ry_param_config.def:

_Originally posted by @TonyTeng22 in https://github.com/hzyitc/openwrt-redmi-ax3000/discussions/72_

hzyitc, 2024-06-23:
麻烦提供下原厂rom。

TonyTeng22, 2024-06-25:
***@***.***:/tmp# cat /proc/mtd
dev:    size   erasesize  name
mtd0: 00080000 00020000 "0:SBL1"
mtd1: 00080000 00020000 "0:MIBIB"
mtd2: 00040000 00020000 "0:BOOTCONFIG"
mtd3: 00040000 00020000 "0:BOOTCONFIG1"
mtd4: 00100000 00020000 "0:QSEE"
mtd5: 00100000 00020000 "0:QSEE_1"
mtd6: 00040000 00020000 "0:DEVCFG"
mtd7: 00040000 00020000 "0:DEVCFG_1"
mtd8: 00040000 00020000 "0:CDT"
mtd9: 00040000 00020000 "0:CDT_1"
mtd10: 00080000 00020000 "0:APPSBLENV"
mtd11: 00140000 00020000 "0:APPSBL"
mtd12: 00140000 00020000 "0:APPSBL_1"
mtd13: 00100000 00020000 "0:ART"
mtd14: 00080000 00020000 "0:TRAINING"
mtd15: 03000000 00020000 "rootfs"
mtd16: 03000000 00020000 "rootfs_1"
mtd17: 00a40000 00020000 "TZPARAM"
mtd18: 00a40000 00020000 "TZBAK"
mtd19: 002fd6f8 0001f000 "kernel"
mtd20: 00369000 0001f000 "wifi_fw"
mtd21: 00004800 0001f000 "bt_fw"
mtd22: 0118f000 0001f000 "ubi_rootfs"
mtd23: 00d90000 0001f000 "rootfs_data"
mtd24: 00706000 0001f000 "tzparam"
mtd25: 00706000 0001f000 "tzbak"

***@***.***:/# cat /proc/cmdline
console=ttyMSM0,115200n8 cnss2.bdf_integrated=0x24 cnss2.bdf_pci0=0x60
cnss2.bdf_pci1=0x60 cnss2.skip_radio_bmap=4 ubi.mtd=rootfs
root=mtd:ubi_rootfs rootfstype=squashfs rootwait swiotlb=1

***@***.***:/tmp# mount
mtd:ubi_rootfs on /rom type squashfs (ro,relatime)
proc on /proc type proc (rw,nosuid,nodev,noexec,noatime)
sysfs on /sys type sysfs (rw,nosuid,nodev,noexec,noatime)
tmpfs on /tmp type tmpfs (rw,nosuid,nodev,noatime)
/dev/ubi0_4 on /overlay type ubifs (rw,noatime)
overlayfs:/overlay on / type overlay
(rw,noatime,lowerdir=/,upperdir=/overlay/upper,workdir=/overlay/work)
tmpfs on /dev type tmpfs (rw,nosuid,relatime,size=512k,mode=755)
devpts on /dev/pts type devpts
(rw,nosuid,noexec,relatime,mode=600,ptmxmode=000)
/dev/mtdblock20 on /lib/firmware/IPQ5018/WIFI_FW type squashfs (ro,relatime)
debugfs on /sys/kernel/debug type debugfs (rw,noatime)
ubi1:tzparam on /etc/tzcfg type ubifs (rw,relatime)
ubi2:tzbak on /etc/tzbak type ubifs (rw,relatime)

因为没有nand的编程器，请问需要如何操作才能导出你要的数据？

以下是dd导出的：

 ROM.tar.gz
<https://drive.google.com/file/d/1SM-i-3tNkx9UTW18FaMM8dCg8By_kCoF/view?usp=drive_web>
 mtdblock15
<https://drive.google.com/file/d/1pG_qHtRyX77631dydZMgnNYOTHJ9Lnr7/view?usp=drive_web>
 mtdblock17
<https://drive.google.com/file/d/1JHBL8WIAecItSuCYgEob7IVnz8qE78ug/view?usp=drive_web>
 mtdblock22
<https://drive.google.com/file/d/1lIWETEXDz56s6MA4EqnYacYozZrjP1F0/view?usp=drive_web>

TonyTeng22, 2024-06-25:
内核版本：Linux OpenWrt 4.4.60 #5 SMP Thu Jan 19 09:30:40 CST 2023 armv7l
GNU/Linux

 mtdblock20
<https://drive.google.com/file/d/1QFq5xDBva8ybnymPIXV3V2l95B8t-SGc/view?usp=drive_web>

 mtdblock19
<https://drive.google.com/file/d/1Q25mBmzNKMuvU5wh92MPc7cZCUopgNrt/view?usp=drive_web>

metalbug, 2024-06-26:
太棒了，终于有人搞这个了，30买了个二手的，又花了10块钱进群，刷了个什么NWRT，结果使用体验非常差，挺恶心的。求大佬适配CMCC RAX3000Q/QY

TonyTeng22, 2024-07-05:
> 麻烦提供下原厂rom。

大佬，CMCC AX3000Q/QY 分析的怎么样？ 值得研究吗？

hzyitc, 2024-07-07:
> > 麻烦提供下原厂rom。
> 
> 大佬，CMCC AX3000Q/QY 分析的怎么样？ 值得研究吗？

需要时间，最近工作比较忙

metalbug, 2024-07-07:
> > > 麻烦提供下原厂rom。
> > 
> > 
> > 大佬，CMCC AX3000Q/QY 分析的怎么样？ 值得研究吗？
> 
> 需要时间，最近工作比较忙

加油大佬，就等你了！

TonyTeng22, 2024-07-16:
@hzyitc 有空看下 CMCC AX3000Q/QY ，是否还需要提供其他文件？

hzyitc, 2024-07-17:
dmesg只有一小节，不清楚用的是哪个DT。需要完整的串口启动日志。

```
strings /proc/device-tree/model
strings /proc/device-tree/compatible
```

以及有线拓扑：

```
swconfig switch1 show
cat /etc/config/network
```

lo5a, 2024-07-18:
root@OpenWrt:~# swconfig switch1 show
swconfig list
swconfig dev <dev> [port <port>|vlan <vlan>] (help|set <key> <value>|get <key>|load <config>|show)
root@OpenWrt:~# cat /etc/config/network

config interface 'loopback'
        option ifname 'lo'
        option proto 'static'
        option ipaddr '127.0.0.1'
        option netmask '255.0.0.0'

config globals 'globals'
        option ula_prefix 'auto'

config switch
        option name 'switch0'

config interface 'lan'
        option force_link '1'
        option type 'bridge'
        option netmask '255.255.255.0'
        option ip6assign '60'
        option multicast_querier '0'
        option hostname 'RAX3000-QY-FB14'
        option ipaddr '192.168.10.1'
        option igmp_snooping '1'
        option ieee1905managed '1'
        option ifname 'eth1.2 eth1.4 eth1.5'
        option proto 'static'

config alias 'lan6'
        option interface 'lan'
        option proto 'dhcpv6'
        option reqaddress 'try'
        option reqprefix 'auto'
        option disabled '1'

config interface 'wan'
        option hostname 'RAX3000-QY-FB14'
        option ifname 'eth0'
        option disabled '0'
        option ipaddr '10.232.XX.XXX'
        option netmask '255.255.255.0'
        option gateway '10.232.XX.XXX'
        option proto 'static'
        option peerdns '0'
        list dns '114.114.114.114'

config interface 'wan6'
        option ifname 'eth0'
        option disabled '0'
        option sourcefilter '0'
        option defaultroute '1'
        option fakeroute '1'
        option peerdns '1'
        option reqaddress 'try'
        option reqprefix 'auto'
        option noslaaconly '0'
        option ip6gw 'fe80::1'
        option proto 'static'
        option ip6prefix '2001:db8:1::1/64'
        option ip6addr 'fe80::96be:9ff:fe50:fb14/64'
        list dns 'fe80::1'

config switch
        option name 'switch1'
        option reset '1'
        option enable_vlan '1'

config switch_vlan
        option device 'switch1'
        option vlan '1'
        option ports '3 1'

config switch_vlan
        option device 'switch1'
        option vlan '2'
        option ports '6t 2'

config switch_vlan
        option device 'switch1'
        option vlan '4'
        option ports '6t 4'

config switch_vlan
        option device 'switch1'
        option vlan '5'
        option ports '6t 5'

lo5a, 2024-07-18:
root@OpenWrt:~# cat /etc/config/network

config interface 'loopback'
        option ifname 'lo'
        option proto 'static'
        option ipaddr '127.0.0.1'
        option netmask '255.0.0.0'

config globals 'globals'
        option ula_prefix 'auto'

config switch
        option name 'switch0'

config interface 'lan'
        option force_link '1'
        option type 'bridge'
        option netmask '255.255.255.0'
        option ip6assign '60'
        option multicast_querier '0'
        option hostname 'RAX3000-QY-FB14'
        option ipaddr '192.168.10.1'
        option igmp_snooping '1'
        option ieee1905managed '1'
        option ifname 'eth1.2 eth1.4 eth1.5'
        option proto 'static'

config alias 'lan6'
        option interface 'lan'
        option proto 'dhcpv6'
        option reqaddress 'try'
        option reqprefix 'auto'
        option disabled '1'

config interface 'wan'
        option hostname 'RAX3000-QY-FB14'
        option ifname 'eth0'
        option disabled '0'
        option ipaddr '10.232.XX.XXX'
        option netmask '255.255.255.0'
        option gateway '10.232.XX.XXX'
        option proto 'static'
        option peerdns '0'
        list dns '114.114.114.114'

config interface 'wan6'
        option ifname 'eth0'
        option disabled '0'
        option sourcefilter '0'
        option defaultroute '1'
        option fakeroute '1'
        option peerdns '1'
        option reqaddress 'try'
        option reqprefix 'auto'
        option noslaaconly '0'
        option ip6gw 'fe80::1'
        option proto 'static'
        option ip6prefix '2001:db8:1::1/64'
        option ip6addr 'fe80::96be:9ff:fe50:fb14/64'
        list dns 'fe80::1'

config switch
        option name 'switch1'
        option reset '1'
        option enable_vlan '1'

config switch_vlan
        option device 'switch1'
        option vlan '1'
        option ports '3 1'

config switch_vlan
        option device 'switch1'
        option vlan '2'
        option ports '6t 2'

config switch_vlan
        option device 'switch1'
        option vlan '4'
        option ports '6t 4'

config switch_vlan
        option device 'switch1'
        option vlan '5'
        option ports '6t 5'

lo5a, 2024-07-18:
root@OpenWrt:~# strings /proc/device-tree/model
Qualcomm Technologies, Inc. IPQ5018/AP-MP02.1

lo5a, 2024-07-18:
root@OpenWrt:~# strings /proc/device-tree/compatible
qcom,ipq5018-mp02.1
qcom,ipq5018

lo5a, 2024-07-18:
root@OpenWrt:~# dmesg
[  292.655321] wlan: [1070:I:ANY] ol_ath_vap_set_param: he_bf_cap=0x33
[  292.669887] wlan: [1070:I:ANY] ol_ath_vap_set_param: VDEV params: AC/VHT sounding mode:HE|SU/MU sounding mode:SU|Trig/Non-Trig sounding mode:Non-Trigged
[  292.676144] wlan: [1070:I:ANY] MBO Initialized
[  292.689932] wlan: [1070:I:ANY] OCE Initialized
[  292.694159] wlan: [1070:I:ANY] osif_nss_ol_vap_create: NSS wifi offload VAP create IF 33 nss_id -1
[  292.698614] wlan: [1070:I:ANY] osif_nss_ol_vap_create: NSS radio_if 32
[  292.707682] wlan: [0:I:ANY] osif_nss_vdev_cfg_callback: VDEV configuration success: 0
[  292.714132] wlan: [1070:I:ANY] osif_nss_ol_vap_create: vap create 84de7480 : if_num 33
[  292.737254] wlan: [1070:I:ANY] osif_create_vap_complete: TX Checksum:1|SG:1|TSO:1|LRO:0
[  292.737289] wlan: [1070:I:ANY] WLAN-NSS: VAP NSS ops initialized
[  292.745426] wlan: [1070:I:ANY] VAP device ath0 created osifp: (84de7480) os_if: (8628c000)
[  293.512005] wlan: [1186:I:ANY] osif_nss_vdev_set_cfg: setting me mode 6 target type 29
[  293.512037] wlan: [1186:I:ANY] osif_nss_vdev_set_cfg: Mcast command 6
[  293.709366] wlan: [1215:I:ANY] ieee80211_ucfg_setparam: Setting Max Stations:64
[  294.367837] 8021q: adding VLAN 0 to HW filter on device ath0
[  294.370230] device ath0 entered promiscuous mode
[  294.372679] br-lan: port 4(ath0) entered forwarding state
[  294.377312] br-lan: port 4(ath0) entered forwarding state
[  294.448230] wlan: [917:I:ANY] wlan_cfg80211_do_acs: vap-0(ath0):ACS Params
[  294.448262] wlan: [917:I:ANY] wlan_cfg80211_do_acs: ht_enabled:1|ht40_enabled:0|vht_enabled:1|hw_mode:23|chwidth:20|
[  294.792282] wlan: [1450:I:ANY] ol_ath_ucfg_setparam: Starting the target before sending the command
[  294.792342] wlan: [1450:I:ANY] ol_ath_wifi_ssr: ol_ath_wifi_ssr: SSR event 2 bus_type 1
[  294.800261] remoteproc remoteproc2: powering up q6v5_wcss_userpd2
[  294.811704] Subsystem error monitoring/handling services are up
[  294.814396] remoteproc remoteproc2: remote processor q6v5_wcss_userpd2 is now up
[  294.820224] wlan: [1450:I:ANY] ol_ath_wifi_ssr: ol_ath_wifi_ssr: SSR event 3 bus_type 1
[  294.827816] wlan: [1450:I:ANY] ol_ath_target_start: subsystem_get success
[  294.835720] cnss[41]: INFO: Waiting for FW ready. Device: 0xfffb, FW ready timeout: 15 seconds
[  294.887315] cnss[41]: INFO: QMI WLFW service connected, state: 0x1
[  294.887707] cnss[41]: INFO: device_id : 65531 mem mode : [2]
[  294.892394] cnss[41]: INFO: platform name: Qualcomm Technologies, Inc. IPQ5018/AP-MP02.1
[  294.899890] cnss[41]: INFO: Target capability: chip_id: 0x0, chip_family: 0x4, board_id: 0xff, soc_id: 0xffffffff, fw_version: 0x25087fff, fw_build_timestamp: 2021-09-16 02:40, otp_version: 0x0 eeprom_caldata_read_timeout 0s
[  294.906491] cnss[41]: INFO: Device BAR Info pa: 0x81e00000, va: 0x9a980000, size: 0x200000
[  294.926244] cnss[41]: INFO: Using Boardid from bootargs:0x60
[  294.935033] cnss[41]: INFO: BDF location : 0x4d300000
[  294.940120] cnss[41]: INFO: BDF qcn6122/bdwlan.b60 size 131072
[  294.947309] cnss[41]: INFO: per device BDF location : 0x4d320000
[  294.950886] cnss[41]: INFO: CALDATA qcn6122/caldata_1.bin size 131072 offset 0x20000
[  295.057095] cnss[41]: INFO: FW ready received for device 0xfffb
[  295.057230] wlan: [1450:I:ANY] pld_ahb_wlan_enable: cnss_mode: Mission mode
[  295.062244] cnss[41]: INFO: Sending mode message, mode: MISSION(0), state: 0x7
[  295.078932] wlan: [1450:I:ANY] qca_napi_create: qca_napi_create: napi instance 0 created on pipe 4
[  295.078998] wlan: [1450:I:ANY] htc_wmi_init: HT Create . 85997000
[  295.086863] wlan: [1450:I:ANY] htc_wmi_init: Skipping BMI Done
[  295.093022] wlan: [1450:I:ANY] htc_wmi_init: host_enable 0
[  295.108662] wlan: [1450:I:ANY] ol_target_init_complete: CDP soc attach success
[  295.108696] wlan: [1450:I:ANY] ol_target_init_complete: Soc attach success NSS config 0
[  295.115006] wlan: [1450:I:ANY] htc_set_async_ep: htc_set_async_ep: htc_handle 85997000, ep 2, value 1
[  295.125083] wlan: [533:I:TIF] HE_CAP Info = 9a08010d
[  295.132240] wlan: [533:I:TIF] Num HW modes = 1
[  295.137265] wlan: [533:I:TIF] Num PHY = 1
[  295.141432] wlan: [533:I:TIF] FW Build Version (Ext.) = 0
[  295.145511] wlan: [533:I:TIF] HDL Version: MajorVersion.MinorVersion = 0.0
[  295.150972] wlan: [533:I:TIF] Preferred HW Mode = 0 Num Radios = 1
[  295.157834] wlan: [533:W:TIF] add_11ax_mode_flags: 11AX wireless modes are not populated by the target in reg capabilities, adding 11AX wireless modes
[  295.163843] wlan: [533:I:TIF] init_deinit_add_11ax_modes: hw_mode_id = 0 phy_id = 0 wireless modes = 0
[  295.177380] wlan: [533:I:TIF] init_deinit_get_total_num_ema_vaps: num_radios: 1
[  295.186576] wlan: [533:I:TIF] init_deinit_get_total_num_ema_vaps: ema_max_vap_cnt: 0
[  295.193833] wlan: [533:I:TIF] init_deinit_get_total_num_vdevs_for_soc: num_vdevs: 9
[  295.201802] wlan: [533:I:TIF] init_deinit_get_total_num_ema_vaps: num_radios: 1
[  295.209146] wlan: [533:I:TIF] init_deinit_get_total_num_ema_vaps: ema_max_vap_cnt: 0
[  295.216423] wlan: [533:I:TIF] init_deinit_get_total_num_vdevs_for_soc: num_vdevs: 9
[  295.224429] wlan: [533:I:TIF] init_deinit_get_total_num_ema_vaps: num_radios: 1
[  295.231801] wlan: [533:I:TIF] init_deinit_get_total_num_ema_vaps: ema_max_vap_cnt: 0
[  295.239108] wlan: [533:I:TIF] init_deinit_get_total_num_vdevs_for_soc: num_vdevs: 9
[  295.247085] wlan: [533:I:TIF] init_deinit_get_total_num_ema_vaps: num_radios: 1
[... cut ...]
[  302.879892] wlan: [0:I:ANY] wlan_acs_start_scan_report: [EXT] Invoking ACS module for ACS report
[  319.793540] wlan: [3934:I:ANY] ieee80211_ioctl_siwscan: preempt_scan
[  319.823257] wlan: [3940:I:ANY] ieee80211_ioctl_siwscan: preempt_scan
[14567.493200] sh (474): drop_caches: 3
[16119.200643] port_id:5 port_link:1 speed:2 duplex:1
[16416.678231] port_id:5 port_link:0 speed:0 duplex:0
[38293.938321] port_id:1 port_link:0 speed:0 duplex:0
[39831.850263] port_id:1 port_link:1 speed:1 duplex:1

lo5a, 2024-07-18:
@hzyitc 不知道是否满足您的要求。

hzyitc, 2024-07-18:
@lo5a 可以运行下下面的命令，然后插拔下每个网口，看看各个网口对应的交换片的哪个口吗？

```
while true; do clear; swconfig dev switch1 show | grep link; sleep 1; done
```

根据提供的network，它交换片连线有点奇怪，好像 1是WAN，2 4 5是LAN，3 6是CPU

hzyitc, 2024-07-18:
dmesg信息还是不完整。最好是能用串口记录，从uboot到系统完整启动的所有日志。

不过这个现在暂时可以不用。

但最后测试时还是需要串口兜底的，在启动失败或者网口配置错误时，可以救砖。

lo5a, 2024-07-19:
[image]

lo5a, 2024-07-19:
1 2 4 5确实如您所说，3 6就不懂了。

lo5a, 2024-07-19:
我没有拆机，没用ttl解ssh，所以没有串口记录，其他大佬能否提供一下呢?

lo5a, 2024-07-19:
众人拾柴火焰高，希望大家能踊跃参与，让 @hzyitc 大佬能早日出包，造福RAX3000QY机主，感谢hzyitc大佬的辛勤付出！

TonyTeng22, 2024-07-22:
拆机了一个，完整串口打印如下：
 

U-Boot 2016.01 (Jan 26 2022 - 18:11:36 +0800), Build: jenkins-RAX3000Q-76

DRAM:  smem ram ptable found: ver: 1 len: 4
256 MiB
USB: Node Not found, skipping initiaPIC controller support serial NAND
ID = 21e521e5
Vendor = e5
Device = 21
Serial Nand Device Found With ID : 0xe5 0x21
Serial NAND device Manufature:DS35M1GA-IB
Device Size:128 MiB, Page size:2048, Spare Size:64, ECC:4-bit
SF: Unsupported flash IDs: manuf 00, jedec 0000, ext_jedec 0000
ipq_spi: SPI Flash not found (bus/cs/speed/mode) = (0/0/48000000/0)
128 MiB
MMC:   sdhci: Node Not found, skipping initialization

*** Warning - bad CRC, using default environment

PCI1 is not defined in the device tree
In:    serial@78AF000
Out:   serial@78AF000
Err:   serial@78AF000
machid: 8040000
eth0 MAC Address from ART is not valid
eth1 MAC Address from ART is not valid
Hit any key to stop autoboot:  0 
ubi0: attaching mtd1
ubi0: scanning is finished
ubi0: attached mtd1 (name "mtd=0", size 48 MiB)
ubi0: PEB size: 131072 bytes (128 KiB), LEB size: 126976 bytes
ubi0: min./max. I/O unit sizes: 2048/2048, sub-page size 2048
ubi0: VID header offset: 2048 (aligned 2048), data offset: 4096
ubi0: good PEBs: 384, bad PEBs: 0, corrupted PEBs: 0
ubi0: user volume: 5, internal volumes: 1, max. volumes count: 128
ubi0: max/mean erase counter: 52/16, WL threshold: 4096, image sequence number: 294337820
ubi0: available PEBs: 0, total reserved PEBs: 384, PEBs reserved for bad PEB handling: 20
Read 0 bytes from volume kernel to 44000000
No size specified -> Using max size (3132792)
## Loading kernel from FIT Image at 44000000 ...
   Using 'config@mp02.1' configuration
   Trying 'kernel@1' kernel subimage
     Description:  ARM OpenWrt Linux-4.4.60
     Type:         Kernel Image
     Compression:  lzma compressed
     Data Start:   0x440000e4
     Data Size:    1702507 Bytes = 1.6 MiB
     Architecture: ARM
     OS:           Linux
     Load Address: 0x41208000
     Entry Point:  0x41208000
     Hash algo:    crc32
     Hash value:   ba6c0e89
     Hash algo:    sha1
     Hash value:   be8c07326178a5e20d9ba5ed8ae191c1f38e264f
   Verifying Hash Integrity ... crc32+ sha1+ OK
## Loading fdt from FIT Image at 44000000 ...
   Using 'config@mp02.1' configuration
   Trying 'fdt@mp02.1' fdt subimage
     Description:  ARM OpenWrt qcom-ipq50xx-mpxx device tree blob
     Type:         Flat Device Tree
     Compression:  uncompressed
     Data Start:   0x44203950
     Data Size:    59736 Bytes = 58.3 KiB
     Architecture: ARM
     Hash algo:    crc32
     Hash value:   eeefe877
     Hash algo:    sha1
[... 791 lines cut ...]
[   63.522243] wlan: [829:I:ANY] desired hw mode: 31
[   63.526848] wlan: [829:I:ANY] ieee80211_ucfg_set_freq_internal: 
[   63.526848]  Channel is configured already!!
[   63.531397] wlan: [829:E:MBSSIE] ieee80211_ucfg_set_txvap: MBSSID is not enabled
[   63.541703] wlan: [0:I:ANY] wlan_acs_start_scan_report: [EXT] Invoking ACS module for ACS report
****** Starting QDSS for qcn6122 pci0 ********
****** Starting QDSS for qcn6122 pci1 ********
***** QDSS Tracing Configuration completed *******
****** Starting QDSS for Integrated ********
[   64.397337] br-lan: port 7(ath16) entered forwarding state
[   64.997320] br-lan: port 5(ath06) entered forwarding state
[   69.082840] br-lan: port 7(ath16) entered disabled state
[   72.100783] br-lan: port 7(ath16) entered forwarding state
[   72.100899] br-lan: port 7(ath16) entered forwarding state
[   72.105262] 8021q: adding VLAN 0 to HW filter on device ath16
[   72.111921] wlan: [829:I:ANY] DES SSID SET=xxxx-5G_WiFi5
[   72.116829] wlan: [829:I:ANY] desired hw mode: 31
[   72.122286] wlan: [829:I:ANY] ieee80211_ucfg_set_freq_internal: 
[   72.122286]  Channel is configured already!!
[   72.126606] wlan: [829:E:MBSSIE] ieee80211_ucfg_set_txvap: MBSSID is not enabled
[   72.136908] wlan: [0:I:ANY] wlan_acs_start_scan_report: [EXT] Invoking ACS module for ACS report
[   74.097328] br-lan: port 7(ath16) entered forwarding state

OpenWrt login: root
Password:

TonyTeng22, 2024-07-22:
Uboot env 以及支持的命令如下 ：
IPQ5018# printe 
baudrate=115200
bootargs=console=ttyMSM0,115200n8 cnss2.bdf_integrated=0x24 cnss2.bdf_pci0=0x60 cnss2.bdf_pci1=0x60 cnss2.skip_radio_bmap=4
bootcmd=bootipq
bootdelay=5
eth1addr=00:11:22:33:44:56
ethact=eth0
ethaddr=00:11:22:33:44:55
fdt_high=0x4A400000
fdtcontroladdr=4a9c4004
flash_type=11
ipaddr=192.168.10.10
machid=8040000
netmask=255.255.255.0
serverip=192.168.10.19
soc_hw_version=20180101
soc_version_major=1
soc_version_minor=1
stderr=serial@78AF000
stdin=serial@78AF000
stdout=serial@78AF000

Environment size: 542/262140 bytes
IPQ5018# ?
?       - alias for 'help'
ar8xxx_dump- Dump ar8xxx registers
base    - print or set address offset
bdinfo  - print Board Info structure
bootelf - Boot from an ELF image in memory
bootipq - bootipq from flash device
bootm   - boot application image from memory
bootp   - boot image via network using BOOTP/TFTP protocol
bootvx  - Boot vxWorks from an ELF image
bootz   - boot Linux zImage image from memory
canary  - test stack canary
chpart  - change active partition
cmp     - memory compare
coninfo - print console devices and information
cp      - memory copy
crc32   - checksum calculation
dhcp    - boot image via network using DHCP/TFTP protocol
dm      - Driver model low level access
echo    - echo args to console
editenv - edit environment variable
env     - environment handling commands
erase   - erase FLASH memory
exectzt - execute TZT

exit    - exit script
false   - do nothing, unsuccessfully
fatinfo - print information about filesystem
fatload - load binary file from a dos filesystem
fatls   - list files in a directory (default /)
fatsize - determine a file's size
fatwrite- write file into a dos filesystem
fdt     - flattened device tree utility commands
flash   - flash part_name 
        flash part_name load_addr file_size 

flasherase- flerase part_name 

flinfo  - print FLASH memory information
fuseipq - fuse QFPROM registers from memory

go      - start application at address 'addr'
help    - print command description/usage
i2c     - I2C sub-system
imxtract- extract a part of a multi-image
ipq5018_mdio- IPQ5018 mdio utility commands
ipq_mdio- IPQ mdio utility commands
is_sec_boot_enabled- check secure boot fuse is enabled or not

itest   - return true/false on integer compare
loop    - infinite loop on address range
md      - memory display
mii     - MII utility commands
mm      - memory modify (auto-incrementing address)
mmc     - MMC sub system
mmcinfo - display MMC info
mtdparts- define flash/nand partitions
mtest   - simple RAM read/write test
mw      - memory write (fill)
nand    - NAND sub-system
nboot   - boot from NAND device
nfs     - boot image via network using NFS protocol
nm      - memory modify (constant address)
part    - disk partition related commands
pci     - list and access PCI Configuration Space
ping    - send ICMP ECHO_REQUEST to network host
printenv- print environment variables
protect - enable or disable FLASH write protection
reset   - Perform RESET of the CPU
run     - run commands in an environment variable
runmulticore- Enable and schedule secondary cores
saveenv - save environment variables to persistent storage
secure_authenticate- authenticate the signed image

setenv  - set environment variables
setexpr - set environment variable as the result of eval expression
sf      - SPI flash sub-system
showvar - print local hushshell variables
sleep   - delay execution for some time
smeminfo- print SMEM FLASH information
source  - run script from memory
test    - minimal test like /bin/sh
tftpboot- boot image via network using TFTP protocol
tftpput - TFTP put command, for uploading files to a server
true    - do nothing, successfully
tzt     - load and run tzt

uart    - UART sub-system
ubi     - ubi commands
usb     - USB sub-system
usbboot - boot from USB device
version - print monitor, compiler and linker version
zip     - zip a memory region

hzyitc, 2024-07-22:
已推送。

仅限有串口，懂救砖的用户测试。

不保证能启动。

TonyTeng22, 2024-07-23:
有人编译成功的吗？方便发出来，还有刷机方法方便简单写一下吗？

TonyTeng22, 2024-07-23:
测试可以启动，但是mac地址获取失败：

U-Boot 2016.01 (Jan 26 2022 - 18:11:36 +0800), Build: jenkins-RAX3000Q-76

DRAM:  smem ram ptable found: ver: 1 len: 4
256 MiB
USB: Node Not found, skipping initiaPIC controller support serial NAND
ID = 21e521e5
Vendor = e5
Device = 21
Serial Nand Device Found With ID : 0xe5 0x21
Serial NAND device Manufature:DS35M1GA-IB
Device Size:128 MiB, Page size:2048, Spare Size:64, ECC:4-bit
SF: Unsupported flash IDs: manuf 00, jedec 0000, ext_jedec 0000
ipq_spi: SPI Flash not found (bus/cs/speed/mode) = (0/0/48000000/0)
128 MiB
MMC:   sdhci: Node Not found, skipping initialization

PCI1 is not defined in the device tree
In:    serial@78AF000
Out:   serial@78AF000
Err:   serial@78AF000
machid: 8040000
eth0 MAC Address from ART is not valid
eth1 MAC Address from ART is not valid
Hit any key to stop autoboot:  0 
ubi0: attaching mtd1
ubi0: scanning is finished
ubi0: attached mtd1 (name "mtd=0", size 48 MiB)
ubi0: PEB size: 131072 bytes (128 KiB), LEB size: 126976 bytes
ubi0: min./max. I/O unit sizes: 2048/2048, sub-page size 2048
ubi0: VID header offset: 2048 (aligned 2048), data offset: 4096
ubi0: good PEBs: 384, bad PEBs: 0, corrupted PEBs: 0
ubi0: user volume: 3, internal volumes: 1, max. volumes count: 128
ubi0: max/mean erase counter: 2/0, WL threshold: 4096, image sequence number: 1711145382
ubi0: available PEBs: 0, total reserved PEBs: 384, PEBs reserved for bad PEB handling: 20
Read 0 bytes from volume kernel to 44000000
No size specified -> Using max size (4063232)
## Loading kernel from FIT Image at 44000000 ...
   Using 'config@mp02.1' configuration
   Trying 'kernel-1' kernel subimage
     Description:  ARM OpenWrt Linux-5.15.150
     Type:         Kernel Image
     Compression:  lzma compressed
     Data Start:   0x440000e4
     Data Size:    3923210 Bytes = 3.7 MiB
     Architecture: ARM
     OS:           Linux
     Load Address: 0x41208000
     Entry Point:  0x41208000
     Hash algo:    crc32
     Hash value:   50b70ec0
     Hash algo:    sha1
     Hash value:   1aefc0e7a1564d1aeaf55668cba213cddd87286c
   Verifying Hash Integrity ... crc32+ sha1+ OK
## Loading fdt from FIT Image at 44000000 ...
   Using 'config@mp02.1' configuration
   Trying 'fdt-1' fdt subimage
     Description:  ARM OpenWrt cmcc_rax3000q device tree blob
     Type:         Flat Device Tree
     Compression:  uncompressed
     Data Start:   0x443bdf28
     Data Size:    17329 Bytes = 16.9 KiB
     Architecture: ARM
     Hash algo:    crc32
     Hash value:   6535fdaf
     Hash algo:    sha1
     Hash value:   c34853a078887d5905410f3d5b795a6af2c96b2b
   Verifying Hash Integrity ... crc32+ sha1+ OK
   Booting using the fdt blob at 0x443bdf28
[... 663 lines cut ...]
Tx1024Byte  : 0
Tx1518Byte  : 0
TxMaxByte   : 0
TxOverSize  : 0
TxByte      : 34648
TxCollision : 0
TxAbortCol  : 0
TxMultiCol  : 0
TxSingleCol : 0
TxExcDefer  : 0
TxDefer     : 0
TxLateCol   : 0
RxUniCast   : 20
TxUniCast   : 16

        enable_eee: ???
        pvid: 1
        link: port:6 link:up speed:1000baseT full-duplex 
VLAN 1:
        vid: 1
        ports: 2 4 5 6 
VLAN 2:
        vid: 2
        ports: 1 3 
root@OpenWrt:/#

TonyTeng22, 2024-07-23:
[image]
[image]

TonyTeng22, 2024-07-23:
[image]

lo5a, 2024-07-23:
等待 uboot刷机，感谢大佬的付出！

hzyitc, 2024-07-23:
无线BDF参数写错了，已修复。

ligehhh, 2024-07-23:
大佬 有时间可以技术交流一下吗

TonyTeng22, 2024-07-24:
> 无线BDF参数写错了，已修复。

23日的 Release 中没有生成CMCC AX3000Q 的文件，是不是自动编译出错了 ？

TonyTeng22, 2024-07-24:
问题似乎没解决，没有无线， mac不对
[image]

TonyTeng22, 2024-07-24:
[image]

TonyTeng22, 2024-07-25:
官方的固件不是从uboot环境变了读取的，uboot 以及 env 没有清除过，官方应该是从配置文件读取的

TonyTeng22, 2024-07-25:
[image]
[image]

TonyTeng22, 2024-07-25:
[image]

TonyTeng22, 2024-07-25:
再次感谢 hzyitc 的辛勤付出，CMCC RAX3000Q/QY 可以刷机成功，除了MAC无法自动读取外其他功能都正常。
注意：Wifi 默认关闭，需要手动打开。
固件地址：[下载地址](https://github.com/hzyitc/openwrt-redmi-ax3000/releases/download/ci-ipq50xx-mainline-kernel-5.15-openwrt-23.05-20240724-153345-UTC/openwrt-ipq50xx-arm-cmcc_rax3000q-squashfs-nand-factory.ubi)
-------------------------  刷机步骤  --------------------------
setenv ethaddr '替换成设备标签上的MAC地址'
setenv eth1addr '替换成设备标签上的MAC地址'
setenv ipaddr 192.168.1.2
setenv serverip 192.168.1.1
tftpboot openwrt-ipq50xx-arm-cmcc_rax3000q-squashfs-nand-factory.ubi
flash rootfs_1
flash rootfs
saveenv
reset
-----------------------------  end -------------------------------------

[openwrt_klog.txt](https://github.com/user-attachments/files/16370174/openwrt_klog.txt)

metalbug, 2024-07-25:
点赞，支持，有空试一下，感谢大佬付出

lo5a, 2024-07-29:
@TonyTeng22  wifi 5g的速率能达到多少?和原厂固件相比差很多么?

TonyTeng22, 2024-07-30:
@lo5a 
[image]

TonyTeng22, 2024-07-30:
@hzyitc 编译时可以提供sysupgrade固件吗？这样就不需要每次都拆机串口刷机了。如果能提供不死uboot就更好了

hzyitc, 2024-07-30:
其他分支也更新了，麻烦测试下

hzyitc, 2024-07-30:
> @hzyitc 编译时可以提供sysupgrade固件吗？这样就不需要每次都拆机串口刷机了。如果能提供不死uboot就更好了

sysupgrade直接用ubi包

当前不可用是因为我没做适配。得研究下这个设备的ab升级流程。

TonyTeng22, 2024-07-30:
k4.4 CMCC RAX3000Q 刷入无限重启
[k4.4_keepingRestart.txt](https://github.com/user-attachments/files/16423593/k4.4_keepingRestart.txt)

lo5a, 2024-07-30:
> > @hzyitc 编译时可以提供sysupgrade固件吗？这样就不需要每次都拆机串口刷机了。如果能提供不死uboot就更好了
> 
> sysupgrade直接用ubi包
> 
> 当前不可用是因为我没做适配。得研究下这个设备的ab升级流程。

大佬最好能够把不拆机刷机包研究出来，便于大伙刷机测试，为大佬点赞！

TonyTeng22, 2024-07-30:
k5.4正式版测试WiFi异常， 没有加载私有驱动，无5G热点
[IPQ5018_klog_5.4 .txt](https://github.com/user-attachments/files/16424428/IPQ5018_klog_5.4.txt)

TonyTeng22, 2024-07-30:
k5.4pre 预览版 5G有160M设置选项但是，设置后无热点 ，80M 正常
[IPQ5018_k54pre_160Mno-ok.txt](https://github.com/user-attachments/files/16424566/IPQ5018_k54pre_160Mno-ok.txt)

TonyTeng22, 2024-07-30:
截止7.30日目前能用的版本是：20240724153345 的 [5.15 pre-release](https://github.com/hzyitc/openwrt-redmi-ax3000/releases/download/ci-ipq50xx-mainline-kernel-5.15-openwrt-23.05-20240724-153345-UTC/openwrt-ipq50xx-arm-cmcc_rax3000q-squashfs-nand-factory.ubi)

metalbug, 2024-07-31:
其实2.4g可以不要，我用路由器，都是把2.4g关掉，还能减少发热。
希望加入nss，希望5g信号好一点，希望可以中继

[uboot.zip](https://github.com/user-attachments/files/16436017/uboot.zip)
mtd write /tmp/rax3000qy_uboot.mbn /dev/mtd11
mtd write /tmp/rax3000qy_mibib.bin /dev/mtd1

这里有个uboot可以直接刷入，但是刷完再进uboot就进不去了，不知道什么原因

liufanl, 2024-08-01:
> @lo5a 可以运行下下面的命令，然后插拔下每个网口，看看各个网口对应的交换片的哪个口吗？
> 
> ```
> while true; do clear; swconfig dev switch1 show | grep link; sleep 1; done
> ```
> 
> 根据提供的network，它交换片连线有点奇怪，好像 1是WAN，2 4 5是LAN，3 6是CPU。（3 6 可能说反了）

大佬，经过你的分析，这个交换片连线具体是怎么连的啊？1是WAN，245是LAN，036分别连的什么？理论上0应该是CPU。现在这个设备刷了其他相同硬件配置的固件，可能会出现环路，是不是应该和这个交换机连接有关啊？？

liufanl, 2024-08-01:
> 再次感谢 hzyitc 的辛勤付出，CMCC RAX3000Q/QY 可以刷机成功，除了MAC无法自动读取外其他功能都正常。 注意：Wifi 默认关闭，需要手动打开。 [...]

eth1addr 的地址应该是标签上的地址+1 ，不然你的LAN和WAN的MAC地址会是一样的。

TonyTeng22, 2024-08-02:
> eth1addr 的地址应该是标签上的地址+1 ，不然你的LAN和WAN的MAC地址会是一样的。

跨网段mac层是隔离的，一个mac足矣

tangmuyang, 2024-08-03:
> 其实2.4g可以不要，我用路由器，都是把2.4g关掉，还能减少发热。 希望加入nss，希望5g信号好一点，希望可以中继
> 
> [uboot.zip](https://github.com/user-attachments/files/16436017/uboot.zip) mtd write /tmp/rax3000qy_uboot.mbn /dev/mtd11 mtd write /tmp/rax3000qy_mibib.bin /dev/mtd1
> 
> 这里有个uboot可以直接刷入，但是刷完再进uboot就进不去了，不知道什么原因

感谢分享uboot，nwrt固件是基于qsdk的32位系统，qcawifi闭源驱动，一般使用应该可以吧，真要跑一些服务，内存太小了。

LoadingZhang, 2024-08-11:
> 其实2.4g可以不要，我用路由器，都是把2.4g关掉，还能减少发热。 希望加入nss，希望5g信号好一点，希望可以中继
> 
> [uboot.zip](https://github.com/user-attachments/files/16436017/uboot.zip) mtd write /tmp/rax3000qy_uboot.mbn /dev/mtd11 mtd write /tmp/rax3000qy_mibib.bin /dev/mtd1
> 
> 这里有个uboot可以直接刷入，但是刷完再进uboot就进不去了，不知道什么原因

@metalbug 我刷了之后开机长按reset会卡启动，需要连ttl输入httpd才能进入刷机界面

TonyTeng22, 2024-08-24:
> 其实2.4g可以不要，我用路由器，都是把2.4g关掉，还能减少发热。 希望加入nss，希望5g信号好一点，希望可以中继
> 
> [uboot.zip](https://github.com/user-attachments/files/16436017/uboot.zip) mtd write /tmp/rax3000qy_uboot.mbn /dev/mtd11 mtd write /tmp/rax3000qy_mibib.bin /dev/mtd1
> 
> 这里有个uboot可以直接刷入，但是刷完再进uboot就进不去了，不知道什么原因

此 uboot 的 bootargs 有误 需改为: 
setenv bootargs console=ttyMSM0,115200n8 cnss2.bdf_integrated=0x24 cnss2.bdf_pci0=0x60 cnss2.bdf_pci1=0x60 cnss2.skip_radio_bmap=4
save;reset

TonyTeng22, 2024-08-24:
刷好不死uboot后，推荐一个openwrt.ai 的固件：https://dl.openwrt.ai/firmare/ipq50xx-arm/cmcc_rax3000q/

lo5a, 2024-08-25:
> 刷好不死uboot后，推荐一个openwrt.ai 的固件：https://dl.openwrt.ai/firmware/ipq50xx-arm/cmcc_rax3000q/08.23.2024-kwrt-ipq50xx-arm-cmcc_rax3000q-squashfs-nand-factory.ubi

感谢分享！有改好的不死uboot吗?这个固件稳定吗?无线的速度如何?

TonyTeng22, 2024-08-27:
> > 刷好不死uboot后，推荐一个openwrt.ai 的固件：https://dl.openwrt.ai/firmware/ipq50xx-arm/cmcc_rax3000q/08.23.2024-kwrt-ipq50xx-arm-cmcc_rax3000q-squashfs-nand-factory.ubi
> 
> 感谢分享！有改好的不死uboot吗?这个固件稳定吗?无线的速度如何?

速度一般吧：300 - 400mbps

[image]

hzyitc, 2024-08-27:
5.15没有nss，吃cpu的，速度慢是正常现象

[37 more replies]