hzyitc/openwrt-redmi-ax3000 #79: 请教如何添加 SPI NAND 的支持？

sseseed, 2024-07-24:
看到你也在研究 IPQ5018，我最近也在搞这个，但我水平实在太业余，网上找到一些源码算是编译出来了，但因为试的机子是 PZ-L8，使用的是 ESMT F50D1G41LB 闪存，启动时就识别不正常，前一句还正常识别nand: device found, Manufacturer ID: 0xa1, Chip ID: 0xa5，后面就是nand: unknown nand 2gib 1,8v 8-bit这样的错误信息，原厂固件显示的正常的应该是nand: 128 MiB, SLC, erase size: 128 KiB, page size: 2048, OOB size: 128。所以考虑到是不是源码不支持这个闪存。请问怎么添加这个闪存的支持？

hzyitc, 2024-07-24:
F50D1G41LB 不是这个ID。

请拆机确认你板子上用的NAND芯片

sseseed, 2024-07-25:
> F50D1G41LB 不是这个ID。
> 
> 请拆机确认你板子上用的NAND芯片

首先，谢谢你的回复。我对比过原厂固件启动信息，的确也是显示 nand: device found, Manufacturer ID: 0xa1, Chip ID: 0xa5，闪存型号我没认真拆开看，因为有散热片和屏蔽压着所以没拆到底，只是看网上别人的拆解图的确是F50D1G41LB ，但现在自己编译的固件和原厂固件都识别出这个 ID，应该也没问题吧。就是后面具体参数那一行识别错了，启动的时候过了这几行就是出分区表的了。所以我就感觉会不会是原厂固件有添加这个闪存的信息能支持它，而我用的源码并不支持这个闪存。用这种 SPI NAND 就是省成本的结果，如果是用大片多针脚那种常见的 NAND，估计兼容性就强得多了。

potatoigit, 2024-07-25:
应该是0x11

sseseed, 2024-07-27:
补充一下 TTL 启动信息：
Hit any key to stop autoboot:  0
ubi0: attaching mtd1
ubi0: scanning is finished
ubi0: volume 2 ("rootfs_data") re-sized from 9 to 362 LEBs
ubi0: attached mtd1 (name "mtd=0", size 58 MiB)
ubi0: PEB size: 131072 bytes (128 KiB), LEB size: 126976 bytes
ubi0: min./max. I/O unit sizes: 2048/2048, sub-page size 2048
ubi0: VID header offset: 2048 (aligned 2048), data offset: 4096
ubi0: good PEBs: 464, bad PEBs: 0, corrupted PEBs: 0
ubi0: user volume: 3, internal volumes: 1, max. volumes count: 128
ubi0: max/mean erase counter: 4/2, WL threshold: 4096, image sequence number: 2021019266
ubi0: available PEBs: 0, total reserved PEBs: 464, PEBs reserved for bad PEB handling: 20
Read 0 bytes from volume kernel to 44000000
No size specified -> Using max size (3809280)
## Loading kernel from FIT Image at 44000000 ...
   Using 'config@mp02.1' configuration
   Trying 'kernel-1' kernel subimage
     Description:  ARM OpenWrt Linux-5.15.161
     Type:         Kernel Image
     Compression:  lzma compressed
     Data Start:   0x440000e4
     Data Size:    3697970 Bytes = 3.5 MiB
     Architecture: ARM
     OS:           Linux
     Load Address: 0x41208000
     Entry Point:  0x41208000
     Hash algo:    crc32
     Hash value:   77767bed
     Hash algo:    sha1
     Hash value:   2823c8b0479ae599d6b8984cfc3064bb43234cac
   Verifying Hash Integrity ... crc32+ sha1+ OK
## Loading fdt from FIT Image at 44000000 ...
   Using 'config@mp02.1' configuration
   Trying 'fdt-1' fdt subimage
     Description:  ARM OpenWrt pz_pz-l8 device tree blob
     Type:         Flat Device Tree
     Compression:  uncompressed
     Data Start:   0x44386f4c
     Data Size:    17393 Bytes = 17 KiB
     Architecture: ARM
     Hash algo:    crc32
     Hash value:   edf9fb18
     Hash algo:    sha1
     Hash value:   440ad955bf4dd6987661186944cc4e0a3d941b44
   Verifying Hash Integrity ... crc32+ sha1+ OK
   Booting using the fdt blob at 0x44386f4c
   Uncompressing Kernel Image ... OK
   Loading Device Tree to 4a3f8000, end 4a3ff3f0 ... OK
Could not find PCI in device tree
Could not find PCI in device tree
Using machid 0x8040000 from environment

Starting kernel ...

[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 5.15.161 (runner@fv-az1345-867) (arm-openwrt-linux-muslgnueabi-gcc (OpenWrt GCC 13.3.0 r0-73c2119) 13.3.0, GNU ld (GNU Binutils) 2.42) #0 SMP PREEMPT Fri Jul 26 14:19:10 2024
[    0.000000] CPU: ARMv7 Processor [51af8014] revision 4 (ARMv7), cr=10c0383d
[    0.000000] CPU: div instructions available: patching division code
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
[    0.000000] OF: fdt: Machine model: PZ PZ-L8
[    0.000000] OF: fdt: Ignoring memory range 0x40000000 - 0x41000000
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] Zone ranges:
[    0.000000]   Normal   [mem 0x0000000041000000-0x000000004fffffff]
[    0.000000]   HighMem  empty
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000041000000-0x000000004a3fffff]
[    0.000000]   node   0: [mem 0x000000004a400000-0x000000004dffffff]
[... 86 lines cut ...]
[    0.868097] nand: Unknown NAND 2GiB 1,8V 8-bit
[    0.873813] nand: 2048 MiB, SLC, erase size: 64 KiB, page size: 1024, OOB size: 16
[    0.878148] QPIC controller hw version Major:2, Minor:1
[    0.885621] nand: ECC (step, strength) = (512, 4) does not fit in OOB
[    0.885632] qcom-nandc 79b0000.qpic-nand: No valid ECC settings possible
[    0.897468] qcom-nandc: probe of 79b0000.qpic-nand failed with error -28
[    1.287256] i2c_dev: i2c /dev entries driver
[    1.288187] cpufreq: cpufreq_online: CPU0: Running at unlisted initial frequency: 24000 KHz, changing to: 800000 KHz
[    1.290895] sdhci: Secure Digital Host Controller Interface driver
[    1.301181] sdhci: Copyright(c) Pierre Ossman
[    1.307108] sdhci-pltfm: SDHCI platform and OF driver helper
[    1.314536] remoteproc remoteproc0: releasing cd00000.remoteproc
[    1.318486]  cs_system_cfg: CoreSight Configuration manager initialised
[    1.324841] NET: Registered PF_INET6 protocol family
[    1.331359] Segment Routing with IPv6
[    1.334836] In-situ OAM (IOAM) with IPv6
[    1.338471] NET: Registered PF_PACKET protocol family
[    1.342390] 8021q: 802.1Q VLAN Support v1.8
[    1.347526] Registering SWP/SWPB emulation handler
[    1.372389] remoteproc remoteproc0: cd00000.remoteproc is available
[    1.373697] remoteproc remoteproc1: 4ab000.remoteproc_pd1 is available
[    1.379218] remoteproc remoteproc2: cd00000.remoteproc:remoteproc_pd2 is available
[    1.385199] UBI error: cannot open mtd rootfs, error -2
[    1.391833] clk: Disabling unused clocks
[    1.397186] ALSA devic[    1.404238] Waiting for root device /dev/ubiblock0_1...

sseseed, 2024-07-28:
> F50D1G41LB 不是这个ID。
> 
> 请拆机确认你板子上用的NAND芯片

经过一番的瞎探索，终于找到了添加 SPI NAND 的 patch 文件并添加了 FM25LS01 进去，
{"FM25LS01 SPI NAND 1G 1.8V",
		{ .id = {0xa1, 0xa5} },
		SZ_2K, SZ_128, SZ_128K, 0, 2, 128, NAND_ECC_INFO(1, SZ_512) },
但不懂这个结构体，应该有些参数设错了，特别是 ECC 那里，我在  https://www.fmsh.com/62934bb9-d96d-5f04-e1ce-dd9a26dbfebf/ 查到 FM25LS01 的 ECC 是 1bit，所以就填成 1,SZ_512，也不知道对不对。目前能成功到识别分区那里了，但后面读不到数据提示 ECC 错误。部分关键的 TTL 回显如下：

[    0.867862] nand: device found, Manufacturer ID: 0xa1, Chip ID: 0xa5
[    0.867909] nand: Unknown FM25LS01 SPI NAND 1G 1.8V
[    0.873733] nand: 128 MiB, SLC, erase size: 128 KiB, page size: 2048, OOB size: 128
[    0.878408] QPIC controller hw version Major:2, Minor:1
[    0.886111] 17 fixed-partitions partitions found on MTD device qcom_nand.0
[    0.891223] Creating 17 MTD partitions on "qcom_nand.0":
[    0.898153] 0x000000000000-0x000000080000 : "0:SBL1"
[    0.905109] 0x000000080000-0x000000100000 : "0:MIBIB"
[    0.910138] 0x000000100000-0x000000140000 : "0:BOOTCONFIG"
[    0.914616] 0x000000140000-0x000000180000 : "0:BOOTCONFIG1"
[    0.920055] 0x000000180000-0x000000280000 : "0:QSEE"
[    0.926815] 0x000000280000-0x000000380000 : "0:QSEE_1"
[    0.931764] 0x000000380000-0x0000003c0000 : "0:DEVCFG"
[    0.935535] 0x0000003c0000-0x000000400000 : "0:DEVCFG_1"
[    0.940821] 0x000000400000-0x000000440000 : "0:CDT"
[    0.946324] 0x000000440000-0x000000480000 : "0:CDT_1"
[    0.950828] 0x000000480000-0x000000500000 : "0:APPSBLENV"
[    0.956449] 0x000000500000-0x000000640000 : "0:APPSBL"
[    0.962891] 0x000000640000-0x000000780000 : "0:APPSBL_1"
[    0.968129] 0x000000780000-0x000000880000 : "0:ART"
[    0.972979] 0x000000880000-0x000000900000 : "0:TRAINING"
[    0.976980] 0x000000900000-0x000004300000 : "rootfs"
[    1.064560] mtd: setting mtd15 (rootfs) as root device
[    1.065941] mtdsplit: error occured while reading from "rootfs"
[    1.068709] 0x000004300000-0x000007d00000 : "rootfs_1"
[    1.557315] i2c_dev: i2c /dev entries driver
[    1.558363] cpufreq: cpufreq_online: CPU0: Running at unlisted initial frequency: 24000 KHz, changing to: 800000 KHz
[    1.561003] sdhci: Secure Digital Host Controller Interface driver
[    1.571225] sdhci: Copyright(c) Pierre Ossman
[    1.577194] sdhci-pltfm: SDHCI platform and OF driver helper
[    1.584402] remoteproc remoteproc0: releasing cd00000.remoteproc
[    1.588620]  cs_system_cfg: CoreSight Configuration manager initialised
[    1.594946] NET: Registered PF_INET6 protocol family
[    1.601500] Segment Routing with IPv6
[    1.604897] In-situ OAM (IOAM) with IPv6
[    1.608569] NET: Registered PF_PACKET protocol family
[    1.612457] 8021q: 802.1Q VLAN Support v1.8
[    1.617587] Registering SWP/SWPB emulation handler
[    1.642534] remoteproc remoteproc0: cd00000.remoteproc is available
[    1.643945] remoteproc remoteproc1: 4ab000.remoteproc_pd1 is available
[    1.649375] remoteproc remoteproc2: cd00000.remoteproc:remoteproc_pd2 is available
[    1.655250] ubi0: attaching mtd15
[    1.663580] ubi0 warning: ubi_io_read: error -74 (ECC error) while reading 64 bytes from PEB 0:0, read only 64 bytes, retry
[    1.666420] ubi0 warning: ubi_io_read: error -74 (ECC error) while reading 64 bytes from PEB 0:0, read only 64 bytes, retry
[    1.677328] ubi0 warning: ubi_io_read: error -74 (ECC error) while reading 64 bytes from PEB 0:0, read only 64 bytes, retry
[    1.688621] ubi0 error: ubi_io_read: error -74 (ECC error) while reading 64 bytes from PEB 0:0, read 64 bytes
[    1.698361] CPU: 0 PID: 1 Comm: swapper/0 Not tainted 5.15.150 #0
[    1.708375] Hardware name: Generic DT based system
[    1.714460] [<8030e0b0>] (unwind_backtrace) from [<80309bd8>] (show_stack+0x10/0x14)
[    1.719143] [<80309bd8>] (show_stack) from [<8067a780>] (dump_stack_lvl+0x40/0x4c)
[    1.727042] [<8067a780>] (dump_stack_lvl) from [<80791d88>] (ubi_io_read+0x138/0x33c)
[    1.734421] [<80791d88>] (ubi_io_read) from [<807921e0>] (ubi_io_read_ec_hdr+0x44/0x1f4)
[    1.742318] [<807921e0>] (ubi_io_read_ec_hdr) from [<807972f4>] (ubi_attach+0x428/0x1458)
[    1.750479] [<807972f4>] (ubi_attach) from [<8078c4e4>] (ubi_attach_mtd_dev+0x494/0xb34)
[    1.758552] [<8078c4e4>] (ubi_attach_mtd_dev) from [<80d1e7f0>] (ubi_init+0x124/0x354)
[    1.766712] [<80d1e7f0>] (ubi_init) from [<8030180c>] (do_one_initcall+0x54/0x1f4)
[    1.774436] [<8030180c>] (do_one_initcall) from [<80d013d8>] (kernel_init_freeable+0x218/0x26c)
[    1.781991] [<80d013d8>] (kernel_init_freeable) from [<809f9a1c>] (kernel_init+0x14/0x124)
[    1.790584] [<809f9a1c>] (kernel_init) from [<80300124>] (ret_from_fork+0x14/0x30)
[    1.798914] Exception stack(0x81029fb0 to 0x81029ff8)
[... 2 lines cut ...]
[    1.819752] 9fe0: 00000000 00000000 00000000 00000000 00000013 00000000
[    1.830145] ubi0 warning: ubi_io_read: error -74 (ECC error) while reading 64 bytes from PEB 1:0, read only 64 bytes, retry
[    1.835616] ubi0 warning: ubi_io_read: error -74 (ECC error) while reading 64 bytes from PEB 1:0, read only 64 bytes, retry
[    1.846791] ubi0 warning: ubi_io_read: error -74 (ECC error) while reading 64 bytes from PEB 1:0, read only 64 bytes, retry
[    1.857866] ubi0 error: ubi_io_read: error -74 (ECC error) while reading 64 bytes from PEB 1:0, read 64 bytes
[    1.867733] CPU: 0 PID: 1 Comm: swapper/0 Not tainted 5.15.150 #0
[    1.877732] Hardware name: Generic DT based system
[    1.883815] [<8030e0b0>] (unwind_backtrace) from [<80309bd8>] (show_stack+0x10/0x14)
[... cut ...]
[    1.959941] [<809f9a1c>] (kernel_init) from [<80300124>] (ret_from_fork+0x14/0x30)
[    1.968271] Exception stack(0x81029fb0 to 0x81029ff8)
[    1.975825] 9fa0:                                     00000000 00000000 00000000 00000000
[    1.980951] 9fc0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[    1.989110] 9fe0: 00000000 00000000 00000000 00000000 00000013 00000000
[    1.999424] ubi0 warning: ubi_io_read: error -74 (ECC error) while reading 64 bytes from PEB 2:0, read only 64 bytes, retry

后面是不停地循环。。。

sseseed, 2024-07-28:
> 应该是0x11

麻烦帮看下添加 FM25LS01 这个 SPI NAND 的结构体是不是哪里有错？芯片简单说明：https://www.fmsh.com/62934bb9-d96d-5f04-e1ce-dd9a26dbfebf/ 
谢谢。
{"FM25LS01 SPI NAND 1G 1.8V",
{ .id = {0xa1, 0xa5} },
SZ_2K, SZ_128, SZ_128K, 0, 2, 128, NAND_ECC_INFO(1, SZ_512) },

potatoigit, 2024-07-28:
此类问题无法得到保证回答，请自行多对比学习并实践验证

potatoigit, 2024-07-28:
> > 应该是0x11
> 
> 麻烦帮看下添加 FM25LS01 这个 SPI NAND 的结构体是不是哪里有错？芯片简单说明：https://www.fmsh.com/62934bb9-d96d-5f04-e1ce-dd9a26dbfebf/ 谢谢。 {"FM25LS01 SPI NAND 1G 1.8V", { .id = {0xa1, 0xa5} }, SZ_2K, SZ_128, SZ_128K, 0, 2, 128, NAND_ECC_INFO(1, SZ_512) },

此类问题无法得到保证回答，请自行多对比学习并实践验证。

potatoigit, 2024-07-28:
请尝试NAND_ECC_INFO(4, SZ_512)

potatoigit, 2024-07-28:
> > 应该是0x11
> 
> 麻烦帮看下添加 FM25LS01 这个 SPI NAND 的结构体是不是哪里有错？芯片简单说明：https://www.fmsh.com/62934bb9-d96d-5f04-e1ce-dd9a26dbfebf/ 谢谢。 {"FM25LS01 SPI NAND 1G 1.8V", { .id = {0xa1, 0xa5} }, SZ_2K, SZ_128, SZ_128K, 0, 2, 128, NAND_ECC_INFO(1, SZ_512) },

NAND_ECC_INFO(1, SZ_512)

potatoigit, 2024-07-28:
> > 应该是0x11
> 
> 麻烦帮看下添加 FM25LS01 这个 SPI NAND 的结构体是不是哪里有错？芯片简单说明：https://www.fmsh.com/62934bb9-d96d-5f04-e1ce-dd9a26dbfebf/ 谢谢。 {"FM25LS01 SPI NAND 1G 1.8V", { .id = {0xa1, 0xa5} }, SZ_2K, SZ_128, SZ_128K, 0, 2, 128, NAND_ECC_INFO(1, SZ_512) },

请尝试NAND_ECC_INFO(1, SZ_512)

potatoigit, 2024-07-28:
> > 应该是0x11
> 
> 麻烦帮看下添加 FM25LS01 这个 SPI NAND 的结构体是不是哪里有错？芯片简单说明：https://www.fmsh.com/62934bb9-d96d-5f04-e1ce-dd9a26dbfebf/ 谢谢。 {"FM25LS01 SPI NAND 1G 1.8V", { .id = {0xa1, 0xa5} }, SZ_2K, SZ_128, SZ_128K, 0, 2, 128, NAND_ECC_INFO(1, SZ_512) },

请尝试NAND_ECC_INFO(4, SZ_512)

potatoigit, 2024-07-28:
> > 应该是0x11
> 
> 麻烦帮看下添加 FM25LS01 这个 SPI NAND 的结构体是不是哪里有错？芯片简单说明：https://www.fmsh.com/62934bb9-d96d-5f04-e1ce-dd9a26dbfebf/ 谢谢。 {"FM25LS01 SPI NAND 1G 1.8V", { .id = {0xa1, 0xa5} }, SZ_2K, SZ_128, SZ_128K, 0, 2, 128, NAND_ECC_INFO(1, SZ_512) },

您的patch里的oob size为64，和上面的128不一致，且也应为128

sseseed, 2024-07-28:
> > > 应该是0x11
> > 
> > 
> > 麻烦帮看下添加 FM25LS01 这个 SPI NAND 的结构体是不是哪里有错？芯片简单说明：https://www.fmsh.com/62934bb9-d96d-5f04-e1ce-dd9a26dbfebf/ 谢谢。 {"FM25LS01 SPI NAND 1G 1.8V", { .id = {0xa1, 0xa5} }, SZ_2K, SZ_128, SZ_128K, 0, 2, 128, NAND_ECC_INFO(1, SZ_512) },
> 
> 请尝试NAND_ECC_INFO(4, SZ_512)

这个试过了，也是不行。可能不只一个参数有问题。

sseseed, 2024-07-29:
> > > 应该是0x11
> > 
> > 
> > 麻烦帮看下添加 FM25LS01 这个 SPI NAND 的结构体是不是哪里有错？芯片简单说明：https://www.fmsh.com/62934bb9-d96d-5f04-e1ce-dd9a26dbfebf/ 谢谢。 {"FM25LS01 SPI NAND 1G 1.8V", { .id = {0xa1, 0xa5} }, SZ_2K, SZ_128, SZ_128K, 0, 2, 128, NAND_ECC_INFO(1, SZ_512) },
> 
> 请尝试NAND_ECC_INFO(4, SZ_512)

找到了另一个形式的对 FM25LS01 的结构体，但不会套过去...
{
		.id = { 0xa1, 0xa5 },
		.page_size = 2048,
		.erase_blk_size = 0x00020000,
		.pgs_per_blk = 64,
		.no_of_blocks = 1024,
		.spare_size = 64,
		.density = 0x08000000,
		.otp_region = 0x2000,
		.no_of_addr_cycle = 0x3,
		.num_bits_ecc_correctability = 4,
		.timing_mode_support = 0,
		.quad_mode = true,
		.check_quad_config = true,
		.name = "FM25LS01",
	},

JiaY-shi, 2024-08-12:
```
From 6ee246291159ce8fa235aa4412941413e073c779 Mon Sep 17 00:00:00 2001
From: JiaY-shi <shi05275@163.com>
Date: Sun, 4 Aug 2024 13:19:32 +0800
Subject: mtd: nand: add ipq5018 spi nand support

---
 drivers/mtd/nand/raw/qcom_nandc.c | 191 ++++++++++++++++++++++++++++--
 1 file changed, 178 insertions(+), 13 deletions(-)

diff --git a/drivers/mtd/nand/raw/qcom_nandc.c b/drivers/mtd/nand/raw/qcom_nandc.c
index b8cff9240b28..9d0e690980b3 100644
--- a/drivers/mtd/nand/raw/qcom_nandc.c
+++ b/drivers/mtd/nand/raw/qcom_nandc.c
@@ -35,13 +35,22 @@
 #define	NAND_DEV_CMD0			0xa0
 #define	NAND_DEV_CMD1			0xa4
 #define	NAND_DEV_CMD2			0xa8
+#define	NAND_DEV_CMD3			0xd0
+#define	NAND_DEV_CMD4			0xd4
+#define	NAND_DEV_CMD5			0xd8
+#define	NAND_DEV_CMD6			0xdc
+#define	NAND_DEV_CMD7			0xb0
+#define	NAND_DEV_CMD8			0xb4
+#define	NAND_DEV_CMD9			0xb8
 #define	NAND_DEV_CMD_VLD		0xac
+#define	NAND_FLASH_SPI_CFG		0xc0
+#define	NAND_SPI_NUM_ADDR_CYCLES	0xc4
+#define	NAND_SPI_BUSY_CHECK_WAIT_CNT	0xc8
 #define	SFLASHC_BURST_CFG		0xe0
 #define	NAND_ERASED_CW_DETECT_CFG	0xe8
 #define	NAND_ERASED_CW_DETECT_STATUS	0xec
 #define	NAND_EBI2_ECC_BUF_CFG		0xf0
 #define	FLASH_BUF_ACC			0x100
-
 #define	NAND_CTRL			0xf00
 #define	NAND_VERSION			0xf08
 #define	NAND_READ_LOCATION_0		0xf20
@@ -52,6 +61,7 @@
 #define	NAND_READ_LOCATION_LAST_CW_1	0xf44
 #define	NAND_READ_LOCATION_LAST_CW_2	0xf48
 #define	NAND_READ_LOCATION_LAST_CW_3	0xf4c
+#define	NAND_QSPI_MSTR_CONFIG		0xf60

 /* dummy register offsets, used by write_reg_dma */
 #define	NAND_DEV_CMD1_RESTORE		0xdead
@@ -189,6 +199,28 @@
 #define	ECC_BCH_4BIT	BIT(2)
 #define	ECC_BCH_8BIT	BIT(3)

+/* QSPI NAND config reg bits */
+#define	LOAD_CLK_CNTR_INIT_EN	(1 << 28)
+#define	CLK_CNTR_INIT_VAL_VEC	0x924
+#define	FEA_STATUS_DEV_ADDR	0xc0
+#define	SPI_CFG	(1 << 0)
+
+/* CMD register value for qspi nand */
+#define	CMD0_VAL	0x1080D8D8
+#define	CMD1_VAL	0xF00F3000
+#define	CMD2_VAL	0xF0FF709F
+#define	CMD3_VAL	0x3F310015
+#define	CMD7_VAL	0x04061F0F
+#define	CMD_VLD_VAL	0xd
+#define	SPI_NUM_ADDR	0xDA4DB
+#define	WAIT_CNT	0x10
+
+/* QSPI NAND CMD reg bits value */
+#define	SPI_WP		(1 << 28)
+#define	SPI_HOLD	(1 << 27)
+#define	SPI_TRANSFER_MODE_x1	(1 << 29)
[... 330 lines cut ...]
 };

 static const struct qcom_nandc_props sdx55_nandc_props = {
@@ -3526,6 +3687,10 @@ static const struct of_device_id qcom_nandc_of_match[] = {
 		.compatible = "qcom,sdx55-nand",
 		.data = &sdx55_nandc_props,
 	},
+	{
+		.compatible = "qcom,ipq5018-nand",
+		.data = &ipq5018_nandc_props,
+	},
 	{}
 };
 MODULE_DEVICE_TABLE(of, qcom_nandc_of_match);
@@ -3542,4 +3707,4 @@ module_platform_driver(qcom_nandc_driver);

 MODULE_AUTHOR("Archit Taneja <architt@codeaurora.org>");
 MODULE_DESCRIPTION("Qualcomm NAND Controller driver");
-MODULE_LICENSE("GPL v2");
+MODULE_LICENSE("GPL v2");
\ No newline at end of file
-- 
2.43.0
```
试试这个，测试过，基于linux-6.6内核