hzyitc/openwrt-redmi-ax3000 #81: WiFi speeds

eeddww, 2024-07-28:
First of all, great work. I remember people saying that openwrt on the ipq5000 would be impossible. now, I have a ra80(sorry, I think this firmware isn't really made for this one) and a ra81. I have installed the firmware from [here](https://github.com/hzyitc/openwrt-redmi-ax3000/releases/tag/ci-ipq50xx-qsdk-kernel-5.4-openwrt-21.02-qsdk-11.5.05.841.1029-20240723-140038-UTC) in both of them successfully (it's the build with nssoffloading. The ra80 works as my main router (using pppoe), while the ra81 is an access point. The ethernet speeds are great (900mbps-1gbps), equal or even better than stock firmware. My problem is with the wifi speed., it is extremely low (not even surpassing 30mbps) Maybe I'm doing something wrong in my config, or it is normal to get this speeds in these builds? 
My main router (ra80)
[image]
My AP
[image]

I enabled this setting in the main router, thought it would help ¿?
[image]

Thank you in advance.

mitchell931993, 2024-08-19:
Tangent: did you need to do anything special to get it working on the ra80. Are you running stable ? Or a different branch ?

eeddww, 2024-08-21:
> Tangent: did you need to do anything special to get it working on the ra80. Are you running stable ? Or a different branch ?

I just used the 5.4 stable branch. I read how to do it here https://github.com/hzyitc/openwrt-redmi-ax3000/issues/2#issuecomment-2117261253 . Neither of my devices works great when it comes to wifi tho, don't know what I'm doing wrong

kimaguraDroid, 2024-09-04:
Hi @eeddww ,
I have a Xiaomi AX3000 ra82, and I want to install OpenWrt.
Can you confirm the steps to update on this router? On 2 other devices like Xiaomi AX3600 (I have 2 with OpenWrt) the first requirements is to enable SSH, but for this AX3000 I tried but I did not succeed.

eeddww, 2024-09-06:
> Hi @eeddww , I have a Xiaomi AX3000 ra82, and I want to install OpenWrt. Can you confirm the steps to update on this router? On 2 other devices like Xiaomi AX3600 (I have 2 with OpenWrt) the first requirements is to enable SSH, but for this AX3000 I tried but I did not succeed.
> 
> finally, I get ssh to my AX3000 RA82 with next steps:
> 
> 1. reset
> 2. connect with cable with valid network address like 192.168.31.5 and open browser to 192.168.31.1
> 3. follow initial steps to configure like a new Xiaomi Router.
> 4. After save changes in steps 3, activate DHCP in Windows to get new IP. Should be like 192.171.31.*
> 5. From xmir-patcher from [openwrt-xiaomi/xmir-patcher](https://github.com/openwrt-xiaomi/xmir-patcher) run.bat (windows)
> 6. press 1 and set ip to 192.171.31.1
> 7. press 2 to install SSH
> 8. press 8 -> 7 to install permanent SSH access

Hi, I planned to respond to you but you beat me to it lol. I also used xmir patcher to get ssh access. I got bored of waiting for the usb thingy that I ordered from aliexpress. Have you flashed openwrt already? are you speeds slow too?

Thomasefski, 2024-10-15:
Well I did succeed to gain ssh access on the ra82. Tried to flash openwrt but now the device does nothing. Anyone have a fix on getting it operational again?

eeddww, 2024-10-19:
I hope that the dude who commented about how he got his router bricked (and deleted the comment) managed to fix it lol.

Thomasefski, 2024-10-19:
I did! Still didn't manage to flash openwrt on it tho

averyanalex, 2024-10-29:
I flashed [2024-07-27](https://github.com/hzyitc/openwrt-redmi-ax3000/releases/tag/ci-ipq50xx-qsdk-kernel-5.4-openwrt-21.02-qsdk-11.5.05.841.1029-20240727-173350-UTC) on chinese AX3000 (RA80), unaccaptable slow wifi, especially 5G (only works ~50cm near router and very slow), 2.4G ~30 mbps in another room. Reverted to stock firmware.

htbuddha, 2024-10-29:
I'm so glad to see people are finding ways to flash OpenWRT on AX3000 RA80.

I'm new to this and have never attempted it. Hopefully in a little while updates will come out to make wifi signal better, and maybe a guide for people who have never flashed openwrt on a Xiaomi router device.

balangandio, 2025-04-11:
> I flashed [2024-07-27](https://github.com/hzyitc/openwrt-redmi-ax3000/releases/tag/ci-ipq50xx-qsdk-kernel-5.4-openwrt-21.02-qsdk-11.5.05.841.1029-20240727-173350-UTC) on chinese AX3000 (RA80), [...]

I did the same. I tried to play with the country code and txpower but had no luck, the 5Ghz radio reception remains very poor. 

Could you explain how you reverted to the stock firmware ? I had no success uploading the miwifi_ra80 bin file via the Luci web interface.

averyanalex, 2025-04-21:
> I did the same. I tried to play with the country code and txpower but had no luck, the 5Ghz radio reception remains very poor.
>
> Could you explain how you reverted to the stock firmware ? I had no [...]

I don't remember exact process, but I just set prev a/b partition as default. No flashing, just 1 command and reboot. But it will work only if you flashed custom firmware once and never updated ut.