hzyitc/openwrt-redmi-ax3000 #83: 两个 QSDK 分支云编译都报同样的错误

sseseed, 2024-08-05:
开始编译没几秒就报错了，两个 QSDK 版本都是一样的错误，编译的是自己适配加上去的 IPQ5018 机子，本地编译 make menuconfig 保存的 config 文件，本地编译也是这样的错，以为是自己适配得有问题，换其它型号机子也是一样，又以为是本地环境不对，转去云编译也是一样的错误。

WARNING: Makefile 'package/kernel/qca/qca-nss-clients/Makefile' has a dependency on 'kmod-qca-nss-drv-ipsecmgr', which does not exist
WARNING: Makefile 'package/kernel/qca/qca-nss-clients/Makefile' has a dependency on 'kmod-qca-nss-drv-dtlsmgr', which does not exist
WARNING: Makefile 'package/kernel/qca/qca-nss-clients/Makefile' has a dependency on 'kmod-qca-nss-drv-capwapmgr', which does not exist
WARNING: Makefile 'package/kernel/qca/qca-nss-clients/Makefile' has a dependency on 'kmod-qca-ovsmgr', which does not exist
WARNING: Makefile 'package/kernel/qca/qca-nss-clients/Makefile' has a dependency on 'kmod-qca-nss-cfi-cryptoapi', which does not exist
WARNING: Makefile 'package/kernel/qca/qca-nss-clients/Makefile' has a build dependency on 'kmod-qca-nss-crypto', which does not exist
WARNING: Makefile 'package/kernel/qca/qca-nss-drv/Makefile' has a dependency on 'kmod-qca-nss-gmac', which does not exist
WARNING: Makefile 'package/kernel/qca/qca-nss-ecm/Makefile' has a dependency on 'kmod-qca-ovsmgr', which does not exist
WARNING: Makefile 'package/kernel/qca/qca-nss-ecm/Makefile' has a dependency on 'kmod-shortcut-fe-drv', which does not exist
WARNING: Makefile 'package/kernel/qca/qca-nss-ecm/Makefile' has a dependency on 'kmod-qca-mcs', which does not exist
WARNING: Makefile 'package/kernel/qca/qca-nss-ecm/Makefile' has a dependency on 'kmod-qca-ovsmgr', which does not exist
WARNING: Makefile 'package/kernel/qca/qca-nss-ecm/Makefile' has a dependency on 'kmod-qca-mcs', which does not exist
WARNING: Makefile 'package/kernel/qca/qca-nss-ecm/Makefile' has a dependency on 'kmod-qca-ovsmgr', which does not exist
WARNING: Makefile 'package/kernel/qca/qca-nss-ecm/Makefile' has a dependency on 'kmod-shortcut-fe-drv', which does not exist
WARNING: Makefile 'package/kernel/qca/qca-nss-ecm/Makefile' has a dependency on 'kmod-qca-mcs', which does not exist
WARNING: Makefile 'package/kernel/qca/qca-nss-ecm/Makefile' has a dependency on 'kmod-qca-hyfi-bridge', which does not exist
WARNING: Makefile 'package/kernel/qca/qca-nss-ecm/Makefile' has a dependency on 'kmod-emesh-sp', which does not exist
WARNING: Makefile 'package/kernel/qca/qca-nss-ecm/Makefile' has a dependency on 'kmod-qca-ovsmgr', which does not exist
WARNING: Makefile 'package/kernel/qca/qca-nss-ecm/Makefile' has a dependency on 'kmod-shortcut-fe-drv', which does not exist
WARNING: Makefile 'package/kernel/qca/qca-nss-ecm/Makefile' has a dependency on 'kmod-qca-mcs', which does not exist
WARNING: Makefile 'package/kernel/qca/qca-nss-ecm/Makefile' has a dependency on 'kmod-qca-hyfi-bridge', which does not exist
WARNING: Makefile 'package/kernel/qca/qca-nss-ecm/Makefile' has a dependency on 'kmod-emesh-sp', which does not exist
WARNING: Makefile 'package/kernel/qca/qca-nss-ecm/Makefile' has a dependency on 'kmod-shortcut-fe-drv', which does not exist
WARNING: Makefile 'package/kernel/qca/qca-nss-ecm/Makefile' has a dependency on 'kmod-qca-ovsmgr', which does not exist
WARNING: Makefile 'package/kernel/qca/qca-nss-ecm/Makefile' has a dependency on 'kmod-qca-mcs', which does not exist
WARNING: Makefile 'package/kernel/qca/qca-nss-ecm/Makefile' has a dependency on 'kmod-qca-hyfi-bridge', which does not exist
WARNING: Makefile 'package/kernel/qca/qca-nss-ecm/Makefile' has a dependency on 'kmod-emesh-sp', which does not exist
WARNING: Makefile 'package/kernel/qca/qca-ssdk/Makefile' has a dependency on 'kmod-qca-rfs', which does not exist
generic.mk:6: common-ubnt.mk: No such file or directory
make[6]: *** No rule to make target 'common-ubnt.mk'.  Stop.
make[5]: *** [Makefile:20: image-prereq] Error 2
make[4]: *** [Makefile:11: prereq] Error 2
time: target/linux/prereq#0.51#0.05#0.56
    ERROR: target/linux failed to build.
make[3]: *** [target/Makefile:25: target/linux/prereq] Error 1
make[2]: *** [target/Makefile:17: /home/runner/work/OpenWrt/OpenWrt/staging_dir/target-mips_24kc_musl/stamp/.target_prereq] Error 2
make[1]: *** [/home/runner/work/OpenWrt/OpenWrt/include/toplevel.mk:20[8](https://github.com/sseseed/OpenWrt/actions/runs/10245433203/job/28340787881#step:8:9): prereq] Error 2
make: *** [/home/runner/work/OpenWrt/OpenWrt/include/toplevel.mk:228: world] Error 2

WhaleFell, 2024-12-02:
Same problem.

hzyitc, 2025-02-08:
参考我的action