hzyitc/openwrt-redmi-ax3000 #84: 关于无线nss的一些建议

breeze303, 2024-08-19:
目前对于ipq60xx的开源无线nss已经非常成熟，能达到千兆切0cpu占用https://github.com/LiBwrt-op/openwrt-6.x
想在ipq50xx上使用https://github.com/qosmio/nss-packages，但需要将内核升级到6.6才能适应该nss，5.x内核适配过程中遇到问题，不知到大佬有没有兴趣整合下这个nss

breeze303, 2024-08-22:
> 在适配6.6了，但需要时间

我也在捣鼓这平台，有机会一起玩下
https://github.com/LiBwrt-op/openwrt-6.x.git

breeze303, 2024-09-10:
[image]
64位目前带上nss及无线编译是通过了没刷入测试没机器，32位卡在这个bug一直没法解决不知道哪里导致无法获取

yyangv, 2025-03-22:
发现有个ipq50xx的分支，是否有帮助。

https://github.com/LiBwrt/openwrt-6.x/tree/ipq50xx

作为一个程序员，最新有这一块需求，也想研究研究。

breeze303, 2025-03-22:
> [@hzyitc](https://github.com/hzyitc) 发现有个ipq50xx的分支，是否有帮助。
> 
> https://github.com/LiBwrt/openwrt-6.x/tree/ipq50xx
> 
> 作为一个程序员，最新有这一块需求，也想研究研究。

这个我都不维护了，你用https://github.com/VIKINGYFY/immortalwrt，https://github.com/VIKINGYFY/immortalwrt/commit/ca09e54dda20894d993e6c6061fc655224369dbf我在适配后羿哪吒的时候发现nss不需要过多的修复就是正常的，但wifi部分的dts暂未修复

yyangv, 2025-03-22:
> > [@hzyitc](https://github.com/hzyitc) 发现有个ipq50xx的分支，是否有帮助。
> > https://github.com/LiBwrt/openwrt-6.x/tree/ipq50xx
> > 作为一个程序员，最新有这一块需求，也想研究研究。
> 
> 这个我都不维护了 你用https://github.com/VIKINGYFY/immortalwrt
> 
> [VIKINGYFY/immortalwrt@ca09e54](https://github.com/VIKINGYFY/immortalwrt/commit/ca09e54dda20894d993e6c6061fc655224369dbf)
> 
> 我在适配后羿哪吒的时候发现nss不需要过多的修复就是正常的，但wifi部分的dts暂未修复

谢谢！https://github.com/VIKINGYFY/immortalwrt 这个构建目标Target都没有 IPQ50xx，发现 Openwrt 官网也没有 AX3000 的固件，有很多工作要做啊！

我还是先用他的吧，这一行我还是个小白。

borintkkk, 2025-03-30:
> 谢谢！https://github.com/VIKINGYFY/immortalwrt 这个构建目标Target都没有 IPQ50xx，发现 Openwrt 官网也没有 AX3000 的固件，有很多工作要做啊！
>
> 我还是先用他的吧，这一行我还是个小白。

[[immortalwrt](https://github.com/VIKINGYFY/immortalwrt/tree/main)/[target](https://github.com/VIKINGYFY/immortalwrt/tree/main/target)/[linux](https://github.com/VIKINGYFY/immortalwrt/tree/main/target/linux)
/qualcommax](https://github.com/VIKINGYFY/immortalwrt/tree/main/target/linux/qualcommax/ipq50xx)   有ipq50xx ，有人会折腾吗