hzyitc/openwrt-redmi-ax3000 #88: XIaomi Mesh System AX3000-RA82

BASHEER-A, 2024-09-10:
I wanna ask you is this software Support show me my system AX 3000 model RA82 
,
By the way  , 

Serial connection parameters , 1.8v not 3.3v

BASHEER-A, 2024-09-10:
[image] this model

Galtor, 2024-09-30:
It's not working. I've tried all three firmwares and no luck, all three attempts resulted in a brick. It can be recovered using the standard Xiaomi TFTP method. 
I can provide more device information if the developer wants to add support. Feel free to ask any other questions.

```
root@XiaoQiang:~# cat /etc/openwrt_release 
DISTRIB_ID='OpenWrt'
DISTRIB_RELEASE='18.06-SNAPSHOT'
DISTRIB_REVISION='unknown'
DISTRIB_TARGET='ipq/ipq50xx'
DISTRIB_ARCH='arm_cortex-a7'
DISTRIB_DESCRIPTION='OpenWrt 18.06-SNAPSHOT unknown'
DISTRIB_TAINTS='no-all busybox'

root@XiaoQiang:~# cat /etc/openwrt_version 
unknown

root@XiaoQiang:~# bootinfo

ROM    ver: config core 'version'
	# ROM ver
	option ROM '1.4.31'
	# channel
	option CHANNEL 'release'
	# hardware platform R1AC or R1N etc.
	option HARDWARE 'RA82'
	# CFE ver
	option UBOOT '1.0.2'
	# Linux Kernel ver
	option LINUX '4.4.16'
	# RAMFS ver
	option RAMFS '0.0.1'
	# SQUASHFS ver
	option SQAFS '0.0.1'
	# ROOTFS ver
	option ROOTFS '0.0.1'
  #build time
  option BUILDTIME 'Thu, 07 Sep 2023 04:09:20 +0000'
  #build timestamp
  option BUILDTS '1694059760'
  #build git tag
   option GTAG 'commit 2021e96342a56819306987c575af9f27c4f37214'
Hardware  : Ver. A
ROM    sum: 
System    : Dual - 2
KERNEL    : ubi.mtd=rootfs_1 root=mtd:ubi_rootfs rootfstype=squashfs cnss2.bdf_integrated=0x24 cnss2.bdf_pci0=0x60 cnss2.bdf_pci1=0x60 cnss2.skip_radio_bmap=4 rootwait uart_en=1 swiotlb=1

MTD  table:
dev:    size   erasesize  name
mtd0: 00080000 00020000 "0:SBL1"
mtd1: 00080000 00020000 "0:MIBIB"
mtd2: 00040000 00020000 "0:BOOTCONFIG"
mtd3: 00040000 00020000 "0:BOOTCONFIG1"
mtd4: 00100000 00020000 "0:QSEE"
mtd5: 00100000 00020000 "0:QSEE_1"
mtd6: 00040000 00020000 "0:DEVCFG"
mtd7: 00040000 00020000 "0:DEVCFG_1"
mtd8: 00040000 00020000 "0:CDT"
mtd9: 00040000 00020000 "0:CDT_1"
mtd10: 00080000 00020000 "0:APPSBLENV"
mtd11: 00140000 00020000 "0:APPSBL_1"
mtd12: 00140000 00020000 "0:APPSBL"
mtd13: 00100000 00020000 "0:ART"
mtd14: 00080000 00020000 "0:TRAINING"
mtd15: 00080000 00020000 "bdata"
mtd16: 00080000 00020000 "crash"
mtd17: 00080000 00020000 "crash_syslog"
mtd18: 02400000 00020000 "rootfs"
mtd19: 02400000 00020000 "rootfs_1"
mtd20: 01f00000 00020000 "overlay"
mtd21: 00080000 00020000 "rsvd0"
[... 72 lines cut ...]
mtd3_0BOOTCONFIG1.bin: data
mtd4_0QSEE.bin:        ELF 64-bit LSB executable, ARM aarch64, version 1 (SYSV), statically linked, no section header
mtd5_0QSEE1.bin:       ELF 64-bit LSB executable, ARM aarch64, version 1 (SYSV), statically linked, no section header
mtd6_0DEVCFG.bin:      ELF 64-bit LSB executable, ARM aarch64, version 1 (SYSV), statically linked, no section header
mtd7_0DEVCFG1.bin:     ELF 64-bit LSB executable, ARM aarch64, version 1 (SYSV), statically linked, no section header
mtd8_0CDT.bin:         data
mtd9_0CDT1.bin:        data
mtd10_0APPSBLENV.bin:  data
mtd11_0APPSBL1.bin:    ELF 32-bit LSB executable, ARM, EABI5 version 1 (SYSV), statically linked, no section header
mtd12_0APPSBL.bin:     ELF 32-bit LSB executable, ARM, EABI5 version 1 (SYSV), statically linked, no section header
mtd13_0ART.bin:        data
mtd14_0TRAINING.bin:   data
mtd15_bdata.bin:       data
mtd16_crash.bin:       ISO-8859 text, with very long lines (65536), with no line terminators
mtd17_crashsyslog.bin: ISO-8859 text, with very long lines (65536), with no line terminators
mtd18_rootfs.bin:      UBI image, version 1
mtd19_rootfs1.bin:     UBI image, version 1
mtd20_overlay.bin:     UBI image, version 1
mtd21_rsvd0.bin:       ISO-8859 text, with very long lines (65536), with no line terminators
mtd22_kernel.bin:      data
mtd23_ubirootfs.bin:   Squashfs filesystem, little endian, version 4.0, xz compressed, 21154186 bytes, 4398 inodes, blocksize: 262144 bytes, created: Thu Sep  7 04:23:44 2023

```

Partition rootfs appears to be kernel + ubirootfs, but kernel in xiaomi firmware has 40 extra bytes in the header (some secure firm?)

giannidallatorre, 2024-12-17:
@BASHEER-A 

Apparently, a user on the OpenWRT forum managed to install OpenWRT on an RA80 device. From what I understand, the hardware is identical to the RA82.

Here is the link to the post where they explained what they did:
https://forum.openwrt.org/t/adding-openwrt-support-for-xiaomi-redmi-router-ax6s-xiaomi-router-ax3200/111085/1875

I might get my hands dirty and give it a try myself in the next few days, but I thought it was worth sharing with you. You might find something useful here, especially since you’re already a few steps ahead.

Thomasefski, 2025-01-20:
Has anyone managed to do this as of yet? I tried, multiple methods, no success so far :(

hzyitc, 2025-02-08:
please provide the stock rom. and make sure you known how to recovery when crash.

invictus737, 2025-03-15:
Here is recovery rom for RA82
https://cdn.awsusor1.fds.api.mi-img.com/xiaoqiang/rom/ra82/miwifi_ra82_all_7eb36_1.3.27_INT.bin

Regards

moda20, 2025-06-25:
> [@BASHEER-A](https://github.com/BASHEER-A)
> 
> Apparently, a user on the OpenWRT forum managed to install OpenWRT on an RA80 device. From what I understand, the hardware is identical to the RA82.
> 
> Here is the link to the post where they explained what they did: [forum.openwrt.org/t/adding-openwrt-support-for-xiaomi-redmi-router-ax6s-xiaomi-router-ax3200/111085/1875](https://forum.openwrt.org/t/adding-openwrt-support-for-xiaomi-redmi-router-ax6s-xiaomi-router-ax3200/111085/1875)
> 
> I might get my hands dirty and give it a try myself in the next few days, but I thought it was worth sharing with you. You might find something useful here, especially since you’re already a few steps ahead.

I used this post as a tutorial, everything worked fine, except that i got my RA82 model to reboot into a bootloop, so be careful of it

moda20, 2025-06-26:
@BASHEER-A no my router went into a bootloop that i can't get out of, no ssh or network access is available, i am trying to find a way to reset it to a stable state, but after flashing it is not usable

giannidallatorre, 2025-12-24:
Hey @moda20,

I saw you mentioned there’s no SSH or network access on your router, which makes it sound dead, but TFTP recovery might still work. The key thing is that the router’s bootloader runs before the OS, so it can listen for a firmware image over the network even if OpenWrt is stuck or bootlooping.

You can use the firmware @invictus737 shared:

https://cdn.awsusor1.fds.api.mi-img.com/xiaoqiang/rom/ra82/miwifi_ra82_all_7eb36_1.3.27_INT.bin

1. Connect your PC to the router via Ethernet and set a static IP (e.g., 192.168.31.100).
2. Run a TFTP server with the ROM in the root folder.
3. Hold the reset button, power on the router, and keep holding for ~5–10s until recovery mode starts (LED blinks orange/white).
4. Send the ROM via TFTP:

Linux

```sh
tftp -m binary 192.168.31.1 -c put miwifi_ra82_all_7eb36_1.3.27_INT.bin
```

Windows:

```cmd
tftp -i 192.168.31.1 PUT miwifi_ra82_all_7eb36_1.3.27_INT.bin
```

5. Wait for the router to finish flashing and reboot (LED should go solid green).

Have you tried this yet?

moda20, 2025-12-24:
@giannidallatorre your first comment steps don't make sense, why would i start a TFTP server on my PC if i need to put the firmware onto the router ? the router should have a TFTP server be running automatically for any of this to be viable in the first place. 

I also can't connect my router directly to my pc as i am using a mac; 
Here what i did : 
- holding the reset button did do something as now my main router recognizes the connection but no IP is handed out, setting the static ip doesn't change that, but if you are correct that is expected.
- I used the BSD tftp client that ships with macOs and tried to put the file but it was timing out all the time (i renamed the file, hopefully that is not an issue) : 

```
tftp> put miwifi_ra82_all_7eb36_1_3_27_INT.bin
putting miwifi_ra82_all_7eb36_1_3_27_INT.bin to 192.168.<>:miwifi_ra82_all_7eb36_1_3_27_INT.bin [octet]
receive_packet: timeout
Try 1, didn't receive answer from remote.
receive_packet: timeout
Try 2, didn't receive answer from remote.
receive_packet: timeout
Try 3, didn't receive answer from remote.
receive_packet: timeout
Try 4, didn't receive answer from remote.
receive_packet: timeout
Try 5, didn't receive answer from remote.
send_wrq: No route to host
Cannot send WRQ packet
```

moda20, 2026-11-06:
@giannidallatorre I finally got to do this, i had to use an AI agent on my laptop with WIFI and an eth port running ubuntu. 

It turns out that the bootloader TFTP service is not the only thing needing to be used. The U-Boot server running on the recovery mode on the AX3000 requires a DHCP server with an address : 192.168.31.100/24. the DHCP server will give the AP an IP (which you can put statically to any IP in the accepted range). 

The Firmware is also a thing to check, it seems there is a limitation on which firmware to flash, If you device previously had a firmware 1.13 for example it can't downgrade to 1.28 even with this recovery method seems like a ROM update happens on regular updates. SO if you don't care about version, getting the latest version is your best bet. 
The firmware name is also a fixed value: `hex(IP).img`. example : `192.168.31.2` → **`C0A81F02.img`** (`192.168.31.2` = `C0.A8.1F.02`). the IP (here 192.168.31.2) is the static IP you are going to give to the AP via the DHCP server.

After everything is ready then you can retrigger the recovery by : Plug the Ethernet cable into one of the LAN ports (not WAN) => unplugging power  => hold the recessed reset button => pug in power + keep holding the reset button until the system LED flashes orange. Then the flow should start on it's own. flashing will start instantly and will finish after some seconds of the LED flashing white/blue 

Here are the thing you can pass to your Agent to handle this for you or you can do it manually it's relatively easy : 

### Working procedure (PC on Linux)

**1. Direct Ethernet: PC NIC ↔ a LAN port of the router.** No switch in between.

**2. Static PC IP matching the bootloader's `serverip`:**

```bash
# 192.168.31.100 is the IP the U-Boot recovery expects ("serverip=192.168.31.100")
nmcli con mod "Wired connection 1" ipv4.method manual ipv4.addresses 192.168.31.100/24
nmcli con up "Wired connection 1" device enp0s31f6     # your NIC name
# NetworkManager sometimes drops the IP while the cable is unlinked —
# 'ip addr add 192.168.31.100/24 dev <iface>' right before booting the router fixes it
```

**3. Firmware + expected filename** (hex of the DHCP-leased address + `.img`):

```bash
sudo mkdir -p /srv/tftp && cd /srv/tftp
# take the INT firmware (this is what fixed it; 1.3.27 all-image is older & led to a no-boot state)
sudo wget -O C0A81F02.img \
  https://cdn.awsusor1.fds.api.mi-img.com/xiaoqiang/rom/ra82/miwifi_ra82_firmware_db06e_1.4.31_INT.bin
```

**4. One DHCP + TFTP server** (dnsmasq serves both — this is the piece every bare-TFTP attempt misses). Pin the lease to `192.168.31.2`: that's why the filename is `C0A81F02.img`:

```bash
sudo dnsmasq --no-daemon --port=0 \
  --interface=enp0s31f6 --bind-interfaces \
  --dhcp-range=192.168.31.2,192.168.31.2,255.255.255.0 \
  --enable-tftp --tftp-root=/srv/tftp \
  --dhcp-boot=C0A81F02.img --bootp-dynamic --log-dhcp
```

(leave it running in a terminal; optionally verify by downloading the file: `curl tftp://192.168.31.100/C0A81F02.img -o /tmp/check.img` and compare the SHA256SUM.)

**5. Trigger recovery:** unplug router power → **hold Reset** → plug power back in while holding → keep holding until the LED changes (fast orange blink) → release. The bootloader then: DHCP-discover → gets `.2` → requests `C0A81F02.img` via TFTP → downloads (26 MB took ~10 s at 100 Mbit, the recovery link is 100 Mbit anyway) → shows **flashing blue**.

**6. Wait until blue flashing stops, then power-cycle manually.** First boot can show solid red up to ~80 s (normal), then white → Wi-Fi comes up, `192.168.31.1` pings, stock is back.

**Wire-proof this actually happens** (think more debugging from dhcp log):

```bash
sudo tcpdump -i enp0s31f6 -Unn    # free to see: BOOTP/DHCP from router MAC + RRQ C0A81F02.img
```

Note: **TFTP recovery only accepts signed (stock) images** — OpenWrt images cannot be flashed this way (secure boot + RSA check). It restores stock firmware only.

giannidallatorre, 2026-11-06:
nice, thanks for coming back with this, it's genuinely useful.

I had an AI agent helping me as well, and it suggests that what you found is actually the generic Xiaomi U-Boot behaviour, not RA82-specific. The same thing is documented for the AX6000 (RA72) and the AX3000T: U-Boot does DHCP first, gets its own IP from the lease, and only then asks for `hex(ip).img`. No lease, no RRQ. That's the whole mystery, and it also explains why my earlier TFTP-only suggestion went nowhere.

Two things that might save the next person some pain: pin the lease by MAC (`--dhcp-host=<mac>,192.168.31.2`) instead of a one-address range, and if NetworkManager keeps dropping the IP, `nmcli dev set <iface> managed no` is less racy than re-adding it right before boot.

The caveat at the end is the important one though: signed images only, so this is a safety net, not an install path. OpenWrt on the RA82 still has to go through stock SSH + the inactive rootfs slot.