hzyitc/openwrt-redmi-ax3000 #89: CR8808(CR8818?) ipq5018版本无法启动

baili233, 2024-09-13:
尝试过5.15 5.4 4.4等版本均无法启动，报错如下图，貌似是找不到闪存
[image]
而原版小米系统启动如下图
[image]

baili233, 2024-09-14:
> 刷的是哪个ubi，ax3000还是881x，要对版本刷

刷的是881x，应该是没有添加我这个型号的NAND闪存，我自己添加（虽然不知道对不对）编译后成功启动了，
+   {"GD5F1GM7REYIG SPI NAND 1G 1.8V",
+       { .id = {0xc8, 0x81} },
+       SZ_2K, SZ_128, SZ_128K, 0, 2, 128, NAND_ECC_INFO(8, SZ_512) },
[image]

closed 2024-10-25 by baili233

zeroneiko, 2025-01-23:
手上有一台2024.02生产的8818，遇到了和大佬一样的情况，请问大佬是通过修改哪一个patch文件实现适配的

baili233, 2025-01-23:
> 手上有一台2024.02生产的8818，遇到了和大佬一样的情况，请问大佬是通过修改哪一个patch文件实现适配的

可以加个patch放到target/linux/ipq50xx/patches下

[422-add-GD5Fx-nand-support.patch](https://github.com/user-attachments/files/18519351/422-add-GD5Fx-nand-support.patch)

zeroneiko, 2025-01-23:
感谢大佬相助！成功编译出GD5F1GM7REYIG NAND的8818可用的固件

baili233, 2025-01-30:
> * [ ]  8818遇刷后重启概率回到原系统，感觉像是crash之后触发failsafe类机制了，有可能是uboot不一样，待商讨

我也遇到了，所以我两个rootfs都刷了

zeroneiko, 2025-02-05:
> > * [ ]   8818遇刷后重启概率回到原系统，感觉像是crash之后触发failsafe类机制了，有可能是uboot不一样，待商讨
> 
> 我也遇到了，所以我两个rootfs都刷了

手头有两台8818，一台23年一台24，实测用下来都有突然回滚系统的问题，初步推测不是自编译系统的问题

[openwrt-ipq50xx-arm-xiaomi_cr881x-squashfs-nand-factory.zip](https://github.com/user-attachments/files/18676850/openwrt-ipq50xx-arm-xiaomi_cr881x-squashfs-nand-factory.zip)

使用该魔改脚本刷写两个rootfs
```shell
# Check your partition table
# cat /proc/mtd
[[ "$(grep '"rootfs"' /proc/mtd | cut -d':' -f1)" == "mtd18" ]] || exit
[[ "$(grep '"rootfs_1"' /proc/mtd | cut -d':' -f1)" == "mtd19" ]] || exit

# Detect the current system slot and flash into the other one
# cat /proc/cmdline
mtd="$(grep -oE 'ubi.mtd=[a-zA-Z0-9\-\_]*' /proc/cmdline | cut -d'=' -f2)"
if [[ "$mtd" == "rootfs" ]]; then
    # Flash it as system 2
    ubiformat /dev/mtd19 -f openwrt-ipq50xx-arm-xiaomi_cr881x-squashfs-nand-factory.ubi
    #v2防回滚措施
    ubiformat /dev/mtd18 -f openwrt-ipq50xx-arm-xiaomi_cr881x-squashfs-nand-factory.ubi
    nvram set flag_try_sys2_failed=0
    nvram set flag_boot_rootfs=1
    nvram set flag_last_success=1
    nvram commit
elif [[ "$mtd" == "rootfs_1" ]]; then
    # Flash it as system 1
    ubiformat /dev/mtd18 -f openwrt-ipq50xx-arm-xiaomi_cr881x-squashfs-nand-factory.ubi
    #v2防回滚措施
    ubiformat /dev/mtd19 -f openwrt-ipq50xx-arm-xiaomi_cr881x-squashfs-nand-factory.ubi
    nvram set flag_try_sys1_failed=0
    nvram set flag_boot_rootfs=0
    nvram set flag_last_success=0
    nvram commit
fi

# Reboot
reboot
```

vietanhtwdk, 2026-10-24:
> > > * [ ]    8818遇刷后重启概率回到原系统,感觉像是crash之后触发failsafe类机制了,有可能是uboot不一样,待商讨
> > 
> > 
> > 我也遇到了,所以我两个rootfs都刷了
> 
> 手头有两台8818,一台23年一台24,实测用下来都有突然回滚系统的问题,初步推测不是自编译系统的问题
> 
> [openwrt-ipq50xx-arm-xiaomi_cr881x-squashfs-nand-factory.zip](https://github.com/user-attachments/files/18676850/openwrt-ipq50xx-arm-xiaomi_cr881x-squashfs-nand-factory.zip)
> 
> 使用该魔改脚本刷写两个rootfs
> 
> # Check your partition table
> # cat /proc/mtd
> [[ "$(grep '"rootfs"' /proc/mtd | cut -d':' -f1)" == "mtd18" ]] || exit
> [[ "$(grep '"rootfs_1"' /proc/mtd | cut -d':' -f1)" == "mtd19" ]] || exit
> 
> # Detect the current system slot and flash into the other one
> # cat /proc/cmdline
> mtd="$(grep -oE 'ubi.mtd=[a-zA-Z0-9\-\_]*' /proc/cmdline | cut -d'=' -f2)"
> if [[ "$mtd" == "rootfs" ]]; then
>     # Flash it as system 2
>     ubiformat /dev/mtd19 -f openwrt-ipq50xx-arm-xiaomi_cr881x-squashfs-nand-factory.ubi
>     #v2防回滚措施
>     ubiformat /dev/mtd18 -f openwrt-ipq50xx-arm-xiaomi_cr881x-squashfs-nand-factory.ubi
>     nvram set flag_try_sys2_failed=0
>     nvram set flag_boot_rootfs=1
>     nvram set flag_last_success=1
>     nvram commit
> elif [[ "$mtd" == "rootfs_1" ]]; then
>     # Flash it as system 1
>     ubiformat /dev/mtd18 -f openwrt-ipq50xx-arm-xiaomi_cr881x-squashfs-nand-factory.ubi
>     #v2防回滚措施
>     ubiformat /dev/mtd19 -f openwrt-ipq50xx-arm-xiaomi_cr881x-squashfs-nand-factory.ubi
>     nvram set flag_try_sys1_failed=0
>     nvram set flag_boot_rootfs=0
>     nvram set flag_last_success=0
>     nvram commit
> fi
> 
> # Reboot
> reboot
> 感谢各位佬的指点,发在这里作此备忘

Thanks for sharing, this work with cr8808 ver.b aka cr8818. Also for anyone install this remember to regen ssh key 

```
   rm /etc/dropbear/dropbear_*_host_key && /etc/init.d/dropbear restart
```