hzyitc/openwrt-redmi-ax3000 #92: 5.4分支选择64位编译报错

baili233, 2024-09-29:
代码未作修改，分支：ipq50xx-qsdk-kernel-5.4-openwrt-21.02-qsdk-11.5.05.841.1029，架构选择了aarch64，cr8818
[image]

baili233, 2024-09-29:
顺便分享一下，改512m内存的方法
[image]
[image]
使用的内存型号是k4b4g1646e bcma，cdt使用的是https://github.com/lu-1991/cdt-gpt-ipq60xx-generate
这个工具生成的cdt-AP-MP03.3_512M16_DDR3_LM512.bin
[cdt-AP-MP03.3_512M16_DDR3_LM512.zip](https://github.com/user-attachments/files/17179274/cdt-AP-MP03.3_512M16_DDR3_LM512.zip)
在uboot中刷入 flash 0:CDT flash 0:CDT_1

code-reboot-F, 2025-02-20:
你可以试试固件下面的的交叉编译工具toolchain，替换原有的，我也不确定会不会有用

不过还有一种邪门方法，就是编译加入V=s，找到“Werror”前的错误所在行和错误类型，直接修改源码，大部分问题都可以百度到，但是编出来的玩意能不能用就是另外一回事了😂

hzyitc, 2025-05-13:
https://github.com/hzyitc/openwrt-redmi-ax3000/blob/ipq50xx-qsdk-kernel-5.4-openwrt-21.02-qsdk-11.5.05.841.1029/.github/workflows/ci.yml

closed 2025-05-13 by hzyitc