hzyitc/openwrt-redmi-ax3000 #99: mt76 failed to build in GitHub Action when I added `passwall` app. / 当我添加 passwall 插件时, CI 中的 mt76 编译出错.

WhaleFell, 2024-12-03:
This is my CI link: [GitHub Action Log](https://github.com/Undefined-R-0/openwrt-redmi-ax3000-offical/actions/runs/12133755239/job/33829801621)

Related commit: https://github.com/Undefined-R-0/openwrt-redmi-ax3000-offical/tree/fba463174199ff002922ad299031a99e66a819d6

[image]

WhaleFell, 2024-12-05:
`make -j1 V=s` log:

```log
make[3]: Entering directory '/home/adminbobo/boboworkspace/openwrt-redmi-ax3000-offical/feeds/packages/utils/gl-mifi-mcu'
rm -f /home/adminbobo/boboworkspace/openwrt-redmi-ax3000-offical/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq50xx_arm/gl-mifi-mcu-1/.built
touch /home/adminbobo/boboworkspace/openwrt-redmi-ax3000-offical/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq50xx_arm/gl-mifi-mcu-1/.built_check
make -C /home/adminbobo/boboworkspace/openwrt-redmi-ax3000-offical/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq50xx_arm/linux-5.4-qsdk-11.5.0.5 KCFLAGS="-ffile-prefix-map=/home/adminbobo/boboworkspace/openwrt-redmi-ax3000-offical/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi=target-arm_cortex-a7+neon-vfpv4_musl_eabi -fno-caller-saves " HOSTCFLAGS="-O2 -I/home/adminbobo/boboworkspace/openwrt-redmi-ax3000-offical/staging_dir/host/include -I/home/adminbobo/boboworkspace/openwrt-redmi-ax3000-offical/staging_dir/hostpkg/include -I/home/adminbobo/boboworkspace/openwrt-redmi-ax3000-offical/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/host/include -Wall -Wmissing-prototypes -Wstrict-prototypes" CROSS_COMPILE="arm-openwrt-linux-muslgnueabi-" ARCH="arm" KBUILD_HAVE_NLS=no KBUILD_BUILD_USER="" KBUILD_BUILD_HOST="" KBUILD_BUILD_TIMESTAMP="Mon Apr 17 13:15:36 2023" KBUILD_BUILD_VERSION="0" HOST_LOADLIBES="-L/home/adminbobo/boboworkspace/openwrt-redmi-ax3000-offical/staging_dir/host/lib" KBUILD_HOSTLDLIBS="-L/home/adminbobo/boboworkspace/openwrt-redmi-ax3000-offical/staging_dir/host/lib" CONFIG_SHELL="bash" V=''  cmd_syscalls=  M="/home/adminbobo/boboworkspace/openwrt-redmi-ax3000-offical/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq50xx_arm/gl-mifi-mcu-1" modules
make[4]: Entering directory '/home/adminbobo/boboworkspace/openwrt-redmi-ax3000-offical/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq50xx_arm/linux-5.4-qsdk-11.5.0.5'
  CC [M]  /home/adminbobo/boboworkspace/openwrt-redmi-ax3000-offical/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq50xx_arm/gl-mifi-mcu-1/module.o
/home/adminbobo/boboworkspace/openwrt-redmi-ax3000-offical/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq50xx_arm/gl-mifi-mcu-1/module.c:196:20: error: conflicting types for built-in function 'exit' [-Werror=builtin-declaration-mismatch]
 static void __exit exit(void)
                    ^~~~
cc1: all warnings being treated as errors
make[5]: *** [scripts/Makefile.build:297: /home/adminbobo/boboworkspace/openwrt-redmi-ax3000-offical/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq50xx_arm/gl-mifi-mcu-1/module.o] Error 1
make[4]: *** [Makefile:1734: /home/adminbobo/boboworkspace/openwrt-redmi-ax3000-offical/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq50xx_arm/gl-mifi-mcu-1] Error 2
make[4]: Leaving directory '/home/adminbobo/boboworkspace/openwrt-redmi-ax3000-offical/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq50xx_arm/linux-5.4-qsdk-11.5.0.5'
make[3]: *** [Makefile:39: /home/adminbobo/boboworkspace/openwrt-redmi-ax3000-offical/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq50xx_arm/gl-mifi-mcu-1/.built] Error 2
make[3]: Leaving directory '/home/adminbobo/boboworkspace/openwrt-redmi-ax3000-offical/feeds/packages/utils/gl-mifi-mcu'
time: package/feeds/packages/gl-mifi-mcu/compile#1.27#0.63#1.87
    ERROR: package/feeds/packages/gl-mifi-mcu failed to build.
make[2]: *** [package/Makefile:116: package/feeds/packages/gl-mifi-mcu/compile] Error 1
make[2]: Leaving directory '/home/adminbobo/boboworkspace/openwrt-redmi-ax3000-offical'
make[1]: *** [package/Makefile:110: /home/adminbobo/boboworkspace/openwrt-redmi-ax3000-offical/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/adminbobo/boboworkspace/openwrt-redmi-ax3000-offical'
make: *** [/home/adminbobo/boboworkspace/openwrt-redmi-ax3000-offical/include/toplevel.mk:230: world] Error 2
```

gl-mifi-mcu failed to build. I think there's a problem with the libraries that Passwall depends on.

eritpchy, 2025-01-24:
根据项目中的ci.yml， 出错好像是必然的，应该忽略错误， 改了这个后面还有其他错😊
https://github.com/openwrt/packages/commit/7118a45a9b091439348cd5341bcd632e1b2a90e8
sed -i 's/(init/(gl_mifi_mcu_init/g' feeds/packages/utils/gl-mifi-mcu/src/module.c
sed -i 's/ init/ gl_mifi_mcu_init/g' feeds/packages/utils/gl-mifi-mcu/src/module.c
sed -i 's/(exit/(gl_mifi_mcu_exit/g' feeds/packages/utils/gl-mifi-mcu/src/module.c
sed -i 's/ exit/ gl_mifi_mcu_exit/g' feeds/packages/utils/gl-mifi-mcu/src/module.c

hzyitc, 2025-02-08:
忽略即可。qsdk的版本，有不少内核模块都会编译错误。因为内核版本和op版本并不是完全匹配的。高通内核改了不少东西，懒的修了。

加个IGNORE_ERRORS=1就行。