hzyitc/openwrt-redmi-ax3000 #103: CMCC RAX3000Q怎么用交换机设置双wan？

Tzwxyaa, 2025-01-13:
交换机显示的CPU (eth0)为未连接状态，如下图。
[image]
这样如何设置双wan？

hzyitc, 2025-02-08:
不用管eth0的链路状态，实际上是up的。直接配置即可。

hzyitc, 2025-02-08:
而且双wan跟cpu链路不相关，可以通过vlan实现的。

borintkkk, 2025-03-05:
> 交换机显示的CPU (eth0)为未连接状态，如下图。 [image] 这样如何设置双wan？

用vlan ，新增vlanid 10 ， cpu eth0  已标记， 然后 需要的wan2 口改成未标记，  最后，添加wan口的时候， 硬件接口选择 eth0.10，或者手动输入eth0.10

borintkkk, 2025-03-05:
> > 交换机显示的CPU (eth0)为未连接状态，如下图。 [image] 这样如何设置双wan？
> 
> 用vlan ，新增vlanid 10 ， cpu eth0 已标记， 然后 需要的wan2 口改成未标记， 最后，添加wan口的时候， 硬件接口选择 eth0.10，或者手动输入eth0.10

[image]

[image]