hzyitc/openwrt-redmi-ax3000 #104: kmod-tun缺失，无法安装Tailscale或者ZeroTier，望修复

RiceCloud, 2025-01-16:
kmod-tun缺失，无法安装Tailscale或者ZeroTier，望修复

eritpchy, 2025-01-24:
https://github.com/eritpchy/openwrt-redmi-ax3000 重新build了一个看起来是有kmod tun的不过好像openwrt 太老了没有luci-app-zerotier这个包， 你可以试试

RiceCloud, 2025-01-28:
> [https://github.com/eritpcy/openwrt-redmi-ax3000](https://github.com/eritpchy/openwrt-redmi-ax3000)重新构建了一个看起来有kmod tun的不过好像openwrt太老了没有luci-app-zerotier这个包，你可以尝试一下

2025-01-24 15:52:22的版本依然没有这个模块，报错cannot find dependency kmod-tun for zerotier。看到有说需要编译时候Kernel modules > Network Support > kmod-tun添加进去

eritpchy, 2025-01-28:
刚看了一眼，有的， 不仅我编译的版本有 hzyitc的也有， 可能是你的路由访问github.io被墙了 下载不到core package， 你也可以上传ipk手动安装
zerotier可以装的 我前几天装过， 不过由于我搞不定fullcore nat 改用其他的了

https://github.com/hzyitc/openwrt-redmi-ax3000/blob/gh-pages/ipq50xx-qsdk-kernel-5.4-openwrt-21.02-qsdk-11.5.05.841.1029/ci-20240727-173350-ab1f9ffa/kmod-tun_5.4-qsdk-11.5.0.5-1_arm_cortex-a7_neon-vfpv4.ipk

https://github.com/eritpchy/openwrt-redmi-ax3000/blob/gh-pages/ipq50xx-qsdk-kernel-5.4-openwrt-21.02-qsdk-11.5.05.841.1029/ci-20250124-155222-6f7246e1/kmod-tun_5.4-qsdk-11.5.0.5-1_arm_cortex-a7_neon-vfpv4.ipk

[image]

RiceCloud, 2025-01-28:
> 刚看了一眼，有的， 不仅我编译的版本有 hzyitc的也有， 可能是你的路由访问github.io被墙了 下载不到core package， 你也可以上传ipk手动安装 zerotier可以装的 我前几天装过， 不过由于我搞不定fullcore nat 改用其他的了 [...]

还真是，github.io的core package从路由无法访问，浏览器却可以直接下载，很奇怪。感谢大佬在除夕的百忙之中详细解答，祝您新年快乐，阖家欢乐。

closed 2025-02-08 by hzyitc