hzyitc/openwrt-redmi-ax3000 #108: 小米ax3000折腾后不通电了

obesewang, 2025-02-25:
之前尝试刷入uboot然后断电reset开机发现进不去uboot,并且官方修复工具也无效.
查看mtd分区发现mtd11和mtd12的分区文件名好像不对,但是尝试重新写入文件发现好像无效.
就手贱尝试擦除了mtd11,结果现在上电发现完全不通电了,求问现在还有什么挽救方法吗?

ps:刷回官方rom就好,因为发现刷了openwrt后wifi问题很多

yyangv, 2025-03-22:
我按照这个教程：https://mao.fan/article/195

我还特意把文件下载到路由器/tmp里面，跑的脚本，重启就没有任何反应了。

```txt
root@XiaoQiang:/tmp# ./run.sh 
dev:    size   erasesize  name
mtd0: 00080000 00020000 "0:SBL1"
mtd1: 00080000 00020000 "0:MIBIB"
mtd2: 00040000 00020000 "0:BOOTCONFIG"
mtd3: 00040000 00020000 "0:BOOTCONFIG1"
mtd4: 00100000 00020000 "0:QSEE"
mtd5: 00100000 00020000 "0:QSEE_1"
mtd6: 00040000 00020000 "0:DEVCFG"
mtd7: 00040000 00020000 "0:DEVCFG_1"
mtd8: 00040000 00020000 "0:CDT"
mtd9: 00040000 00020000 "0:CDT_1"
mtd10: 00080000 00020000 "0:APPSBLENV"
mtd11: 00140000 00020000 "0:APPSBL"
mtd12: 00140000 00020000 "0:APPSBL_1"
mtd13: 00100000 00020000 "0:ART"
mtd14: 00080000 00020000 "0:TRAINING"
mtd15: 00080000 00020000 "bdata"
mtd16: 00080000 00020000 "crash"
mtd17: 00080000 00020000 "crash_syslog"
mtd18: 02400000 00020000 "rootfs"
mtd19: 02400000 00020000 "rootfs_1"
mtd20: 01f00000 00020000 "overlay"
mtd21: 00080000 00020000 "rsvd0"
mtd22: 00364000 0001f000 "kernel"
mtd23: 0120b000 0001f000 "ubi_rootfs"
mtd24: 01b20000 0001f000 "data"
ubi.mtd=rootfs_1 root=mtd:ubi_rootfs rootfstype=squashfs cnss2.bdf_integrated=0x24 cnss2.bdf_pci0=0x60 cnss2.bdf_pci1=0x60 cnss2.skip_radio_bmap=4 rootwait swiotlb=1
ubiformat: mtd18 (nand), size 37748736 bytes (36.0 MiB), 288 eraseblocks of 131072 bytes (128.0 KiB), min. I/O size 2048 bytes
libscan: scanning eraseblock 287 -- 100 % complete  
ubiformat: 288 eraseblocks have valid erase counter, mean value is 1
ubiformat: flashing eraseblock 90 -- 100 % complete  
ubiformat: formatting eraseblock 287 -- 100 % complete  
root@XiaoQiang:/tmp# Connection to 192.168.31.1 closed by remote host.
```

nj-dw, 2025-10-02:
我也遇到了，刷机执行sh后重启路由器就不通电了，请问怎么处理！不通电还怎么刷还原？

trigsoft, 2025-10-19:
我是刷了大 root 分区那个就不行了。
wifi 不行，有试过 openwrt23版本吗？

WTepiphany, 2026-09-07:
我也是，怎么刷回官方的rom啊