hzyitc/openwrt-redmi-ax3000 #113: 刷如openwrt后如何通过ssh再升级openwrt？

mykofzone, 2025-04-21:
没有UART，通过ssh升级有问题（如下），还可以通过其他方式升级么？譬如openwrt的web升级？

一开始问题是nvram找不到，后来改用fw_setenv，可以设置成功，但是发现开机后还是这个固件，还需要设置什么么？

mtd="$(grep -oE 'ubi.mtd=[a-zA-Z0-9\-\_]*' /proc/cmdline | cut -d'=' -f2)"
if [[ "$mtd" == "rootfs" ]]; then
	# Flash it as system 2
	ubiformat /dev/mtd19 -f openwrt-ipq50xx-arm-redmi_ax3000-squashfs-nand-factory.ubi
	fw_setenv flag_try_sys2_failed 0
	fw_setenv flag_boot_rootfs 1
	fw_setenv flag_last_success 1
	#nvram commit
elif [[ "$mtd" == "rootfs_1" ]]; then
	# Flash it as system 1
	ubiformat /dev/mtd18 -f openwrt-ipq50xx-arm-redmi_ax3000-squashfs-nand-factory.ubi
	fw_setenv flag_try_sys1_failed 0
	fw_setenv flag_boot_rootfs 0
	fw_setenv flag_last_success 0
	#nvram commit
fi

mykofzone, 2025-04-30:
我昨天发现之前用ssh刷如的版本好像生效了，难道是要刷如后断电才行？我之前是reboot的。

swingsc2019, 2025-06-08:
好多插件都安装不上说缺少依赖欸，想更新到24.10版本不，哪位大佬有固件给我刷一下不