hzyitc/openwrt-redmi-ax3000 #116: 请问大佬 ，在ubuntu 20.04 下编译ipq50xx-qsdk-kernel-5.4-openwrt-21.02-qsdk-11.5.05.841.1029报错debugfs_netdev.c:139:6: error: passing argument 5 of 'ieee80211_if_read' from incompatible pointer type [-Werror=incompatible-pointer-types]怎么解决？

yimin8, 2025-05-05:
请问大佬 ，在ubuntu 20.04 下编译ipq50xx-qsdk-kernel-5.4-openwrt-21.02-qsdk-11.5.05.841.1029报错debugfs_netdev.c:139:6: error: passing argument 5 of 'ieee80211_if_read' from incompatible pointer type [-Werror=incompatible-pointer-types]怎么解决？详细信息如下：

make -C "/home/yimin/openwrt-redmi-ax3000/build_dir/target-aarch64_cortex-a53_musl/linux-ipq50xx_aarch64/backports-20210222-5.4-qsdk-11.5.0.5" KCFLAGS="-fmacro-prefix-map=/home/yimin/openwrt-redmi-ax3000/build_dir/target-aarch64_cortex-a53_musl=target-aarch64_cortex-a53_musl -fno-caller-saves " HOSTCFLAGS="-O2 -I/home/yimin/openwrt-redmi-ax3000/staging_dir/host/include -I/home/yimin/openwrt-redmi-ax3000/staging_dir/hostpkg/include -I/home/yimin/openwrt-redmi-ax3000/staging_dir/target-aarch64_cortex-a53_musl/host/include -Wall -Wmissing-prototypes -Wstrict-prototypes" CROSS_COMPILE="aarch64-openwrt-linux-musl-" ARCH="arm64" KBUILD_HAVE_NLS=no KBUILD_BUILD_USER="" KBUILD_BUILD_HOST="" KBUILD_BUILD_TIMESTAMP="Mon Apr 17 13:15:36 2023" KBUILD_BUILD_VERSION="0" HOST_LOADLIBES="-L/home/yimin/openwrt-redmi-ax3000/staging_dir/host/lib" KBUILD_HOSTLDLIBS="-L/home/yimin/openwrt-redmi-ax3000/staging_dir/host/lib" CONFIG_SHELL="bash" V=''  cmd_syscalls= KBUILD_EXTRA_SYMBOLS="/home/yimin/openwrt-redmi-ax3000/build_dir/target-aarch64_cortex-a53_musl/linux-ipq50xx_aarch64/symvers/qca-nss-clients.symvers /home/yimin/openwrt-redmi-ax3000/build_dir/target-aarch64_cortex-a53_musl/linux-ipq50xx_aarch64/symvers/gpio-button-hotplug.symvers /home/yimin/openwrt-redmi-ax3000/build_dir/target-aarch64_cortex-a53_musl/linux-ipq50xx_aarch64/symvers/qca-nss-drv.symvers /home/yimin/openwrt-redmi-ax3000/build_dir/target-aarch64_cortex-a53_musl/linux-ipq50xx_aarch64/symvers/qca-nss-dp.symvers /home/yimin/openwrt-redmi-ax3000/build_dir/target-aarch64_cortex-a53_musl/linux-ipq50xx_aarch64/symvers/qca-ssdk.symvers" EXTRA_CFLAGS="-I/home/yimin/openwrt-redmi-ax3000/build_dir/target-aarch64_cortex-a53_musl/linux-ipq50xx_aarch64/backports-20210222-5.4-qsdk-11.5.0.5/include -fmacro-prefix-map=/home/yimin/openwrt-redmi-ax3000/build_dir/target-aarch64_cortex-a53_musl/linux-ipq50xx_aarch64/backports-20210222-5.4-qsdk-11.5.0.5=backports-20210222-5.4-qsdk-11.5.0.5  -I/home/yimin/openwrt-redmi-ax3000/staging_dir/target-aarch64_cortex-a53_musl/usr/include/qca-nss-drv -I/home/yimin/openwrt-redmi-ax3000/staging_dir/target-aarch64_cortex-a53_musl/usr/include/qca-nss-clients" KLIB_BUILD="/home/yimin/openwrt-redmi-ax3000/build_dir/target-aarch64_cortex-a53_musl/linux-ipq50xx_aarch64/linux-5.4-qsdk-11.5.0.5" MODPROBE=true KLIB=/lib/modules/5.4.164 KERNEL_SUBLEVEL=4 KBUILD_LDFLAGS_MODULE_PREREQ= olddefconfig
make[4]: Entering directory '/home/yimin/openwrt-redmi-ax3000/build_dir/target-aarch64_cortex-a53_musl/linux-ipq50xx_aarch64/backports-20210222-5.4-qsdk-11.5.0.5'
Generating local configuration database from kernel ... done.
make[6]: 'conf' is up to date.
boolean symbol CRYPTO_LIB_ARC4 tested for 'm'? test forced to 'n'
.config:5:warning: symbol value 'm' invalid for ATH_REG_DYNAMIC_USER_REG_HINTS
#
# configuration written to .config
#
make[4]: Leaving directory '/home/yimin/openwrt-redmi-ax3000/build_dir/target-aarch64_cortex-a53_musl/linux-ipq50xx_aarch64/backports-20210222-5.4-qsdk-11.5.0.5'
rm -rf /home/yimin/openwrt-redmi-ax3000/build_dir/target-aarch64_cortex-a53_musl/linux-ipq50xx_aarch64/backports-20210222-5.4-qsdk-11.5.0.5/modules
make  -C "/home/yimin/openwrt-redmi-ax3000/build_dir/target-aarch64_cortex-a53_musl/linux-ipq50xx_aarch64/backports-20210222-5.4-qsdk-11.5.0.5" KCFLAGS="-fmacro-prefix-map=/home/yimin/openwrt-redmi-ax3000/build_dir/target-aarch64_cortex-a53_musl=target-aarch64_cortex-a53_musl -fno-caller-saves " HOSTCFLAGS="-O2 -I/home/yimin/openwrt-redmi-ax3000/staging_dir/host/include -I/home/yimin/openwrt-redmi-ax3000/staging_dir/hostpkg/include -I/home/yimin/openwrt-redmi-ax3000/staging_dir/target-aarch64_cortex-a53_musl/host/include -Wall -Wmissing-prototypes -Wstrict-prototypes" CROSS_COMPILE="aarch64-openwrt-linux-musl-" ARCH="arm64" KBUILD_HAVE_NLS=no KBUILD_BUILD_USER="" KBUILD_BUILD_HOST="" KBUILD_BUILD_TIMESTAMP="Mon Apr 17 13:15:36 2023" KBUILD_BUILD_VERSION="0" HOST_LOADLIBES="-L/home/yimin/openwrt-redmi-ax3000/staging_dir/host/lib" KBUILD_HOSTLDLIBS="-L/home/yimin/openwrt-redmi-ax3000/staging_dir/host/lib" CONFIG_SHELL="bash" V=''  cmd_syscalls= KBUILD_EXTRA_SYMBOLS="/home/yimin/openwrt-redmi-ax3000/build_dir/target-aarch64_cortex-a53_musl/linux-ipq50xx_aarch64/symvers/qca-nss-clients.symvers /home/yimin/openwrt-redmi-ax3000/build_dir/target-aarch64_cortex-a53_musl/linux-ipq50xx_aarch64/symvers/gpio-button-hotplug.symvers /home/yimin/openwrt-redmi-ax3000/build_dir/target-aarch64_cortex-a53_musl/linux-ipq50xx_aarch64/symvers/qca-nss-drv.symvers /home/yimin/openwrt-redmi-ax3000/build_dir/target-aarch64_cortex-a53_musl/linux-ipq50xx_aarch64/symvers/qca-nss-dp.symvers /home/yimin/openwrt-redmi-ax3000/build_dir/target-aarch64_cortex-a53_musl/linux-ipq50xx_aarch64/symvers/qca-ssdk.symvers" EXTRA_CFLAGS="-I/home/yimin/openwrt-redmi-ax3000/build_dir/target-aarch64_cortex-a53_musl/linux-ipq50xx_aarch64/backports-20210222-5.4-qsdk-11.5.0.5/include -fmacro-prefix-map=/home/yimin/openwrt-redmi-ax3000/build_dir/target-aarch64_cortex-a53_musl/linux-ipq50xx_aarch64/backports-20210222-5.4-qsdk-11.5.0.5=backports-20210222-5.4-qsdk-11.5.0.5  -I/home/yimin/openwrt-redmi-ax3000/staging_dir/target-aarch64_cortex-a53_musl/usr/include/qca-nss-drv -I/home/yimin/openwrt-redmi-ax3000/staging_dir/target-aarch64_cortex-a53_musl/usr/include/qca-nss-clients" KLIB_BUILD="/home/yimin/openwrt-redmi-ax3000/build_dir/target-aarch64_cortex-a53_musl/linux-ipq50xx_aarch64/linux-5.4-qsdk-11.5.0.5" MODPROBE=true KLIB=/lib/modules/5.4.164 KERNEL_SUBLEVEL=4 KBUILD_LDFLAGS_MODULE_PREREQ= modules
make[4]: Entering directory '/home/yimin/openwrt-redmi-ax3000/build_dir/target-aarch64_cortex-a53_musl/linux-ipq50xx_aarch64/backports-20210222-5.4-qsdk-11.5.0.5'
make[5]: 'Kconfig.versions' is up to date.
make[7]: 'Kconfig.versions' is up to date.
make[8]: 'conf' is up to date.
boolean symbol CRYPTO_LIB_ARC4 tested for 'm'? test forced to 'n'
#
# configuration written to .config
#
Building backport-include/backport/autoconf.h ... done.
  CC [M]  /home/yimin/openwrt-redmi-ax3000/build_dir/target-aarch64_cortex-a53_musl/linux-ipq50xx_aarch64/backports-20210222-5.4-qsdk-11.5.0.5/compat/backport-5.10.o
  CC [M]  /home/yimin/openwrt-redmi-ax3000/build_dir/target-aarch64_cortex-a53_musl/linux-ipq50xx_aarch64/backports-20210222-5.4-qsdk-11.5.0.5/compat/backport-5.11.o
  LD [M]  /home/yimin/openwrt-redmi-ax3000/build_dir/target-aarch64_cortex-a53_musl/linux-ipq50xx_aarch64/backports-20210222-5.4-qsdk-11.5.0.5/compat/compat.o
  CC [M]  /home/yimin/openwrt-redmi-ax3000/build_dir/target-aarch64_cortex-a53_musl/linux-ipq50xx_aarch64/backports-20210222-5.4-qsdk-11.5.0.5/drivers/net/wireless/ath/main.o
  CC [M]  /home/yimin/openwrt-redmi-ax3000/build_dir/target-aarch64_cortex-a53_musl/linux-ipq50xx_aarch64/backports-20210222-5.4-qsdk-11.5.0.5/drivers/net/wireless/ath/regd.o
  CC [M]  /home/yimin/openwrt-redmi-ax3000/build_dir/target-aarch64_cortex-a53_musl/linux-ipq50xx_aarch64/backports-20210222-5.4-qsdk-11.5.0.5/drivers/net/wireless/ath/hw.o
  CC [M]  /home/yimin/openwrt-redmi-ax3000/build_dir/target-aarch64_cortex-a53_musl/linux-ipq50xx_aarch64/backports-20210222-5.4-qsdk-11.5.0.5/drivers/net/wireless/ath/key.o
  CC [M]  /home/yimin/openwrt-redmi-ax3000/build_dir/target-aarch64_cortex-a53_musl/linux-ipq50xx_aarch64/backports-20210222-5.4-qsdk-11.5.0.5/drivers/net/wireless/ath/debug.o
  CC [M]  /home/yimin/openwrt-redmi-ax3000/build_dir/target-aarch64_cortex-a53_musl/linux-ipq50xx_aarch64/backports-20210222-5.4-qsdk-11.5.0.5/drivers/net/wireless/ath/dfs_pattern_detector.o
  CC [M]  /home/yimin/openwrt-redmi-ax3000/build_dir/target-aarch64_cortex-a53_musl/linux-ipq50xx_aarch64/backports-20210222-5.4-qsdk-11.5.0.5/drivers/net/wireless/ath/dfs_pri_detector.o
  LD [M]  /home/yimin/openwrt-redmi-ax3000/build_dir/target-aarch64_cortex-a53_musl/linux-ipq50xx_aarch64/backports-20210222-5.4-qsdk-11.5.0.5/drivers/net/wireless/ath/ath.o
  CC [M]  /home/yimin/openwrt-redmi-ax3000/build_dir/target-aarch64_cortex-a53_musl/linux-ipq50xx_aarch64/backports-20210222-5.4-qsdk-11.5.0.5/drivers/net/wireless/ath/ath11k/core.o
  CC [M]  /home/yimin/openwrt-redmi-ax3000/build_dir/target-aarch64_cortex-a53_musl/linux-ipq50xx_aarch64/backports-20210222-5.4-qsdk-11.5.0.5/drivers/net/wireless/ath/ath11k/hal.o
[... cut ...]
    ERROR: package/kernel/mac80211 failed to build (build variant: regular).
make[2]: *** [package/Makefile:116: package/kernel/mac80211/compile] Error 1
make[2]: Leaving directory '/home/yimin/openwrt-redmi-ax3000'
make[1]: *** [package/Makefile:110: /home/yimin/openwrt-redmi-ax3000/staging_dir/target-aarch64_cortex-a53_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/yimin/openwrt-redmi-ax3000'
make: *** [/home/yimin/openwrt-redmi-ax3000/include/toplevel.mk:230：world] 错误 2

hzyitc, 2025-05-13:
.config 的问题

参考 https://github.com/hzyitc/openwrt-redmi-ax3000/blob/ipq50xx-qsdk-kernel-5.4-openwrt-21.02-qsdk-11.5.05.841.1029/.github/workflows/ci.yml#L40

yimin8, 2025-05-18:
> .config 的问题
> 
> 参考 https://github.com/hzyitc/openwrt-redmi-ax3000/blob/ipq50xx-qsdk-kernel-5.4-openwrt-21.02-qsdk-11.5.05.841.1029/.github/workflows/ci.yml#L40

大佬，编译32位不会报错，是编译 64位就报错，是不支持64位吗？？？

yynag, 2025-07-21:
mac80211 编译错误修复：
```c
// build_dir/target-aarch64_cortex-a53_musl/linux-ipq50xx_aarch64/backports-20210222-5.4-qsdk-11.5.0.5/net/mac80211/debugfs_netdev.c

+ ssize_t ieee80211_if_fmt_bmiss_threshold(const struct ieee80211_sub_if_data *sdata,
- int ieee80211_if_fmt_bmiss_threshold(const struct ieee80211_sub_if_data *sdata,
				     char *buf, int buflen)
{
	return snprintf(buf, buflen, "%u\n", sdata->vif.bmiss_threshold);
}
```