hzyitc/openwrt-redmi-ax3000 #123: 5.4-qsdk已无法编译

Cathgao, 2025-05-29:
backports-20210222-5.4-qsdk-11.5.0.5.tar.xz 这个文件已经被删除

```
SHELL= flock /home/gao/openwrt-redmi-ax3000/tmp/.backports-20210222-5.4-qsdk-11.5.0.5.tar.xz.flock -c '  	/home/gao/openwrt-redmi-ax3000/scripts/download.pl "/home/gao/openwrt-redmi-ax3000/dl" "backports-20210222-5.4-qsdk-11.5.0.5.tar.xz" "skip" "" || (  mkdir -p /home/gao/openwrt-redmi-ax3000/tmp/dl && cd /home/gao/openwrt-redmi-ax3000/tmp/dl && rm -rf backports-20210222-5.4-qsdk-11.5.0.5 backports-source ath-kernel && mkdir -p backports-20210222-5.4-qsdk-11.5.0.5 backports-source ath-kernel && tar -C backports-source --strip-components=1 -xJf /home/gao/openwrt-redmi-ax3000/dl/backports-source-v5.12.19-1.tar.xz && tar -C ath-kernel --strip-components=1 -xJf /home/gao/openwrt-redmi-ax3000/dl/ath-kernel-2021-02-22-b157d227.tar.xz && ( cd /home/gao/openwrt-redmi-ax3000/tmp/dl/backports-source && PATH="/sbin:/home/gao/openwrt-redmi-ax3000/staging_dir/toolchain-arm_cortex-a7+neon-vfpv4_gcc-8.4.0_musl_eabi/bin:/home/gao/openwrt-redmi-ax3000/staging_dir/toolchain-arm_cortex-a7+neon-vfpv4_gcc-8.4.0_musl_eabi/bin:/home/gao/openwrt-redmi-ax3000/staging_dir/host/bin:/home/gao/openwrt-redmi-ax3000/staging_dir/host/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin:/snap/bin" ./gentree.py --clean /home/gao/openwrt-redmi-ax3000/tmp/dl/ath-kernel /home/gao/openwrt-redmi-ax3000/tmp/dl/backports-20210222-5.4-qsdk-11.5.0.5 ) && (cd /home/gao/openwrt-redmi-ax3000/tmp/dl; 	tar --numeric-owner --owner=0 --group=0 --mode=a-s --sort=name ${TAR_TIMESTAMP:+--mtime="$TAR_TIMESTAMP"} -c backports-20210222-5.4-qsdk-11.5.0.5 | 	xz -T0 -zc -7e > /home/gao/openwrt-redmi-ax3000/tmp/dl/backports-20210222-5.4-qsdk-11.5.0.5.tar.xz) && mv /home/gao/openwrt-redmi-ax3000/tmp/dl/backports-20210222-5.4-qsdk-11.5.0.5.tar.xz /home/gao/openwrt-redmi-ax3000/dl && rm -rf backports-20210222-5.4-qsdk-11.5.0.5 backports-source ath-kernel;  )    '
+ curl -f --connect-timeout 20 --retry 5 --location https://sources.cdn.openwrt.org/backports-20210222-5.4-qsdk-11.5.0.5.tar.xz
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0Receiving objects:  67% (2022/3012), 178.31 MiB | 207.00 K  0   146    0     0    0     0      0      0 --:--:--  0:00:01 --:--:--     0
curl: (22) The requested URL returned error: 404
Download failed.
+ curl -f --connect-timeout 20 --retry 5 --location https://sources.openwrt.org/backports-20210222-5.4-qsdk-11.5.0.5.tar.xz
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0Receiving objects:  67% (2022/3012), 178.62 MiB | 211.00 K  0     0    0     0    0     0      0      0 --:--:--  0:00:01 --:--:--     0Receiving objects:  67% (2024/3012), 178.98 MiB | 230.00 K  0   146    0     0    0     0      0      0 --:--:--  0:00:02 --:--:--     0
curl: (22) The requested URL returned error: 404
Download failed.
+ curl -f --connect-timeout 20 --retry 5 --location https://mirror2.openwrt.org/sources/backports-20210222-5.4-qsdk-11.5.0.5.tar.xz
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0Receiving objects:  67% (2024/3012), 179.23 MiB | 234.00 K  0   153    0     0    0     0      0      0 --:--:--  0:00:01 --:--:--     0
curl: (22) The requested URL returned error: 404
Download failed.
No more mirrors to try - giving up.
You need to have installed: spatch), 180.66 MiB | 183.00 KiB/s
Try installing the package: coccinelle
If that is too old go grab the code from source:

git clone https://github.com/coccinelle/coccinelle.git

To build you will need: ocaml ncurses-devel

If on SUSE / OpenSUSE you will also need: ocaml-ocamldoc

You have unfulfilled binary requirements
make[2]: *** [Makefile:656: /home/gao/openwrt-redmi-ax3000/dl/backports-20210222-5.4-qsdk-11.5.0.5.tar.xz] Error 1
make[2]: Leaving directory '/home/gao/openwrt-redmi-ax3000/package/kernel/mac80211'
time: package/kernel/mac80211/regular/download#553.09#49.41#170.64
    ERROR: package/kernel/mac80211 failed to build (build variant: regular).
make[1]: *** [package/Makefile:116: package/kernel/mac80211/download] Error 1
```

closed 2025-06-01 by Cathgao

zxc503, 2025-08-04:
我也遇到了这个问题，你解决了吗

6ccg, 2025-09-13:
same issue

Cathgao, 2025-09-13:
下载release里的dl解压到源码目录